fredlcore/BSB-LAN #392: DS18B20 and/or BME280 with ESP32 EVB

dukess, 2021-09-18:
I did small investigation today for possibility of connection temperature sensors to ESP32 EVB.
I'm disappointed. In short words current configuration doesn't allow to works with OneWire sensors because all pins are used for different hardware. I tried pins No 13 and 16 (i2c bus) but nothing happend and no sensors found on bus. Tried to flash simple sketch with Dallas example - it works. Right now i have no idea what thing blocking connection to sensors.
May try connect through pin 12? But i don't know how OneWire will interfere with IR LED on same pin.

So here i see only one possibility to use temperature sensors: BME280. But right now i haven't BME280 (delivery promised me at start of October) and can't test it.

fredlcore, 2021-09-18:
It's true that many pins are connected to hardware built onto the EVB which in some cases may interfere with attached sensors. But I would have thought that i2c would have worked because it's just bare pins and I thought that you can configure any pin for any kind of input/output, so I'm puzzled why it does not work in your setting.
If that should remain true, it would be worth noting which sensors won't be able to work on the EVB in the manual.

dukess, 2021-09-19:
I have good news. I found issue which looks similar for me: https://github.com/espressif/arduino-esp32/issues/755
And there I found fork of OneWire library which eliminate my problem: https://github.com/stickbreaker/OneWire
I think we need merge last improvements from original library to stickbreaker's fork and replace our current OneWire.

Now DS18B20 is working.

fredlcore, 2021-09-19:
Great, thanks, will stickbreaker's library also run on Arduino (Due)?

dukess, 2021-09-19:
Not tried yet. But by description and discussion his libarary is compatible with original version.

1coderookie, 2021-09-21:
Hey guys,
sorry for not answering the last days - I have a herniated disc right now and can't do anything (besides laying in the bed), not even sitting at the desk, so I'll be off for another few (..I hope..) days. I just got online to check my mails today and let you guys know..

> But I would have thought that i2c would have worked because it's just bare pins and I thought that you can configure any pin for any kind of input/output, so I'm puzzled why it does not work in your setting.

No, afaik one can't connect OneWire sensors to the I2C (or SPI) directly, because the I2C is a different bus protocol than the OneWire by Maxim/Dallas and therefore it isn't compatible. However, there is a piece of hardware available from Maxim/Dallas for about a year or so ([DS28E18](https://www.maximintegrated.com/en/products/embedded-security/secure-authenticators/DS28E18.html)) which allows to connect OneWire to that thing and then connect that to the I2C. But I don't have that one, so I can't test it or give further infos/recommendations about that..

fredlcore, 2021-09-21:
Oh, so sorry to hear that! Hope you get better soon!
As for 1Wire - yes, of course I2C is something completely different. What I meant is that these pins are labelled (and can be used for) I2C, but unless you write code that uses these pins for that purpose, they shouldn't be doing anythin "on their own". So if you want to use them for sensors of whatever type, that should be possible, provided that the code on the ESP is dealing with it correctly and no other device is attached to these pins that uses them otherwise. So, for example, you can't really use the pin that is connected to the IR sensor on the Olimex because if there is IR light coming in, the sensor will affect the pin and thus produce invalid results for whatever else you are using it. 
With I2C I thought that no device is connected to it on the Olimex so the labelling in that case is just a suggestion, if I'm seeing things correctly.

1coderookie, 2021-09-21:
Thanks!

I don't think that's correct, but of course I can be wrong. Afaik, the I2C pins are straightly 'coming from/going to' the I2C controller of the ESP itself, so you can only use them for [I2C](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2c.html). There are indeed pins which could or couldn't be used for SPI though, which also are connected to the I2C controller/bus, but the 'plain' I2C-labeled-pins should (imho) go straight and only to the I2C bus.
But as I said - I might be wrong..

dukess, 2021-09-21:
Get well soon, Ulf!
I was able to run the DS18B20 sensor on those contacts that are marked as i2c after OneWire library patching, so here one less problem.

By the way, we are interesting in possibility to use more than 2 BME280? They can be connected through TCA9548 multiplexer and required just few lines of code.
https://learn.adafruit.com/adafruit-tca9548a-1-to-8-i2c-multiplexer-breakout/wiring-and-test

1coderookie, 2021-09-21:
Thanks!

That's interesting and good to know, thanks for that info!

Multiplexer: Sounds interesting, maybe someone could/would use it. Up to me and my needs right now it wouldn't really be necessary, but if it doesn't take too much RAM and wouldn't be a problem somehow for Due/ESP, then I'd say: go for it! ;)

fredlcore, 2021-09-21:
AFAIK, there is no internal "bus" whatsoever in the ESP32, basically all pins can be used as GPIOs (except those that are hardwired to a specific function like flash etc.), the same goes for the I2C pins, see here:
https://www.olimex.com/forum/index.php?topic=7275.msg27738#msg27738
Even on an Arduino you can use the serial RX/TX pins as GPIOs, but as soon as the Arduino uses these for serial communication, it's going awry of course...

dukess, 2021-09-21:
@fredlcore, sure! In any case two free pins (13 and 16) are better than nothing. It is possible use as I2C or for 2 DHT22 or for OneWire + one DHT22.

dukess, 2021-09-22:
> Multiplexer: Sounds interesting, maybe someone could/would use it. Up to me and my needs right now it wouldn't really be necessary, but if it doesn't take too much RAM and wouldn't be a problem somehow for Due/ESP, then I'd say: go for it! ;)

Did it: https://github.com/fredlcore/BSB-LAN/pull/391/commits/2194cc0f36cd049e0e1c03122def02159553c477

fredlcore, 2021-09-23:
Thanks, I wrote my comments into the PR. Once this is settled, I can merge.

closed 2021-09-23 by fredlcore