fredlcore/BSB-LAN #393: [BUG] MQTT active and buggy in access point mode

wbeck77, 2021-09-23:
**BSB-LAN Version** master

**Architecture** ESP32

**Bus system** BSB

**Describe the bug**
If the SW enters access point mode and MQTT is active the web interface is unreachable.
I monitored the serial port. And its obvious trying to connect to the MQTT broker that's not accessible.
It looks like if the MQTT broker connection fails the retries to connect are blocking the web server to answer requests.
Therefore the webfrontend is extremely slow or stuck at all

**To Reproduce**
enable MQTT
disable your WLAN to enforce BSB-LAN to enter access point mode (192.168.4.1)
try to use the web frontend

fredlcore, 2021-09-23:
@dukess and @torbennehmer: I think @wbeck77 is right, once we enter access point mode because the ESP cannot connect to the wifi network, all functions that initiate connections to external addresses need to be disabled. What do you think?

dukess, 2021-09-23:
did it

closed 2021-09-23 by fredlcore

wbeck77, 2021-09-23:
you are great! thank you