fredlcore/BSB-LAN #395: Olimex ESP32 EVB. UEXT1 connector

dukess, 2021-09-25:
@fredlcore For future boards 7x2 connector will looks better for using: I didn't put it into the right position today and don't understand long time why i got "query failed".

By the way today i tried to use ESP32 EVB as autonomous device but can't did it: when i use +5V from PC USB port - device start fine. When i use USB power supply without adapter - MCU starts too. But when i use power supply with adapter MSU won't starts without reset.

I found very similar issue but haven't ideas how to resolve it. (Now i think about LiPo battery)
https://github.com/OLIMEX/ESP32-EVB/issues/22

fredlcore, 2021-09-25:
Ok, that is interesting. So far, we have only heard about this issue here:
https://www.olimex.com/forum/index.php?topic=7447.msg28187#msg28187
Where you need to press reset after powering on. But since I have REV I, that should no longer be a problem.

Your finding makes sense and it would be relatively easy to solve in a new board revision where we could use pin 5 or 6 for example (GPIO16 and 13 respectively) since the serial pins can easily be redefined. However, these are normally used for I2C communication, so I'm not sure if you would lose I2C capabilities with that move, even if you could reassign the i2C pins as well, but there are hardly any free pins left on the EVB.

So far, I'm a bit inclined to leave things as they are, because (correct me if I'm wrong) all you have to do is press the Reset button after turning on power, right? That is a bit of a nuisance, but how often do you power-cycle your Olimex in everyday use? Doing a soft reset via /N does not seem to cause any problems, at least here.
So if my observations are true, it is something you need to remember, but since most of the time you can also just press the reset button when power-cycling the Olimex, it should not be a show-stopper?

As for the 7x2 connectors, I thought of these as well, it's may just a bit weird when you look at the adapter and see 14 holes but only 10 pins on the Olimex. If we stick with 10 pins, we'll add a note in the manual.

fredlcore, 2021-09-25:
I'm not really good in electronics, but would it be an option to add a transistor which would prevent current from flowing on the adapter from the BSB bus to GPIO4 until (after booting) the transistor's base receives current (by setting a pin on the Olimex high) and thus current can flow to GPIO4. But then we'd need a different pin again, so I'm not sure if that really solves the issue.

dukess, 2021-09-27:
> So far, I'm a bit inclined to leave things as they are, because (correct me if I'm wrong) all you have to do is press the Reset button after turning on power, right?

I think too. This case can be added to manual for informing about potential troubles. In my installation i have surge protector relay which time-from-time disconnect my home from power network. But i can use LiPo battery as UPS as solution.

> I'm not really good in electronics, but would it be an option to add a transistor...
Same problem. :) 

> the adapter from the BSB bus to GPIO4 until (after booting) the transistor's base receives current (by setting a pin on the Olimex high) 

Or adding capacitor and some resistors and implementing delaying scheme. Pro: no need to use one more pin. Contra: bigger board, more components and more complex wiring needed.

fredlcore, 2021-09-28:
What I'm wondering: I can reproduce the problem of the Olimex hanging when powering the device up. But why is it no problem when I press the reset button or issue a software-initiated reset via /N? The boot sequence should be the same in all cases, shouldn't it?

Regarding 2X5-headers, look here for example:
https://www.aliexpress.com/item/4000602105321.html?spm=a2g0o.productlist.0.0.5dbd429aZU4zRJ&algo_pvid=b773f64c-36cb-4d62-9c79-6055fb05e5a8&algo_exp_id=b773f64c-36cb-4d62-9c79-6055fb05e5a8-11&pdp_ext_f=%7B%22sku_id%22%3A%2210000003762123666%22%7D

dukess, 2021-09-29:
The difference is most likely that the power supply is not interrupted.

I mean something like this but for board, not cable: https://aliexpress.ru/item/4000124619762.html?spm=a2g0o.productlist.0.0.22521d83RR56A4&algo_pvid=a98bf2ef-5a82-4f4a-aceb-7fe953d2e0d8&algo_expid=a98bf2ef-5a82-4f4a-aceb-7fe953d2e0d8-9&btsid=0b8b037016329096032465614e85e4&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

fredlcore, 2021-09-29:
Yes, the power supply is not interrupted, but if the state of GPIO4 is relevant during boot, it should affect it whether it's during power-on or plain reset, shouldn't it? But I'm no electronics expert...

Why would you need such a thing for the board?

dukess, 2021-10-01:
>Yes, the power supply is not interrupted, but if the state of GPIO4 is relevant during boot, it should affect it whether it's during power-on or plain reset, shouldn't it? But I'm no electronics expert...

Don't know too.

>Why would you need such a thing for the board?

I was just looking for such part so that I could advise using it in new boards later. :) But it not exist.

dukess, 2021-10-05:
Just FYI: I bought JST connector for battery. 
For ESP32 EVB board JST 2.0 PH connectors matched.

During the operation before battery connect, I encountered strange things: many errors "BSB-draht kurzschluss" appepared in boiler log. It can be depend on GPIO4 high state at startup i think.

fredlcore, 2021-10-05:
Hm, that's strange, because the GPIO4 is RX, so if the adapter prevents the EVB from booting, I would assume that it is reading GPIO4, and the adapter puts it into a "wrong" state and therefore it does not boot. 
If "Kurzschluss" is appearing in the logs, then the EVB would be sending data, but GPIO4 should not be an issue then because TX is on GPIO36, so the EVB would have to access GPIO36 randomly during boot. That's something I don't know about. Or do you have an explanation for that?

dukess, 2021-10-05:
Unfortunately, no. Just observation. Perhaps it is connected with something else altogether. I have another oddity: in less than a day, 8314 program stopped being polled correctly. I looked at the debugging output - an incorrect command was used there at all. The reboot helped restore the work. I do not exclude the possibility that we have memory leaks somewhere else.

fredlcore, 2021-10-05:
Have you logged that command? The good thing about the EVB is that it seems to mercilessly crash upon a memory leak. The Arduino would just go on until the code would result in a dead end, so I wonder whether that could really be the case, but of course there is a possibility. If you can replicate this behaviour, that would of course be great...

dukess, 2021-10-05:
Wrong command before reboot:
URL: /JQ=8314
LAN->HEIZ QUR 8314 ENUM_CAT_2d_TEXT - STR8314_TEXT:
DC C2 00 0B 06 00 05 02 1E F2 38
HEIZ->LAN ERR 8314 ENUM_CAT_2d_TEXT - STR8314_TEXT: error 7
DC 80 42 0C 08 05 00 02 1E 07 61 87
#8314:  (parameter not supported)

Correct command after reboot:
LAN->HEIZ QUR 8314 ENUM_CAT_2d_TEXT - STR8314_TEXT:
DC C2 00 0B 06 3D 11 05 1A A1 E7
HEIZ->LAN ANS 8314 ENUM_CAT_2d_TEXT - STR8314_TEXT: 29.7 UNIT_DEG_TEXT
DC 80 42 0E 07 11 3D 05 1A 00 07 6F 7D C8
#8314: 29.7 UNIT_DEG_TEXT

fredlcore, 2021-10-05:
Command ID 0500021E is correct for parameter 8314 for several device families, among others for DEV_NONE which is used when BSB-LAN was not connected to the heater during boot and thus device family is 0. So I assume that device family detection was wrong in some way?

dukess, 2021-10-05:
Clearly! In this case, this could happen if the device was rebooted at the moment when the boiler was disconnected from the electrical network. Yes, it is quite likely.

Perhaps we can add function for boiler family and type detection at runtime for this case. Something like this:
```

  if (millis() - maintenance_timer > 60000) {
    maintenance_timer = millis();
+    if(!my_dev_fam) {
+      SetDevId();
+    }
#if defined(WIFI) && defined(ESP32)
    if (WiFi.status() != WL_CONNECTED) {
      WiFi.disconnect();
```

fredlcore, 2021-10-05:
Great idea!

dukess, 2021-10-08:
@fredlcore Good morning Frederik, can you remember for me why you disabled watchdog for ESP32? It cause problems? Why i ask: my device has been hanging three times this week.

fredlcore, 2021-10-08:
Can you tell me which watchdog you mean and where? I cound only find watch_sockets after a quick search...

dukess, 2021-10-08:
These lines are disabled:

#include <esp_task_wdt.h>
...
#ifdef ESP32
    esp_task_wdt_init(10,true);
    esp_task_wdt_add(NULL);
#endif
...
#ifdef ESP32
        esp_task_wdt_reset();
#endif

dukess, 2021-10-08:
Ok, i will to test it.

> But apart from that we should figure out what leaves your ESP hanging - is the serial output any indication?

Don't know yet because MCU installed on remote location and not connected to any PC/NAS. 
Right now i have only one theory: SD-card + LAN is still unstable.

fredlcore, 2021-10-09:
Could the watchdog also help with the power on problem?

fredlcore, 2021-10-10:
I just found this here:
https://www.olimex.com/forum/index.php?topic=8093.msg30707#msg30707
Could anyone test if this solves the problem? Adding a pulldown resistor between GPIO4 and GND is relatively easy, both to existing boards as well as a potential new revision as GPIO4 and GND are right next to each other on the UEXT1 connector.

The question is a) if this solves the problem during boot when the adapter is plugged in and b) if the pull down resistor is interfering with the incoming data. I'll try to test once I'm back home, but maybe @dukess and others can test this as well?

dukess, 2021-10-10:
> Adding a pulldown resistor between GPIO4 and GND is relatively easy, both 
> to existing boards as well as a potential new revision as GPIO4 and GND are 
> right next to each other on the UEXT1 connector.

Did it. Working fine.

dukess, 2021-10-10:
> Awesome! Which resistance did you choose and where exactly did you put it?

I have 1k resistors so i used it for fast testing. I soldered resistor 
between gnd (2) and tx (3) pins right on uext1 connector.
> if the pull down resistor is interfering with the incoming data.

I have not found any problems yet, except for the fact that another entry 
about a short circuit on the bus appeared in the boiler log when connecting 
power to the controller.

fredlcore, 2021-10-11:
You are right, of course, I mixed up pin 4 of the UEXT1 with GPIO4 ;)...

fredlcore, 2021-10-11:
Ok, I checked and it works with 4.7k as well - I assume that a higher value for a pull-down is better than a smaller one. It's also easy to solder it onto the board when using the outermost row of the board. Luckily, I added markers for the TX and GND pins on that row, so the resistor is easy to add there. I'm thinking of adding a proper location for it in the next board revision, but I'm glad we found and fixed this issue :)
[image]
!

closed 2021-10-11 by fredlcore

fredlcore, 2021-10-11:
My first one was ugly, too, until I remembered that those pins are in the outer row as well ;)...