fredlcore/BSB-LAN #405: Where do I connect the BSB to my RVS21 card ?

Littledodger, 2021-10-16:
Hi there !
I'm looking at this solution to control my heating system (I have a atlantic duo extensa system) but I don't know where I have to connect the BSB bus to the RVS21.831/127 of my system...
On the X86 connector, I have these pins : B2 - M - B1 - 3 - 2 - 1.
Is the pin 2 correspond to CL- and the pin 1 to CL+ ?

Tanks :)

1coderookie, 2021-10-16:
Hi,

sorry, but I'm actually not 100% sure about that, because I found different sayings in different manuals (in one of them, an Atlantic Alfea manual, it says its terminal 2&3). 

_So please look up in your specific system manual again for how to connect the room unit / room sensor._ 
What does that say? Which pins are the two to connect the room sensor unit?

Once you found the correct two terminals and just don't know which one of them is CL+/-, then you can just try either way. If you connect it wrong (CL+ to CL- and vice versa) it shouldn't do any harm, the adapter just won't work. When you found the correct terminals and connect CL+ to CL+ & CL- to CL- correctly, then the red LED at the adapter gets lit and you have access. Then please let us know, which terminals you used.

1coderookie, 2021-10-16:
You're welcome :)
Your link doesn't work for me here btw..

Do you already have a room unit installed? If so, you can also connect the adapter at the terminals right there to have it in your (e.g.) livingroom..

Littledodger, 2021-10-16:
My module is connected to the front controler of my heating system, it's a deported command.

Yes, I looked about HA a few time ago, I have to make some research about it to associate my domotic system with.
I have some hours of work to understand and code a plugin for that... Once done, I'll look at the integration of BSB-LAN...

Littledodger, 2021-10-17:
Hi,
For the moment, I'm just looking for a sollution, I have some other tings to do before dive into scripting... But I think it'll gonna come...