fredlcore/BSB-LAN #413: [BUG] BSB-Lan not starting up on Arduino Due with Austria Email TWPK

Magublafix, 2021-11-07:
**BSB-LAN Version**
#define MAJOR "2"
#define MINOR "0"
#define PATCH "107"
#define COMPILETIME "20210917003308"

**Architecture**
Arduino Due

**Bus system**
BSB

**Describe the bug**
Problem1: BSB-Lan ist not starting up at all and is failing with error "s & 0x80) == 0" failed: file "../source/twi.c", line 279, function: TWI_StartWrite"

Problem2: Startup time until error seems to be very long because the software runs through several BME280 sensor addresses despite I do not use any.

**To Reproduce**
Steps to reproduce the behavior:
1. Download latest master
2. configure ip adresses. disable dhcp
3. run software on connected arduino
4. See error

**Logs**
[20211107_bsblannotstarting2.txt](https://github.com/fredlcore/BSB-LAN/files/7492266/20211107_bsblannotstarting2.txt)

**Expected behavior**
BSB-LAN ist starting up correctly after a few seconds and is visible within the network with the configured ip adresses.

fredlcore, 2021-11-07:
Especially the last line looks really strange to me:
```
s & 0x80) == 0" failed: file "../source/twi.c", line 279, function: TWI_StartWrite
```
@dukess, do you have any idea where this could come from?

fredlcore, 2021-11-07:
I found this solution here:
https://forum.arduino.cc/t/programm-error-assertion-address-0x80-0-failed/133060/3
The address can't be higher than 127. @dukess, is that due to the new OneWire library you found or is this an error on BSB-LAN's part?

dukess, 2021-11-07:
How many BME sensors set in option?

dukess, 2021-11-07:
May it helps: https://github.com/fredlcore/BSB-LAN/pull/414

dukess, 2021-11-07:
I can guess this happened if BME_Sensors == 0xFF in EEPROM or BME_Sensors was more than 80 in settings.

Magublafix, 2021-11-07:
@dukess i tried your fix and it seemed to fix the error, but i still have a problem
Now it tries to get an ip over dhcp despite i have dhcp disabled in the settings:

bool useDHCP = false;                 // Set to false if you want to use a fixed IP.
byte ip_addr[4] = {192,168,1,88};     // Please note the commas instead of dots!!!  Set useDHCP to true if you want DHCP.
byte gateway_addr[4] = {192,168,1,1}; // Gateway address. This is usually your router's IP address. Please note the commas instead of dots!!! Ignored if first value is 0.
byte dns_addr[4] = {192,168,1,1};     // DNS server. Please note the commas instead of dots!!! Ignored if first value is 0.
byte subnet_addr[4] = {255,255,255,0};

As soon as i remove the bsb lan module it will boot correctly without dhcp and the configured ip adress.
New log file:
[20211107_bsblannotstarting3.txt](https://github.com/fredlcore/BSB-LAN/files/7492733/20211107_bsblannotstarting3.txt)

dukess, 2021-11-07:
Thank you for log!
Looks like all settings reading (with some wrong values) from EEPROM.
What we can to do:
1. Erase settings from EEPROM. For this way shorting two pins (31 and 33) before powering. 
OR
2. Set UseEEPROM = 0 in config file. Do not forget set UseEEPROM = 1 in config file after config changing in webConfig page.

P.S. It is your first run BSB-LAN on Due?

Magublafix, 2021-11-07:
That did the trick!
I actually though that the Due has no EEPROM but ok ;)
Actually i already ran an old BSB-LAN Version on it so probably that was the problem ;)

thank you!
Gerald

1coderookie, 2021-11-08:
That's correct, the Due doesn't have an EEPROM itself, but the BSB-LAN adapter for the Due does have one :)
 https://1coderookie.github.io/BSB-LPB-LAN/kap06.html#612-eeprom-l%C3%B6schung-mittels-pinkontakten

Magublafix, 2021-11-13:
Aaah ok that explains it 😉 thank you for the quick help guys! Awesome project!

closed 2021-11-13 by Magublafix