fredlcore/BSB-LAN #417: [FEATURE REQUEST] Improved English translation for 8000-8011 parameters

GSzabados, 2021-11-12:
**Is your feature request related to a problem? Please describe.**
When I look at the status of the heater, I can see only German descriptions of the current states. It is hard to understand quickly what the heater actually does, just after some googling for a translation.

**Describe the solution you'd like**
English translation of the states.

**Describe alternatives you've considered**
Every time use Google Translate, copy paste text, etc.

**Describe your own contribution**
Compiled already a PR from the Brötje IRS-Plus System Handbuch (German) and the LMS14 manual (English).
It includes all the possible states for the 8000-8011 heating parameters what are included in the LMS14 manual. 
(Parameter 8006 unfortunately not part of that.)

**Additional context**

[image]

1coderookie, 2021-11-12:
You need to check a heat pump controller manual (RVS61) for 8004 and 8006, I'll send you one right now.. 

Btw: please don't use G.T. for this, only use the 'official' terms from the manuals. If you're using G.T. for personal usage (to understand certain things better), it's fine of course - in that case I'd recommend www.deepl.com/translator though, it often seems to be more accurate.. ;)

dukess, 2021-11-13:
English translation is looks like abandoned now so if you come as translator then it would be cool!

GSzabados, 2021-11-13:
@dukess, yes, I am trying. I have a few manuals German and English, so I can match the phrases, but it is a tedious job.

If you want have a look on the PR and test it. It should have most of the states translated to English, but not all, the ones for the 8004 and 8006 parameters.

1coderookie, 2021-11-14:
8004&8006 you should find in the english heat pump manual I sent you, that is the manual for the RVS61 type controllers. Or do you miss/need anything else?

fredlcore, 2022-10-12:
This will be no longer relevant with the upcoming version 3.0 as translations will be taken from the heater via the "complete dump". But thanks a lot for your efforts!