fredlcore/BSB-LAN #419: [BUG] Incorrect text assignments

GSzabados, 2021-11-12:
**BSB-LAN Version**
Any

**Architecture**
Any

**Bus system**
BSB

**Describe the bug**
As mentioned in #415

The following assignments seems to be incorrect:
```
#define ENUM8008_16_TEXT ENUM8005_16_TEXT
#define ENUM8008_17_TEXT ENUM8000_16_TEXT
#define ENUM8008_18_TEXT ENUM8000_17_TEXT
#define ENUM8004_95_TEXT ENUM8000_16_TEXT
```
```
#define ENUM8005_16_TEXT "Minimalbegrenzung aktiv"
#define ENUM8000_16_TEXT "Anlagenfrostschutz aktiv"
#define ENUM8000_17_TEXT "Frostschutz aktiv"
```

16: Minimalbegrenzung aktiv (22)
17: Anlagefrostschutz aktiv (23)
18: Frostschutz aktiv (24)
95: Schutzbetrieb Kühlen (149)

**To Reproduce**

**Logs**

**Expected behavior**
Description to match IRS-PLUS System Handbuch 

Parameter 8000_17 and 8008_17 are expected to be _Anlagefrostschutz aktiv_
Parameter 8000_18 and 8008_18 are expected to be _Frostschutz aktiv_
Parameter 8004_95 is expected to be _Schutzbetrieb Kühlen_

**Screenshots**
[image]

**Desktop (if applicable, please complete the following information):**

**Smartphone (if applicable, please complete the following information):**

**Additional context**
Add any other context about the problem here.

1coderookie, 2021-11-12:
Thanks, I just wrote an answer to that in #415 - you're right, seems like a mistake here.

fredlcore, 2021-12-26:
I'm sorry, I have not followed up whose turn is it now in this issue as well as in #415. Do we have validated translations now that I could apply or not?

1coderookie, 2021-12-28:
I dunno about the translations, but at that time there was definitely an error within BSB-LAN for ENUM8000 & ENUM8008: https://github.com/fredlcore/BSB-LAN/pull/415#issuecomment-966928012
so if you didn't already fix that one, at least that one is still 'open'.

fredlcore, 2021-12-28:
Sorry that I have to ask again, but this is what we have currently:
```
8000/20 = not yet defined
8000/21 = not yet defined
8000/22 = Anlagenfrostschutz aktiv (plant frost protection active)
8000/23 = Frostschutz aktiv (frost protection)
8000/24 = Aus (off)
8000/25 = not yet defined

8008/20 = Minimalbegrenzung (minimal limitation)
8008/21 = Minimalbegrenzung, Teillast (minimal limitation, partial load)
8008/22 = Minimalbegrenzung aktiv (minimal limitation active)
8008/23 = Anlagenfrostschutz aktiv (plant frost protection active)
8008/24 = Frostschutz aktiv (frost protection)
8008/25 = Aus (off)
```

I'm fine with any changes that you suggest and which are somehow based on one of the manuals, but it makes me wonder how we got there in the first place. Maybe this was just an error, but maybe this is one of the parameters with different encodings?
If we can ascertain that the options between 8000 and 8008 (and maybe others) here are always the same, I can just reference back to one of them and potential errors won't happen then. But if there is a likelihood that there is a difference between 8000 and 8008 and possibly others, we would need to sort this out first.

Thanks!

1coderookie, 2021-12-29:
As you can see in the screenshot within the above linked answer and the discussion in #415, different manuals clearly state that 
23 = Anlagenfrostschutz aktiv
24 = Frostschutz aktiv
for both 8000 and 8008. 
The error seems to be a typo within this definement:
`#define ENUM8008_17_TEXT ENUM8000_16_TEXT`
`#define ENUM8008_18_TEXT ENUM8000_17_TEXT`
because it should be 
`#define ENUM8008_17_TEXT ENUM8000_17_TEXT`
`#define ENUM8008_18_TEXT ENUM8000_18_TEXT`
I explained it here https://github.com/fredlcore/BSB-LAN/pull/415#issuecomment-966915421 and there I also mentioned that I didn't find a 22 anywhere, only the mentioned 23&24.
So: based on my findings, up to me that's an error/typo.

fredlcore, 2021-12-29:
Yes, I can see and have read that, but I want to make sure that we can safely generalize from these manuals. Because there are not just the two mentioned differences which could be a typo, but basically all the option entries listed above differ from each other. To me, that looks more than just a typo, because that someone who has come up with it in the past (don't know who and don't know when) must have made quite a few errors and misreadings to come up with this. 
I don't mind deviating from the previous definitions and take what you found as the basis, but then it's more than just correcting a typo because we would also have to consolidate (or delete) the other option values between 20 and 25 (and probably more which I haven't checked yet).

1coderookie, 2021-12-29:
Yes, there aren't only the two mentioned ones, I saw that, and I think we should somehow check that (btw: these _17=_16 and so on are also existent with e.g. 8000&8003/8004/8005 (8000 is Status HK1, 8003 Status TWW, 8004 Status Kühlkrei 1, 8005 Status Kessel)), which is a bit weird up to me. 
I will check different manuals and my different controllers once again when I'll find the time (and maybe you can also check them at your LMU7x) and let you know my findings. 
Until then, just leave it ike it is right now.

fredlcore, 2021-12-29:
Yes, it is a bit weird, also because I would think that the Status for a heating circuit would have somewhat different status values compared to a cooling circuit or the boiler itself. These long option lists often tempt to just re-use those from other parameters if one spots a few similarities which could be misleading. On the other hand, it could well be that there is a long, long list of possible options which includes all different types of circuits (cooling, heating etc.) but only a subset of these are used/supported for each type. Then it wouldn't be a problem to just list all of them because these are read-only values anyway and the heater would simply not display a status that it doesn't know about (but it would still be visible in the drop-down list).

The other option would be to go back and just display the numbers only. That would be sad for those where the current list matches (more or less), but less misleading for those where the list doesn't fit. I don't know...

1coderookie, 2021-12-29:
I could imagine that there is something like that list with all possible options where every option has its own number, at least it seemed like this when I studied the different types of manuals. But we'll figure it out somehow I guess.. ;) 
It would be interesting to know though how they manage this when newer controller versions come with additional options - that's why I asked you to maybe also have a look at your LMU7, cuz the LMS definitely came with a few new options I'd say.. 

@GSzabados : Didn't you made a list with all the numbers and options you found? I think I read something like that from you somewhere..

@freetz: The list you posted here https://github.com/fredlcore/BSB-LAN/issues/419#issuecomment-1002313526 - where can I find it, displayed like this? Within the defs.h I only found the definements and stuff for the ENUMS, but I didn't find a listing like the one you posted there yet.

fredlcore, 2021-12-29:
Yes, I'll have a look at it of course, it interests me too :)...
As for the list: It's not really there in plain text; you have to go into _defs.h and look for `const char ENUM8000` for example and then look up each entry such as `ENUM8000_00_TEXT` first in _defs.h again whether it is referenced to a previous entry with the same text (in this case: ENUM6040_03_TEXT). And then use that one (or if it is not referenced the original entry) and look it up in LANG_DE.h or whatever language you are using.
That's why I didn't look up more entries for comparison ;)...

1coderookie, 2021-12-29:
Ah ok, so I was right that I didn't find it in that way how you wrote it down and did it right when I looked it up by myself :) 
(..kinda complicated though lol.. ;) )

fredlcore, 2021-12-29:
Indeed it is, but the price we have to pay for quite some memory savings (well, in relation to the memory available on such a small device ;) )...

fredlcore, 2022-10-12:
This will be no longer relevant with the upcoming version 3.0 as translations will be taken from the heater via the "complete dump". But thanks a lot for your efforts!

closed 2022-10-12 by fredlcore