fredlcore/BSB-LAN #420: ESP32 SDK 2.0.1. Just FYI.

dukess, 2021-11-13:
I'm starting software adaptations for using with new ESP32 SDK. 
At the moment, the program is being built without errors.
To do this, i had to use a new version of the OneWire library.
Now i'm not assured in possibility of working with Dallas sensors, it should be tested.

dukess, 2021-11-14:
@fredlcore Hi, can you remember for me why you used  
`EEPROMClass EEPROM_ESP("nvs", EEPROM_SIZE);`
(EEPROM name definition)
instead 
`EEPROMClass EEPROM_ESP(EEPROM_SIZE);`

Here was glitches?

I got error with name definition when i run test on PlatformIO but haven't problems with Arduino Studio:
```
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FS> 1.0
|-- <SD_MMC> 1.0
|   |-- <FS> 1.0
|-- <EEPROM> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <WiFi> 1.0
|-- <Ethernet> 2.0.0
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <Update> 1.0
|-- <WebServer> 1.0
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <Wire> 1.0.1
Building in release mode
Compiling .pio/build/esp32-evb/src/BSB_LAN.ino.cpp.o
Compiling .pio/build/esp32-evb/src/src/ArduinoMDNS/MDNS.cpp.o
/tmp/tmpw9cod4iu/src/BSB_LAN.ino:510:49: error: invalid conversion from 'const char*' to 'uint32_t {aka unsigned int}' [-fpermissive]
 EEPROMClass EEPROM_ESP((const char *)PSTR("nvs"));
                                                 ^
In file included from /tmp/tmpw9cod4iu/src/BSB_LAN.ino:502:0:
/home/runner/.platformio/packages/framework-arduinoespressif32/libraries/EEPROM/src/EEPROM.h:37:5: note:   initializing argument 1 of 'EEPROMClass::EEPROMClass(uint32_t)'
     EEPROMClass(uint32_t sector);
     ^
*** [.pio/build/esp32-evb/src/BSB_LAN.ino.cpp.o] Error 1
========================== [FAILED] Took 2.04 seconds ==========================
Error: Process completed with exit code 1.
```

fredlcore, 2021-11-14:
Yes, we had a discussion earlier this year about it:
https://github.com/fredlcore/BSB-LAN/issues/329#issuecomment-773272612
By default the IDEs come with different partition layouts for the ESP32, and nvs was the one that seemed to work on all of them. Maybe you could check and see if/why the command no longer lets you select a partition but only accept the size?

fredlcore, 2021-11-14:
I just figured out that the size parameter is no longer supported/used. So 
```
EEPROMClass EEPROM_ESP("nvs");
```
should be sufficient. The problem will be that older IDEs may still not support the new library (which is just 3 weeks old), so changing the code may produce problems there. Since we officially recommend the Arduino IDE, I would only make the change once the newest Arduino IDE also supports this new library version. When that will be, God nows...

dukess, 2021-11-15:
> #329 (comment)

Thank you!

> By default the IDEs come with different partition layouts for the ESP32, and nvs was the one that seemed to work on all of them. Maybe you could check and see if/why the command no longer lets you select a partition but only accept the size?

Compiler for some reason mix unsigned int (32 bit length) and address pointer (32 bit length) I guess.

> The problem will be that older IDEs may still not support the new library

Arduino IDE has no problem with ESP32 SDK 2.0.1.
Got it! I was sure that v 2.0.0 was released by platformio but that not so. And I don't know platformio well enough to manually change frameworks via the command line. 

By the way current DHT library can work unstable/incorrect for same reason as old OneWire library (multicore MCU + interrupts). So we should to change it to https://github.com/beegee-tokyo/DHTesp

Ludn, 2021-12-23:
Sorry for posting this here, but related issues seems to be closed:
Thanks for help and hint on this: i spent the night on getting BSB-LAN v2.0 compiled with Arduino 1.8.15 for Wi-Fi ESP32.
I tried mods noted in [#issue 322](https://github.com/fredlcore/BSB-LAN/issues/322), this mod on EEPROMClass here, different ESP32 SDK versions 1.0.6 to 2.0.1 and several mods of-manual like:
- adding #define ESP32
- deactivating compiler problematic #defines on ONE_WIRE_BUS, DHT_BUS, BME280
..but it's still not compiler-accepted.

My basic mods (followed by manual) are
- removing both directories "ArduinoMDNS" und "WiFiSpi"
- in BSB_LAN_config.h: #define WIFI, with settings

Compiler results somehow close to [#issue 322](https://github.com/fredlcore/BSB-LAN/issues/322) are:
```
D:\Haus\Heizgsplanung, Speicher, Gastherme\BSB-LAN-ESP32\BSB-LAN-master\BSB_LAN\src\OneWire\OneWire.cpp:599:22: warning: extra tokens at end of #undef directive
 #  undef noInterrupts() {portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;portENTER_CRITICAL(&mux)
                      ^
D:\Haus\Heizgsplanung, Speicher, Gastherme\BSB-LAN-ESP32\BSB-LAN-master\BSB_LAN\src\OneWire\OneWire.cpp:600:20: warning: extra tokens at end of #undef directive
 #  undef interrupts() portEXIT_CRITICAL(&mux);}
                    ^
D:\Haus\Heizgsplanung, Speicher, Gastherme\BSB-LAN-ESP32\BSB-LAN-master\BSB_LAN\src\BSB\bsb.cpp: In member function 'void BSB::enableInterface()':
D:\Haus\Heizgsplanung, Speicher, Gastherme\BSB-LAN-ESP32\BSB-LAN-master\BSB_LAN\src\BSB\bsb.cpp:52:34: error: 'UART_RXFIFO_FULL_INT_ENA_M' was not declared in this scope
     uart_intr.intr_enable_mask = UART_RXFIFO_FULL_INT_ENA_M
                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
D:\Haus\Heizgsplanung, Speicher, Gastherme\BSB-LAN-ESP32\BSB-LAN-master\BSB_LAN\src\BSB\bsb.cpp:52:34: note: suggested alternative: 'GPIO_PIN4_INT_ENA_M'
     uart_intr.intr_enable_mask = UART_RXFIFO_FULL_INT_ENA_M
                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                  GPIO_PIN4_INT_ENA_M
D:\Haus\Heizgsplanung, Speicher, Gastherme\BSB-LAN-ESP32\BSB-LAN-master\BSB_LAN\src\BSB\bsb.cpp:53:31: error: 'UART_RXFIFO_TOUT_INT_ENA_M' was not declared in this scope
                             | UART_RXFIFO_TOUT_INT_ENA_M
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
D:\Haus\Heizgsplanung, Speicher, Gastherme\BSB-LAN-ESP32\BSB-LAN-master\BSB_LAN\src\BSB\bsb.cpp:53:31: note: suggested alternative: 'GPIO_PIN4_INT_ENA_M'
                             | UART_RXFIFO_TOUT_INT_ENA_M
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
                               GPIO_PIN4_INT_ENA_M
D:\Haus\Heizgsplanung, Speicher, Gastherme\BSB-LAN-ESP32\BSB-LAN-master\BSB_LAN\src\BSB\bsb.cpp:54:31: error: 'UART_FRM_ERR_INT_ENA_M' was not declared in this scope
                             | UART_FRM_ERR_INT_ENA_M;
                               ^~~~~~~~~~~~~~~~~~~~~~
In file included from D:\Haus\Heizgsplanung, Speicher, Gastherme\BSB-LAN-ESP32\BSB-LAN-master\BSB_LAN\src\OneWire\OneWire.cpp:149:
D:\Haus\Heizgsplanung, Speicher, Gastherme\BSB-LAN-ESP32\BSB-LAN-master\BSB_LAN\src\OneWire\util\OneWire_direct_gpio.h: In function 'void directModeInput(uint32_t)':
D:\Haus\Heizgsplanung, Speicher, Gastherme\BSB-LAN-ESP32\BSB-LAN-master\BSB_LAN\src\OneWire\util\OneWire_direct_gpio.h:161:26: error: 'rtc_gpio_desc' was not declared in this scope
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
D:\Haus\Heizgsplanung, Speicher, Gastherme\BSB-LAN-ESP32\BSB-LAN-master\BSB_LAN\src\OneWire\util\OneWire_direct_gpio.h:161:26: note: suggested alternative: 'rtc_io_desc'
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
                          rtc_io_desc
D:\Haus\Heizgsplanung, Speicher, Gastherme\BSB-LAN-ESP32\BSB-LAN-master\BSB_LAN\src\OneWire\util\OneWire_direct_gpio.h: In function 'void directModeOutput(uint32_t)':
D:\Haus\Heizgsplanung, Speicher, Gastherme\BSB-LAN-ESP32\BSB-LAN-master\BSB_LAN\src\OneWire\util\OneWire_direct_gpio.h:189:26: error: 'rtc_gpio_desc' was not declared in this scope
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
D:\Haus\Heizgsplanung, Speicher, Gastherme\BSB-LAN-ESP32\BSB-LAN-master\BSB_LAN\src\BSB\bsb.cpp:54:31: note: suggested alternative: 'GPIO_PIN26_INT_ENA_M'
                             | UART_FRM_ERR_INT_ENA_M;
                               ^~~~~~~~~~~~~~~~~~~~~~
                               GPIO_PIN26_INT_ENA_M
D:\Haus\Heizgsplanung, Speicher, Gastherme\BSB-LAN-ESP32\BSB-LAN-master\BSB_LAN\src\OneWire\util\OneWire_direct_gpio.h:189:26: note: suggested alternative: 'rtc_io_desc'
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
                          rtc_io_desc
Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt: C:\Users\AG-Ry\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\libraries\WiFi
 Nicht benutzt: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Fehler beim Kompilieren für das Board OLIMEX ESP32-EVB.
```
Has anyone v2 with Wifi for ESP32 running and could give me a hint?

dukess, 2021-12-24:
In SDK 2.0.x some definitions, functions and header files was changed. I added one more header file in bsb.cpp:
`#include "soc/uart_reg.h"`

About OneWire: you can take library from my branch https://github.com/dukess/BSB-LAN/tree/ESP32-SDK-2.0.1 because official library not contatin yet needed changes OR just delete OneWire directory.

fredlcore, 2021-12-24:
Is it confirmed that the official Arduino IDE now uses SDK 2.0? I'm just wondering because otherwise many more users should be complaining about that problem. Or maybe they are all using older versions still? In any case, if Arduino IDE is using SDK 2.0, we'll have to make these adjustments so that it compiles without errors.

dukess, 2021-12-24:
Yep. Arduino Studio for Windows can use ESP 2.0.1

Ludn, 2021-12-24:
Sorry for misunderstanding, i meant ESP32 esspressif version in board administrator. Just can't get it compiled but errors are related to this. Also don't need one wire, DHT or BME support for now.

fredlcore, 2021-12-24:
I just installed the Arduino IDE from scratch on my Mac, version 1.8.19. The board manager for ESP32 only offers me SDK version 1.0.6 as the newest version. I used ESP32 Dev Module as target board.
First thing I did was rename BSB-LAN_config.h.default to BSB-LAN_config.h, remove ArduinoDNS and WiFiSpi folders, then activate `#define WIFI` in config.
Result: Compiler runs through without any errors.
@Ludn: Can you please update your Arduino IDE to 1.8.19 and do exactly as I did? If it doesn't compile properly, you may have to uninstall and reinstall the Arduino IDE and/or the ESP32 board from the board manager.

Ludn, 2021-12-24:
Big thx, let me check on latest Ardu. Interesting, i get newer versions offered here with windows setup, but can choose older version if needed.

fredlcore, 2021-12-24:
I'm using this URL for the ESP32 board manager:
`https://dl.espressif.com/dl/package_esp32_index.json`
Which one do you use that you are being offered the 2.0.2 version?

Ludn, 2021-12-24:
2.0.1

fredlcore, 2021-12-24:
It shouldn't be OS related, I guess, because the JSON file that I'm referring to only lists version up to 1.0.6. Which URL are you using?

Ludn, 2021-12-24:
more precise: [https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json](https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json)

dukess, 2021-12-24:
[image]

@Ludn you can choose version 1.0.6 in board manager

fredlcore, 2021-12-24:
Ok, that's different from what I know, but I see that it's listed in the official docs. Strange that there is a difference between the two, as both seem to be from Espressif...
For now, we should add that BSB-LAN runs on 1.0.6. I would start requiring 2.x once it is supported also on PlatformIO because that's the platform I use to code and it seems that currently I can't simply switch to 2.x there...

Ludn, 2021-12-24:
just checked 1.06 (with Ardu 1.8.15) and 1st error occurs:
```
BSB_LAN:511:23: error: redefinition of 'EEPROMClass EEPROM_ESP'
 EEPROMClass EEPROM_ESP((const char *)PSTR("nvs"));
```

fredlcore, 2021-12-24:
@dukess: I installed 2.0.2 now, but adding `#include "soc/uart_reg.h"` does not stop the above errors...

Ludn, 2021-12-24:
double-checked and compiled fine with 1.0.6.: i had quoted 2 line not deactivated, iam happy for now^^ thx so much

dukess, 2021-12-24:
> just checked 1.0.6. (with Ardu 1.8.15) and 1st error occurs:
> 
> ```
> BSB_LAN:511:23: error: redefinition of 'EEPROMClass EEPROM_ESP'
>  EEPROMClass EEPROM_ESP((const char *)PSTR("nvs"));
> ```

look like you use my branch with 1.0.6

> I installed 2.0.2 now, but adding `#include "soc/uart_reg.h"` does not stop the above errors...

@fredlcore can you show me log?

fredlcore, 2021-12-24:
Sure:
```
/Users/frederik/Downloads/BSB-LAN/BSB_LAN/src/OneWire/OneWire.cpp:599:22: warning: extra tokens at end of #undef directive
 #  undef noInterrupts() {portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;portENTER_CRITICAL(&mux)
                      ^
/Users/frederik/Downloads/BSB-LAN/BSB_LAN/src/OneWire/OneWire.cpp:600:20: warning: extra tokens at end of #undef directive
 #  undef interrupts() portEXIT_CRITICAL(&mux);}
                    ^
/Users/frederik/Downloads/BSB-LAN/BSB_LAN/src/BSB/bsb.cpp: In member function 'void BSB::enableInterface()':
/Users/frederik/Downloads/BSB-LAN/BSB_LAN/src/BSB/bsb.cpp:52:34: error: 'UART_RXFIFO_FULL_INT_ENA_M' was not declared in this scope
     uart_intr.intr_enable_mask = UART_RXFIFO_FULL_INT_ENA_M
                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/frederik/Downloads/BSB-LAN/BSB_LAN/src/OneWire/OneWire.cpp:149:
/Users/frederik/Downloads/BSB-LAN/BSB_LAN/src/OneWire/util/OneWire_direct_gpio.h: In function 'void directModeInput(uint32_t)':
/Users/frederik/Downloads/BSB-LAN/BSB_LAN/src/OneWire/util/OneWire_direct_gpio.h:161:26: error: 'rtc_gpio_desc' was not declared in this scope
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
/Users/frederik/Downloads/BSB-LAN/BSB_LAN/src/OneWire/util/OneWire_direct_gpio.h:161:26: note: suggested alternative: 'rtc_io_desc'
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
                          rtc_io_desc
/Users/frederik/Downloads/BSB-LAN/BSB_LAN/src/OneWire/util/OneWire_direct_gpio.h: In function 'void directModeOutput(uint32_t)':
/Users/frederik/Downloads/BSB-LAN/BSB_LAN/src/OneWire/util/OneWire_direct_gpio.h:189:26: error: 'rtc_gpio_desc' was not declared in this scope
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
/Users/frederik/Downloads/BSB-LAN/BSB_LAN/src/OneWire/util/OneWire_direct_gpio.h:189:26: note: suggested alternative: 'rtc_io_desc'
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
                          rtc_io_desc
/Users/frederik/Downloads/BSB-LAN/BSB_LAN/src/BSB/bsb.cpp:52:34: note: suggested alternative: 'GPIO_PIN4_INT_ENA_M'
     uart_intr.intr_enable_mask = UART_RXFIFO_FULL_INT_ENA_M
                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                  GPIO_PIN4_INT_ENA_M
/Users/frederik/Downloads/BSB-LAN/BSB_LAN/src/BSB/bsb.cpp:53:31: error: 'UART_RXFIFO_TOUT_INT_ENA_M' was not declared in this scope
                             | UART_RXFIFO_TOUT_INT_ENA_M
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/frederik/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/xtensa-esp32-elf-g++ -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4-beta1-189-ga79dc75f0a\"" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/config -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/newlib/platform_include -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/freertos/include -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/freertos/include/esp_additions/freertos -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/freertos/port/xtensa/include -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/freertos/include/esp_additions -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/esp_hw_support/include -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/esp_hw_support/include/soc -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/esp_hw_support/include/soc/esp32 -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/esp_hw_support/port/esp32 -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/heap/include -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/log/include -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/lwip/include/apps -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/lwip/include/apps/sntp -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/lwip/lwip/src/include -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/lwip/port/esp32/include -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/lwip/port/esp32/include/arch -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/soc/include -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/soc/esp32 -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/soc/esp32/include -I/Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/include/hal/esp32/include -I/Users/f [... cut ...]
                               ^~~~~~~~~~~~~~~~~~~~~~
/Users/frederik/Downloads/BSB-LAN/BSB_LAN/src/BSB/bsb.cpp:54:31: note: suggested alternative: 'GPIO_PIN26_INT_ENA_M'
                             | UART_FRM_ERR_INT_ENA_M;
                               ^~~~~~~~~~~~~~~~~~~~~~
                               GPIO_PIN26_INT_ENA_M
Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt: /Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/libraries/WiFi
 Nicht benutzt: /Applications/Arduino.app/Contents/Java/libraries/WiFi
```

fredlcore, 2021-12-24:
Include is added in line 501:
```
#if defined(ESP32)
#include "soc/uart_reg.h"
```

dukess, 2021-12-24:
@fredlcore These changes allow to comlipe project with SDK 2.0.2: https://github.com/dukess/BSB-LAN/commit/0941e33c458e5e0243e56b08d9f22214c45f064c

Ludn, 2021-12-24:
That's good news. Can't actually find where in project (ESP32) is defined. And is there any other advantage of SDK 2.0.2. appart from being updated to latest?

dukess, 2021-12-24:
ESP32 will be defined automatically. And no, this SDK has no advantages excepting more stable work with SD card and Ethernet adapter on OLIMEX ESP32 EVB board.

dukess, 2021-12-24:
So if you use WiFi you have no reason for instant moving to SDK 2.0.x.

fredlcore, 2021-12-24:
@dukess: Cool, thanks! Will try it out and if it works, we could make a note in the manual that we've moved to 2.0 for ESP32 and refer to the official board manager URL and make 2.0.x a requirement. That will prevent us from adding new functions that may only run on 1.0.6 which we would then have to adjust again later.

1coderookie, 2021-12-24:
> we could make a note in the manual that we've moved to 2.0 for ESP32 and refer to the official board manager URL and make 2.0.x a requirement. 

Yes, please let me know when and what to add ;)

Merry Xmas btw!

fredlcore, 2021-12-24:
Yes, merry Christmas to everyone - and thanks for all the support for this project!
@dukess: I just tried to install 2.0 for PlatformIO, but the command from your repo
` pio platform install https://github.com/olicooper/arduino-v2-esp32-platformio`
gives me 
`Error: Could not find one of 'platform.json' manifest files in the package`
I think I can get it to work with manual edits in the platformio.ini file, but not sure if the CI task will cope with the above line...

fredlcore, 2021-12-24:
I got SDK 2.0 installed and applied your changes, but these errors still prevail:
```
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:510:24: error: invalid conversion from 'const char*' to 'uint32_t' {aka 'unsigned int'} [-fpermissive]
 EEPROMClass EEPROM_ESP((const char *)PSTR("nvs"));
In file included from /Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:502:
/Users/frederik/.platformio/packages/framework-arduinoespressif32/libraries/EEPROM/src/EEPROM.h:37:26: note:   initializing argument 1 of 'EEPROMClass::EEPROMClass(uint32_t)'
     EEPROMClass(uint32_t sector);
                 ~~~~~~~~~^~~~~~
Compiling .pio/build/esp32-2.0/src/src/SdFat/MinimumSerial.cpp.o
In file included from /Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:820:
src/BSB_LAN/BSB_LAN_EEPROMconfig.h:168:121: error: 'enable_ota_update' was not declared in this scope
   {CF_OTA_UPDATE,       6, CCAT_GENERAL,  CPI_SWITCH,    CDT_BYTE,           OPT_FL_ADVANCED, CF_OTA_UPDATE_TXT, sizeof(enable_ota_update)}, //immediately apply
                                                                                                                         ^~~~~~~~~~~~~~~~~
src/BSB_LAN/BSB_LAN_EEPROMconfig.h:168:121: note: suggested alternative: 'enable_ipwe'
   {CF_OTA_UPDATE,       6, CCAT_GENERAL,  CPI_SWITCH,    CDT_BYTE,           OPT_FL_ADVANCED, CF_OTA_UPDATE_TXT, sizeof(enable_ota_update)}, //immediately apply
                                                                                                                         ^~~~~~~~~~~~~~~~~
                                                                                                                         enable_ipwe
src/BSB_LAN/BSB_LAN_EEPROMconfig.h:194:124: error: 'mDNS_hostname' was not declared in this scope
   {CF_MDNS_HOSTNAME,    6, CCAT_IPV4,     CPI_TEXT,      CDT_STRING,         OPT_FL_ADVANCED, CF_MDNS_HOSTNAME_TXT, sizeof(mDNS_hostname)},//need reboot
                                                                                                                            ^~~~~~~~~~~~~
src/BSB_LAN/BSB_LAN_EEPROMconfig.h:194:124: note: suggested alternative: 'gethostname'
   {CF_MDNS_HOSTNAME,    6, CCAT_IPV4,     CPI_TEXT,      CDT_STRING,         OPT_FL_ADVANCED, CF_MDNS_HOSTNAME_TXT, sizeof(mDNS_hostname)},//need reboot
                                                                                                                            ^~~~~~~~~~~~~
                                                                                                                            gethostname
src/BSB_LAN/BSB_LAN_EEPROMconfig.h:213:114: error: 'BME_Sensors' was not declared in this scope
   {CF_BMEBUS,           6, CCAT_BMEBUS,   CPI_TEXT,      CDT_BYTE,           OPT_FL_ADVANCED, CF_NUM_TXT, sizeof(BME_Sensors)}, //need reboot
                                                                                                                  ^~~~~~~~~~~
src/BSB_LAN/BSB_LAN_EEPROMconfig.h:213:114: note: suggested alternative: 'numSensors'
   {CF_BMEBUS,           6, CCAT_BMEBUS,   CPI_TEXT,      CDT_BYTE,           OPT_FL_ADVANCED, CF_NUM_TXT, sizeof(BME_Sensors)}, //need reboot
                                                                                                                  ^~~~~~~~~~~
                                                                                                                  numSensors
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino: In function 'void init_ota_update()':
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:1964:6: error: 'enable_ota_update' was not declared in this scope
   if(enable_ota_update) {
      ^~~~~~~~~~~~~~~~~
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:1964:6: note: suggested alternative: 'init_ota_update'
   if(enable_ota_update) {
      ^~~~~~~~~~~~~~~~~
      init_ota_update
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino: In function 'bool SaveConfigFromRAMtoEEPROM()':
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:2484:14: error: 'BME_Sensors' was not declared in this scope
         case CF_ONEWIREBUS:
              ^~~~~~~~~~~
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:2484:14: note: suggested alternative: 'numSensors'
         case CF_ONEWIREBUS:
              ^~~~~~~~~~~
              numSensors
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:2489:15: error: 'enable_ota_update' was not declared in this scope
           needReboot = true;
               ^~~~~~~~~~~~~~   
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:2489:15: note: suggested alternative: 'init_ota_update'
           needReboot = true;
               ^~~~~~~~~~~~~~   
               init_ota_update
Compiling .pio/build/esp32-2.0/src/src/SdFat/SdCard/SdSpiCard.cpp.o
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino: In function 'void loop()':
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:6833:5: error: 'enable_ota_update' was not declared in this scope
  if(enable_ota_update)
     ^~~~~~~~~~~~~~~~~
Compiling .pio/build/esp32-2.0/src/src/SdFat/SdCard/SdSpiCardEX.cpp.o
[... cut ...]
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:6987:52: note: suggested alternative: 'gethostname'
   registerConfigVariable(CF_MDNS_HOSTNAME, (byte *)mDNS_hostname);
                                                    ^~~~~~~~~~~~~
                                                    gethostname
Compiling .pio/build/esp32-2.0/src/src/SdFat/SdCard/SdioCardEX.cpp.o
Compiling .pio/build/esp32-2.0/src/src/SdFat/SdCard/SdioTeensy.cpp.o
*** [.pio/build/esp32-2.0/src/BSB_LAN.ino.cpp.o] Error 1
```

Do you have any idea what to do here, @dukess?

fredlcore, 2021-12-25:
Sorry, I didn't install 2.0.2, so the EEPROM errors are going away, but others still stick:
```
In file included from /Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:820:
src/BSB_LAN/BSB_LAN_EEPROMconfig.h:168:121: error: 'enable_ota_update' was not declared in this scope
   {CF_OTA_UPDATE,       6, CCAT_GENERAL,  CPI_SWITCH,    CDT_BYTE,           OPT_FL_ADVANCED, CF_OTA_UPDATE_TXT, sizeof(enable_ota_update)}, //immediately apply
                                                                                                                         ^~~~~~~~~~~~~~~~~
src/BSB_LAN/BSB_LAN_EEPROMconfig.h:168:121: note: suggested alternative: 'enable_ipwe'
   {CF_OTA_UPDATE,       6, CCAT_GENERAL,  CPI_SWITCH,    CDT_BYTE,           OPT_FL_ADVANCED, CF_OTA_UPDATE_TXT, sizeof(enable_ota_update)}, //immediately apply
                                                                                                                         ^~~~~~~~~~~~~~~~~
                                                                                                                         enable_ipwe
src/BSB_LAN/BSB_LAN_EEPROMconfig.h:194:124: error: 'mDNS_hostname' was not declared in this scope
   {CF_MDNS_HOSTNAME,    6, CCAT_IPV4,     CPI_TEXT,      CDT_STRING,         OPT_FL_ADVANCED, CF_MDNS_HOSTNAME_TXT, sizeof(mDNS_hostname)},//need reboot
                                                                                                                            ^~~~~~~~~~~~~
src/BSB_LAN/BSB_LAN_EEPROMconfig.h:194:124: note: suggested alternative: 'gethostname'
   {CF_MDNS_HOSTNAME,    6, CCAT_IPV4,     CPI_TEXT,      CDT_STRING,         OPT_FL_ADVANCED, CF_MDNS_HOSTNAME_TXT, sizeof(mDNS_hostname)},//need reboot
                                                                                                                            ^~~~~~~~~~~~~
                                                                                                                            gethostname
src/BSB_LAN/BSB_LAN_EEPROMconfig.h:213:114: error: 'BME_Sensors' was not declared in this scope
   {CF_BMEBUS,           6, CCAT_BMEBUS,   CPI_TEXT,      CDT_BYTE,           OPT_FL_ADVANCED, CF_NUM_TXT, sizeof(BME_Sensors)}, //need reboot
                                                                                                                  ^~~~~~~~~~~
src/BSB_LAN/BSB_LAN_EEPROMconfig.h:213:114: note: suggested alternative: 'numSensors'
   {CF_BMEBUS,           6, CCAT_BMEBUS,   CPI_TEXT,      CDT_BYTE,           OPT_FL_ADVANCED, CF_NUM_TXT, sizeof(BME_Sensors)}, //need reboot
                                                                                                                  ^~~~~~~~~~~
                                                                                                                  numSensors
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino: In function 'void init_ota_update()':
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:1964:6: error: 'enable_ota_update' was not declared in this scope
   if(enable_ota_update) {
      ^~~~~~~~~~~~~~~~~
Compiling .pio/build/esp32-2.0/src/src/SdFat/SdCard/SdSpiCard.cpp.o
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:1964:6: note: suggested alternative: 'init_ota_update'
   if(enable_ota_update) {
      ^~~~~~~~~~~~~~~~~
      init_ota_update
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino: In function 'bool SaveConfigFromRAMtoEEPROM()':
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:2484:14: error: 'BME_Sensors' was not declared in this scope
         case CF_ONEWIREBUS:
              ^~~~~~~~~~~
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:2484:14: note: suggested alternative: 'numSensors'
         case CF_ONEWIREBUS:
              ^~~~~~~~~~~
              numSensors
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:2489:15: error: 'enable_ota_update' was not declared in this scope
           needReboot = true;
               ^~~~~~~~~~~~~~   
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:2489:15: note: suggested alternative: 'init_ota_update'
           needReboot = true;
               ^~~~~~~~~~~~~~   
               init_ota_update
Compiling .pio/build/esp32-2.0/src/src/SdFat/SdCard/SdSpiCardEX.cpp.o
Compiling .pio/build/esp32-2.0/src/src/SdFat/SdCard/SdioCardEX.cpp.o
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino: In function 'void loop()':
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:6833:5: error: 'enable_ota_update' was not declared in this scope
  if(enable_ota_update)
     ^~~~~~~~~~~~~~~~~
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:6833:5: note: suggested alternative: 'init_ota_update'
  if(enable_ota_update)
     ^~~~~~~~~~~~~~~~~
     init_ota_update
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino: In function 'void setup()':
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:6985:46: error: 'BME_Sensors' was not declared in this scope
   registerConfigVariable(CF_BMEBUS, (byte *)&BME_Sensors);
                                              ^~~~~~~~~~~
[... cut ...]
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:6987:52: error: 'mDNS_hostname' was not declared in this scope
   registerConfigVariable(CF_MDNS_HOSTNAME, (byte *)mDNS_hostname);
                                                    ^~~~~~~~~~~~~
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:6987:52: note: suggested alternative: 'gethostname'
   registerConfigVariable(CF_MDNS_HOSTNAME, (byte *)mDNS_hostname);
                                                    ^~~~~~~~~~~~~
                                                    gethostname
```

fredlcore, 2021-12-25:
Sorry, my bad, old configuration file...

fredlcore, 2021-12-25:
With this platformio.ini section I got it working with @dukess' changes:
```
[env:esp32-nodemcu]
platform = espressif32
board = nodemcu-32s
framework = arduino
upload_speed = 921600
src_filter = +<*> -<src/ArduinoMDNS/> -<src/WiFiSpi/>
build_flags = -DNO_GLOBAL_EEPROM
#board_build.partitions = custom_partitions.csv
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
```
Thanks!

fredlcore, 2021-12-25:
As mentioned above, the CI tests now fail with platformio-ESP32.yaml:
```
 Platform Manager: Installing git+https://github.com/olicooper/arduino-v2-esp32-platformio
git version 2.34.1
Cloning into '/home/runner/.platformio/.cache/tmp/pkg-installing-dslc7v3z'...
Error: Could not find one of 'platform.json' manifest files in the package
Error: Process completed with exit code 1.
```
I haven't found out how to translate the above platformio.ini line:
`platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git`
into a corresponding `pio` command. Any ideas anyone?

fredlcore, 2021-12-25:
Ok, found the solution for the CI problem, but compiling for the Olimex gives me this error now:
```
In file included from /home/runner/.platformio/packages/framework-arduinoespressif32/libraries/SD_MMC/src/SD_MMC.cpp:24:0:
/home/runner/.platformio/packages/framework-arduinoespressif32/libraries/SD_MMC/src/SD_MMC.cpp: In member function 'bool fs::SDMMCFS::begin(const char*, bool, bool, int)':
/home/runner/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/driver/include/driver/sdmmc_host.h:119:1: sorry, unimplemented: non-trivial designated initializers not supported
 }
 ^
/home/runner/.platformio/packages/framework-arduinoespressif32/libraries/SD_MMC/src/SD_MMC.cpp:45:39: note: in expansion of macro 'SDMMC_SLOT_CONFIG_DEFAULT'
     sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
                                       ^
/home/runner/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/driver/include/driver/sdmmc_host.h:119:1: sorry, unimplemented: non-trivial designated initializers not supported
 }
 ^
/home/runner/.platformio/packages/framework-arduinoespressif32/libraries/SD_MMC/src/SD_MMC.cpp:45:39: note: in expansion of macro 'SDMMC_SLOT_CONFIG_DEFAULT'
     sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
                                       ^
*** [.pio/build/esp32-evb/libc3a/SD_MMC/SD_MMC.cpp.o] Error 1
```

fredlcore, 2021-12-25:
It can be added to `platformio ci` with the `--project-option` command line option:
```
run: platformio ci --project-option = "platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream" --project-option="platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git" --board=esp32-evb ./BSB_LAN
```

fredlcore, 2021-12-25:
@dukess, can we close this issue then?

fredlcore, 2021-12-25:
BTW: Both @1coderookie and me can't test on Windows, could you check out the current master repo and try to compile it under standard Arduino IDE with 2.0.2? Then we should be all set!

fredlcore, 2021-12-25:
A few things I've noted so far:
- BMx280 detection takes quite a long time (20 seconds or so) until setup procedure continues - at least if no such sensor is present. That was not the case under 1.0.6.

- Collision detection is no longer working the way it did. Before, every byte that was sent to the bus was received back at the same time. By comparing what was sent to what was received, we could test if another device was sending on the bus as well: If what we sent was not the same as what we received immediately after, there was probably a different device sending. Now, we don't seem to receive what has been sent. After sending, only one byte is received (0xFF). The next bytes are already the answer to our request. I don't know why we don't receive back what we sent out, but it seems to work because the queries work fine. Still, it would be nice to have the collision check working somehow.

fredlcore, 2021-12-25:
This is the code that needs to be excluded for the ESP32:
```
#if !defined(ESP32)
    serial->flush();
    unsigned long timeout = millis();
    while ((millis()-timeout < 50) && serial->available() == 0) {
      delay(1);
    }
#endif
```
I forgot why we disabled the `flush()` for ESP32 a further bit up in the same function, and it still does not explain why the sent data does not show up again when receiving, but at least it's working for now...

dukess, 2021-12-25:
@1coderookie Merry Christmas! :)
@fredlcore when i tried run BSB-LAN with SDK 2.0.1 it running normally (may collision detection did'nt working).

fredlcore, 2021-12-25:
@dukess: Did you test 2.0.1 with the Olimex? Because collision detection completely breaks waiting for response telegram after sending a telegram. For Arduino IDE, SDK 2.0.2 is released (you don't have to do all this stuff that I need to do for PlatformIO, but that's fine because Arduino IDE is our reference for end users).
Do you have the chance to test the current master branch on a Windows Arduino IDE to see if it works?

dukess, 2021-12-26:
I will try tomorrow to test current version.

dukess, 2021-12-27:
I assemble current build with Arduino Studio 1.8.19 with SDK 2.0.2.  OS: Windows
It works. Startup delay  ~10 seconds (Scan for 1 BM*280 sensor and initial BLE sensors scan ~5 sec)

```
Version: 2.0.111-20211225144617
Found device 8a488b38c1a4
Found service '181a' data len: 19, 12161a188a488b38c1a4ab07e10bab0b57560f
Temp: 19.63°, Humidity: 30.41%, Vbatt: 2.000, Battery: 87%, flg: 0x0f, cout: 86
Found device e6001c38c1a4
Found service '181a' data len: 19, 12161a18e6001c38c1a41f08800b980b550805
Temp: 20.79°, Humidity: 29.44%, Vbatt: 2.000, Battery: 85%, flg: 0x05, cout: 8
Found device d9496b38c1a4
Found service '181a' data len: 19, 12161a18d9496b38c1a4dd07970bd90b5cc30f
Temp: 20.13°, Humidity: 29.67%, Vbatt: 3.000, Battery: 92%, flg: 0x0f, cout: 195
GET /LKJBHksdgfrleneaslkrwklh/JQ=8700,8740,8770,8830,9005,8310,8311,8314,8316 HTTP/1.1
URL: /JQ=8700,8740,8770,8830,9005,8310,8311,8314,8316
Found device 8a488b38c1a4
Found service '181a' data len: 19, 12161a188a488b38c1a4a907e10bab0b57580f
Temp: 19.61°, Humidity: 30.41%, Vbatt: 2.000, Battery: 87%, flg: 0x0f, cout: 88
LAN->HEIZ QUR 8700 ENUM_CAT_2e_TEXT - STR8700_TEXT:
DC C2 00 0B 06 3D 05 05 21 B9 7C
HEIZ->LAN ANS 8700 ENUM_CAT_2e_TEXT - STR8700_TEXT: -10.3 UNIT_DEG_TEXT
DC 80 42 0E 07 05 3D 05 21 00 FD 6D 65 CB
#8700: -10.3 UNIT_DEG_TEXT
LAN->HEIZ QUR 8740 ENUM_CAT_2e_TEXT - STR8740_TEXT:
DC C2 00 0B 06 3D 2D 05 1E 51 A7
HEIZ->LAN ANS 8740 ENUM_CAT_2e_TEXT - STR8740_TEXT: 22.8 UNIT_DEG_TEXT
DC 80 42 0E 07 2D 3D 05 1E 00 05 B5 B0 8A
#8740: 22.8 UNIT_DEG_TEXT
LAN->HEIZ QUR 8770 ENUM_CAT_2e_TEXT - STR8770_TEXT:
DC C2 00 0B 06 3D 2E 05 1E 08 F7
HEIZ->LAN ANS 8770 ENUM_CAT_2e_TEXT - STR8770_TEXT: 20.8 UNIT_DEG_TEXT
DC 80 42 0E 07 2E 3D 05 1E 00 05 32 89 67
#8770: 20.8 UNIT_DEG_TEXT
LAN->HEIZ QUR 8830 ENUM_CAT_2e_TEXT - STR8830_TEXT:
DC C2 00 0B 06 3D 31 05 2F 41 D7
HEIZ->LAN ANS 8830 ENUM_CAT_2e_TEXT - STR8830_TEXT: 47.2 UNIT_DEG_TEXT
DC 80 42 0E 07 31 3D 05 2F 00 0B CB 33 91
#8830: 47.2 UNIT_DEG_TEXT
LAN->HEIZ QUR 9005 ENUM_CAT_2e_TEXT - STR9005_TEXT:
DC C2 00 0B 06 3D 05 08 05 AB C6
HEIZ->LAN ANS 9005 ENUM_CAT_2e_TEXT - STR9005_TEXT: 1.5 UNIT_BAR_TEXT
DC 80 42 0E 07 05 3D 08 05 00 00 0F 90 77
#9005: 1.5 UNIT_BAR_TEXT
LAN->HEIZ QUR 8310 ENUM_CAT_2d_TEXT - STR8310_TEXT:
DC C2 00 0B 06 3D 0D 05 19 A7 86
HEIZ->LAN ANS 8310 ENUM_CAT_2d_TEXT - STR8310_TEXT: 55.0 UNIT_DEG_TEXT
DC 80 42 0E 07 0D 3D 05 19 00 0D BF A7 B2
#8310: 55.0 UNIT_DEG_TEXT
LAN->HEIZ QUR 8311 ENUM_CAT_2d_TEXT - STR8311_TEXT:
DC C2 00 0B 06 3D 0D 09 23 75 F2
HEIZ->LAN ANS 8311 ENUM_CAT_2d_TEXT - STR8311_TEXT: 55.0 UNIT_DEG_TEXT
DC 80 42 0E 07 0D 3D 09 23 00 0D C0 E7 A3
#8311: 55.0 UNIT_DEG_TEXT
LAN->HEIZ QUR 8314 ENUM_CAT_2d_TEXT - STR8314_TEXT:
DC C2 00 0B 06 3D 11 05 1A A1 E7
HEIZ->LAN ANS 8314 ENUM_CAT_2d_TEXT - STR8314_TEXT: 50.1 UNIT_DEG_TEXT
DC 80 42 0E 07 11 3D 05 1A 00 0C 84 ED 77
#8314: 50.1 UNIT_DEG_TEXT
LAN->HEIZ QUR 8316 ENUM_CAT_2d_TEXT - STR8316_TEXT:
DC C2 00 0B 06 3D 05 05 1D 4E A3
HEIZ->LAN ANS 8316 ENUM_CAT_2d_TEXT - STR8316_TEXT: 53.0 UNIT_DEG_TEXT
DC 80 42 0E 07 05 3D 05 1D 00 0D 40 E0 1E
#8316: 53.0 UNIT_DEG_TEXT
Found device d9496b38c1a4
Found service '181a' data len: 19, 12161a18d9496b38c1a4da07960bd80b5cc50f
Temp: 20.10°, Humidity: 29.66%, Vbatt: 3.000, Battery: 92%, flg: 0x0f, cout: 197
BLE scan tack
BLE scan tick
HEIZ->ALL  INF      05000213 04 00 00 5B
DC 80 7F 0F 02 05 00 02 13 04 00 00 5B F5 FC
INF: Brennerstatus: 4
Found device 8a488b38c1a4
Found service '181a' data len: 19, 12161a188a488b38c1a4a907e10bab0b57580f
Temp: 19.61°, Humidity: 30.41%, Vbatt: 2.000, Battery: 87%, flg: 0x0f, cout: 88
RGT1->HEIZ INF      3D2D0215 05 B7 00
DC 86 00 0E 02 3D 2D 02 15 05 B7 00 95 85
HEIZ->ALL  INF 10104 ENUM_CAT_30_TEXT - STR10104_TEXT: 00 00 FF FF FF FF FF FF 00 00
DC 80 7F 15 02 2F 00 02 11 00 00 FF FF FF FF FF FF 00 00 83 90
Found device d9496b38c1a4
```

fredlcore, 2021-12-27:
Hmm, maybe it's faster on your side because the scan does find some sensors? I don't have any sensors installed and the delay at setup is approx. 30 seconds...

fredlcore, 2021-12-27:
As for collision: The current version has collision check disabled for ESP32. If you enable the following section, I would think that you would have the same problems that I mentioned earlier, so please test if possible:
```
#if !defined(ESP32)
    serial->flush();
    unsigned long timeout = millis();
    while ((millis()-timeout < 50) && serial->available() == 0) {
      delay(1);
    }
#endif
```

fredlcore, 2021-12-27:
@dukess: I've had two reports of users with an ESP32 that have encountered the following problem of a fired watchdog without a reboot despite the logs saying so:
```
Address EEPROM option 64: 1505
Size of cmdtbl1: 33984
Size of cmdtbl2: 45888
Size of cmdtbl3: 27120
free RAM: 283164
Init One Wire bus...
numSensors: 0
PPS settings:
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40090734  PS      : 0x00060d33  A0      : 0x8009244c  A1      : 0x3ffb2360  
A2      : 0xcd000000  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060d23  
A6      : 0x00060d20  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0xffffffff  
A10     : 0x3f43fcaa  A11     : 0x000001f4  A12     : 0x3ffb2478  A13     : 0x00000001  
A14     : 0x8d800000  A15     : 0x003fffff  SAR     : 0x0000001a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xcd000000  LBEG    : 0x4008acc0  LEND    : 0x4008acdc  LCOUNT  : 0x00000000  

Backtrace:0x40090731:0x3ffb23600x40092449:0x3ffb23a0 0x400926c5:0x3ffb23c0 0x4008428a:0x3ffb23e0 0x40084306:0x3ffb2410 0x400fae50:0x3ffb2460 0x400faebf:0x3ffb2480 0x400faf29:0x3ffb24a0 0x401013e1:0x3ffb24c0 0x4010148e:0x3ffb24e0 0x4010150a:0x3ffb2500 0x400fadbf:0x3ffb2520 0x400f9ddb:0x3ffb2540 0x400f476c:0x3ffb2560 0x400e4c30:0x3ffb25a0 0x400e4d8e:0x3ffb25e0 0x400e4f3e:0x3ffb2760 0x400d9be9:0x3ffb2780 0x400e80b6:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
```

and 

```
Start network services
query failed
query failed
query failed
6225
query failed
query failed
query failed
6226
Device family: 0
Device variant: 0
 
Setup complete
 
Timeout
GET /K HTTP/1.1
 
0 - Date/Time
1 - Room Unit
2 - Wireless
3 - Time switch program 1
4 - Time switch program 2
5 - Time switch program 3
6 - Time switch program 4/DHW
7 - Time switch program 5
8 - Holiday programs HC1
9 - Holiday programs HC2
10 - Holiday programs HC3
11 - Heating circuit 1
12 - Cooling circuit 1
13 - Heating circuit 2
14 - Cooling circuit 2
15 - Heating circuit 3/P
16 - DHW
17 - Consumer circuit 1
18 - Consumer circuit 2
19 - Swimming pool circuit
20 - Hx pump
21 - Swimming pool
22 - Primary controller/system pump
23 - Primary controller/system pump
24 - Boiler
25 - Sitherm Pro
26 - Heat pump
27 - Energy counter
28 - Cascade
29 - Extra heat source
30 - Solar
31 - Solid fuel boiler
32 - Buffer tank
33 - DHW storage tank
34 - DHW direct charging
35 - General Features
36 - Configuration
37 - LPB
38 - Modbus
39 - Error
40 - Service/special operation
41 - Configuration extension modules
42 - Input/output tests
43 - Status
44 - Diagnosis cascade
45 - Diagnosis producer
46 - Diagnosis consumer
47 - Burner control
48 - User defined
49 - PPS bus
50 - One Wire, DHT & MAX! Sensors
Guru Meditation Error: Core  panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC       0x40155693  PS       0x00060b4  A0       0x80155745  A1       0x3fb5f 
A2       0x3f4882c A3       0x3f48838  A4       0x3f66010  A5       0x00000001 
A6       0x02000000  A7       0x00000000  A8       0x80155693  A9       0x3fb5d 
A10      0x3fffc A11      0x3fa348  A12      0x00000003  A13      0x00000001 
A14      0x00060523  A15      0x00000000  SAR      0x0000001e EXCCAUSE 0x00000005 
EXCVADDR 0x00000000  LBEG     0x400834d  LEND     0x40083509  LCOUNT   0x00000000 
 
ELF file SHA256: 000000000000000
 
Backtrace:0x40155693:0x3fb5f 0x40155742:0x3fb610 0x401555c:0x3fb630 0x4015284d0x3fb650 0x40154fe0x3fb670 0x40155068:0x3fb690 0x40159d9:0x3fb6c 0x40159e9:0x3fb6f 0x400f431:0x3fb720 0x4011a7c0x3fb740 0x4011ac:0x3fb760 0x4011ea:0x3fb7a 0x4011e1d0x3fb7c 0x4011e01:0x3fb7e 0x4008a3a0x3fb810
 
ore 1 register dump:
PC       0x4000c71  PS       0x00060034  A0       0x8008b54  A1       0x3fbb0 
A2       0x3fb058  A3       0x3f51748  A4       0x00000014  A5       0x3fc390 
A6       0x3fc3d  A7       0x00000001  A8       0x00000001  A9       0x3f51749 
A10      0x000000a  A11      0x3fb060  A12      0x8008c0b A13      0x3fc360 
A14      0x00000008  A15      0x00000001  SAR      0x00000012  EXCCAUSE 0x00000005 
EXCVADDR 0x00000000  LBEG     0x4000c6c LEND     0x4000c77  LCOUNT   0x00000000 
 
ELF file SHA256: 000000000000000
 
Backtrace:0x4000c71:0x3fbb00x4008b51:0x3fbb00x4008d24:0x3fbb00x4008dd:0x00000001
 
Rebooting...
```

Do you have an idea what this is coming from? Apparently, these are standard installations...

dukess, 2021-12-28:
(StoreProhibited) this error appears when we tried to set new value to 
variable with "const" flag. They need to check config file.

dukess, 2021-12-28:
Yes, it is possible. I can check it later.

dukess, 2021-12-28:
It would be better if users can decode backtrace log with this tool https://github.com/me-no-dev/EspExceptionDecoder

fredlcore, 2021-12-28:
@MrAntonka: Could it be that you used an "older" _config.h version?

fredlcore, 2021-12-28:
The second report has been solved, that was a solder joint that I accidentially had made. The first one my @MrAntonka is still under investigation.

MrAntonka, 2021-12-28:
HI

this is what I have so far

`
Reading EEPROM
Reading done.
EEPROM schema v.9 Program schema v.9
Address EEPROM option 0: 0
Address EEPROM option 1: 1
Address EEPROM option 2: 2
Address EEPROM option 3: 6
Address EEPROM option 4: 226
Address EEPROM option 5: 306
Address EEPROM option 6: 490
Address EEPROM option 7: 491
Address EEPROM option 8: 492
Address EEPROM option 9: 493
Address EEPROM option 10: 494
Address EEPROM option 11: 495
Address EEPROM option 12: 499
Address EEPROM option 13: 500
Address EEPROM option 14: 660
Address EEPROM option 15: 661
Address EEPROM option 16: 665
Address EEPROM option 17: 825
Address EEPROM option 18: 831
Address EEPROM option 19: 832
Address EEPROM option 20: 836
Address EEPROM option 21: 840
Address EEPROM option 22: 844
Address EEPROM option 23: 848
Address EEPROM option 24: 850
Address EEPROM option 25: 854
Address EEPROM option 26: 858
Address EEPROM option 27: 922
Address EEPROM option 28: 986
Address EEPROM option 29: 987
Address EEPROM option 30: 988
Address EEPROM option 31: 998
Address EEPROM option 32: 999
Address EEPROM option 33: 1159
Address EEPROM option 34: 1160
Address EEPROM option 35: 1164
Address EEPROM option 36: 1165
Address EEPROM option 37: 1166
Address EEPROM option 38: 1167
Address EEPROM option 39: 1171
Address EEPROM option 40: 1203
Address EEPROM option 41: 1235
Address EEPROM option 42: 1267
Address EEPROM option 43: 1299
Address EEPROM option 44: 1301
Address EEPROM option 45: 1302
Address EEPROM option 46: 1303
Address EEPROM option 47: 1304
Address EEPROM option 48: 1336
Address EEPROM option 49: 1400
Address EEPROM option 50: 1420
Address EEPROM option 51: 1440
Address EEPROM option 52: 1460
Address EEPROM option 53: 1461
Address EEPROM option 54: 1462
Address EEPROM option 55: 1463
Address EEPROM option 56: 1464
Address EEPROM option 57: 1465
Address EEPROM option 58: 1466
Address EEPROM option 59: 1498
Address EEPROM option 60: 1499
Address EEPROM option 61: 1500
[... 28 lines cut ...]
`

`PC: 0x40090e20: xPortEnterCriticalTimeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/compare_set.h line 25
EXCVADDR: 0xdd040000

Decoding stack results
0x40090e1d: xPortEnterCriticalTimeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/compare_set.h line 25
0x40092c59: multi_heap_internal_lock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h line 578
0x40092ed5: multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c line 229
0x40084862: heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 145
0x400848de: heap_caps_malloc_prefer at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 226
0x401002b8: mem_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/mem.c line 237
0x40100327: do_memp_malloc_pool at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c line 254
0x40100391: memp_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c line 350
0x40106849: sys_timeout_abs at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/timeouts.c line 198
0x401068f6: sys_timeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/timeouts.c line 323
0x40106972: sys_timeouts_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/timeouts.c line 281
0x40100227: lwip_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/init.c line 378
0x400ff243: tcpip_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c line 610
0x400f9b6c: esp_netif_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c line 320
0x400e69f0: tcpipInit() at /home/r/.arduino15/packages/esp32/hardware/esp32/2.0.2/libraries/WiFi/src/WiFiGeneric.cpp line 530
0x400e6b4e: wifiLowLevelInit(bool) at /home/r/.arduino15/packages/esp32/hardware/esp32/2.0.2/libraries/WiFi/src/WiFiGeneric.cpp line 562
0x400e6cfe: WiFiGenericClass::mode(wifi_mode_t) at /home/r/.arduino15/packages/esp32/hardware/esp32/2.0.2/libraries/WiFi/src/WiFiGeneric.cpp line 1002
0x400df1f5: setup() at /home/r/Downloads/BSB-LAN-master/BSB_LAN/BSB_LAN.ino line 7359
0x400eb9ce: loopTask(void*) at /home/r/.arduino15/packages/esp32/hardware/esp32/2.0.2/cores/esp32/main.cpp line 42`

fredlcore, 2021-12-28:
Hm, this seems to be WiFi-related, I have added this link due to the problems discussed here:
https://github.com/espressif/arduino-esp32/issues/2501#issuecomment-731618196

Does the problem persist if you remove these two lines?
```
  WiFi.disconnect(true);  //disconnect form wifi to set new wifi connection
  WiFi.mode(WIFI_STA); //init wifi mode
```

MrAntonka, 2021-12-28:
when i comment these two lines I got another error.

PC: 0x40090e20: xPortEnterCriticalTimeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/compare_set.h line 25
EXCVADDR: 0xcd040000

Decoding stack results
0x40090e1d: xPortEnterCriticalTimeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/compare_set.h line 25
0x40092c59: multi_heap_internal_lock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h line 578
0x40092ed5: multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c line 229
0x40084862: heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 145
0x400848de: heap_caps_malloc_prefer at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 226
0x401002a4: mem_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/mem.c line 237
0x40100313: do_memp_malloc_pool at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c line 254
0x4010037d: memp_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c line 350
0x40106835: sys_timeout_abs at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/timeouts.c line 198
0x401068e2: sys_timeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/timeouts.c line 323
0x4010695e: sys_timeouts_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/timeouts.c line 281
0x40100213: lwip_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/init.c line 378
0x400ff22f: tcpip_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c line 610
0x400f9b58: esp_netif_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c line 320
0x400e69dc: tcpipInit() at /home/r/.arduino15/packages/esp32/hardware/esp32/2.0.2/libraries/WiFi/src/WiFiGeneric.cpp line 530
0x400e6b3a: wifiLowLevelInit(bool) at /home/r/.arduino15/packages/esp32/hardware/esp32/2.0.2/libraries/WiFi/src/WiFiGeneric.cpp line 562
0x400e6cea: WiFiGenericClass::mode(wifi_mode_t) at /home/r/.arduino15/packages/esp32/hardware/esp32/2.0.2/libraries/WiFi/src/WiFiGeneric.cpp line 1002
0x400e6df0: WiFiGenericClass::enableSTA(bool) at /home/r/.arduino15/packages/esp32/hardware/esp32/2.0.2/libraries/WiFi/src/WiFiGeneric.cpp line 1074
0x400e74d1: WiFiSTAClass::begin() at /home/r/.arduino15/packages/esp32/hardware/esp32/2.0.2/libraries/WiFi/src/WiFiSTA.cpp line 243
0x400df1f2: setup() at /home/r/Downloads/BSB-LAN-master/BSB_LAN/BSB_LAN.ino line 7362
0x400eb9ba: loopTask(void*) at /home/r/.arduino15/packages/esp32/hardware/esp32/2.0.2/cores/esp32/main.cpp line 42

dukess, 2021-12-28:
I can propose to disable onewire, dht and bme sensors. These libraries 
contains time critical sections which can interfere with wifi. Or not.

MrAntonka, 2021-12-28:
@dukess do you mean I have to try comment following lines?

#define ONE_WIRE_BUS
#define DHT_BUS
#define BME280

?

fredlcore, 2021-12-28:
@dukess: @MrAntonka suggested in an e-mail to me that it could have to do with PS-RAM because it seems to work fine on WROOM boards, but not on WROVER boards. So I guess these libraries are probably not the cause of the error. The stack traces also point to WiFi routines. But of course, you could try to disable the libraries first before we go any further, @MrAntonka.

MrAntonka, 2021-12-28:
commentd three lines mentioned in my last post
I got this
Starting SD..Setting up WiFi interfaceGuru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40090088  PS      : 0x00060333  A0      : 0x80091da0  A1      : 0x3ffb2260  
A2      : 0xff040000  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060323  
A6      : 0x00060320  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0xffffffff  
A10     : 0x3f441bca  A11     : 0x000001f4  A12     : 0x3ffb2378  A13     : 0x00000001  
A14     : 0xbf840000  A15     : 0x003fffff  SAR     : 0x0000000d  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xff040000  LBEG    : 0x4008a550  LEND    : 0x4008a56c  LCOUNT  : 0x00000000  

Backtrace:0x40090085:0x3ffb22600x40091d9d:0x3ffb22a0 0x40092019:0x3ffb22c0 0x40083b1a:0x3ffb22e0 0x40083b96:0x3ffb2310 0x400fc6b4:0x3ffb2360 0x400fc723:0x3ffb2380 0x400fc78d:0x3ffb23a0 0x40102c45:0x3ffb23c0 0x40102cf2:0x3ffb23e0 0x40102d6e:0x3ffb2400 0x400fc623:0x3ffb2420 0x400fb63f:0x3ffb2440 0x400f5f78:0x3ffb2460 0x400e5178:0x3ffb24a0 0x400e52d6:0x3ffb24e0 0x400e5486:0x3ffb2660 0x400e558c:0x3ffb2680 0x400e5c6d:0x3ffb26a0 0x400db256:0x3ffb2760 0x400e9486:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...

[46 more replies]

closed 2021-12-28 by fredlcore