fredlcore/BSB-LAN #423: I did Bluetooth Low Energy sensors support for BSB-LAN

dukess, 2021-12-05:
Limitations: 
Will work on ESP32 only.
It consumes  most flash space (OTA will not work).
Experimental and very unstable.

https://github.com/dukess/BSB-LAN/tree/BLE-sensors/BSB_LAN

dukess, 2021-12-07:
@fredlcore , @1coderookie  I want to hear your opinion

fredlcore, 2021-12-07:
Sorry, it slipped away from my list...
For me, 210kb library is a no go. If we want to move to precompiled binaries, we have to offer OTA. For me, the external sensors are not a main priority. We started with a few that were easy to implement, and while Bluetooth would generally be nice because it works without blocking pins, the cost in terms of memory are too high in my opinion.

1coderookie, 2021-12-08:
Well, the idea itself is good though! :) The only problem is -once again- the memory.. :(
I think, for people who don't use a home automation software, it could be nice to have that one included, cuz these sensors aren't too expensive. But then it could also be nice to have that same opportunity with other hardware, like thermostats or whatsoever. So at the end I also think that it's a bit too much of a 'niche product' to have it permanently integrated in BSB-LAN due to the consumption of memory. 
But now that you already made it: If you put this solution in a GitHub repo, I could mention it in the manual and put a link to the repo - just in case there are users out there, who use an ESP32 based setup and who are using this sensors also..?!

dukess, 2021-12-08:
Thanks for the feedback!
Yes, I agree, this functionality will be for geeks.
As for connecting other devices: I think it will not be too difficult to expand the list of supported equipment, especially if there is already a protocol description for it. May be I very optimistic. 
I will ask Frederick to include my code in the main branch. At the moment, it is done this way: if user do not enable the BLE_SENSORS option, then there is no difference at all.

Small description:
MCU in background permanently scans for BLE Advertising telegrams which contains temperature, humidity and other values. 
In config two options added: 
"EnableBLE" - enable or disable sensors scan.
BLE_sensors_macs - list of MAC addresses of BLE sensors. Devices not listed in BLE_sensors_macs will be ignored. Order 
in the list affects the order in the category. At the moment can be added up to 40 sensors. (Programs NN20900-20199)
List of all BT devices around can be displayed through /CO program (If "EnableBLE" is on)

fredlcore, 2021-12-08:
Ok, I understand - is there a standard for these sensors or would that mean that people will be coming with all kind of BLE sensors they want to see implemented and we have to add protocol after protocol for them?
The other thing is the range: Usually, BSB-LAN sits next to the heater which is usually in the basement or under the roof. So a sensor from the living room might not get through to the ESP32.

But what I'm wondering about mostly is the use case: Mostly, these sensors will be imported into a home automation system. For wired sensors, that's not so easy, because you just can't connect a DS18B20 or BLE280 sensor to a PC or wherever your system is running on. But for BLE devices, these can often be connected directly to the server running the home automation system. That was one of the reasons I didn't pursue the implementation of the Shelly devices anymore, because they integrate easy into home automation systems via MQTT etc.

So I really don't see a "killer argument" why we should include it - but if we do it, it will lead to questions and requests that at least I won't be able to handle. So that's why I'm still reluctant to include it...

dukess, 2021-12-08:
I could suggest adding code on the AS IS principle, but your arguments are convincing.

My situation is simpler: BSB-LAN is located almost in the center of the house and there is no full automation, so there is a subconscious desire to cram more functionality into the device.

For my installation, I see such a use case: I will place sensors in rooms to obtain an average temperature to turn on heating circuits 2 and 3. Currently, sensors located in the halls show a temperature higher than in corner rooms.

>that people will be coming with all kind of BLE sensors they want to see implemented and we have to add protocol after protocol for them?

If BLE sensors can send data in broadcast telegrams then no problem to add new device: just 2-10 lines of code. But if device required connection and encryption - it can take many time to implement.

Let's put this question aside for a while. But, if anything, know that BLE support is there.
Now my main task is not to confuse the branches when making changes. :)

fredlcore, 2021-12-08:
In any case: thanks a lot for your effort, and it's good to know that the code is "out there".
I also generally like the idea to use BSB-LAN as a standalone device without the need of an additional home automation system, especially when it comes to sending an average room temperature. But since we already have a whole MQTT stack, maybe it's easier to implement the Shelly devices which can send the temperature and humidity via MQTT?

The other thing I was thinking about for a while (because I already use it ;) ) is reading temperatures from 433 / 868 MHz sensors. MAX is also using this spectrum, but decoding temperatures will vary from one device to another again, so we have the same problem again.

If memory wouldn't be so tight, I would be more open to that, but with several hundred(!) BSB commands still waiting to be implemented, I'm being a bit more reluctant than at other times ;)...

dukess, 2021-12-08:
>maybe it's easier to implement the Shelly devices which can send the temperature and humidity via MQTT?

>The other thing I was thinking about for a while (because I already use it ;) ) is reading temperatures from 433 / 868 MHz sensors. MAX is also using this spectrum, but decoding temperatures will vary from one device to another again, so we have the same problem again.

It is good idea, but, as for all community projects, new functions can be added by peoples with possibilities only. I mean "equipment" in our case.

> but with several hundred(!) BSB commands still waiting to be implemented

Wow! 8-0

fredlcore, 2021-12-25:
As I'm preparing for the 2.0 release of BSB-LAN, I'll close this for now, but feel free to re-open at any time.

closed 2021-12-25 by fredlcore

dukess, 2021-12-27:
I adapted it for SDK 2.0.2

1coderookie, 2022-01-23:
@dukess I just added this function to the manual. Could you please read it and check, if everything is ok like this? 
https://1coderookie.github.io/BSB-LPB-LAN_EN/chap07.html#714-xiaomi-mijia-ble-sensors-lywsd03mmc  
While using your description from above of how to active and configure it, a question arised though. This is the way I wrote it: "A listing of all BT devices found can be displayed through the URL command `/CO` (if "EnableBLE" is activated)." -> does that mean, that /CO lists ALL sensors which were found, even if the MAC isn't listed? Or does it only list the sensors which are listed and also found (which would make more sense because of /CO I guess)?

@dukess @fredlcore For any questions belonging to this BLE function that may arise on the user side and for support, I'd open a thread in the discussions here and link to that instead of let users open issues at dukess' repo. By this, we have all the info and support here at one place. _Would that be ok for you guys?_

fredlcore, 2022-01-23:
Just FYI: this only works on @dukess fork, not with the "official" repo, and probably never will as long as it means that we would have to give opp OTA for that, as I mentioned in the corresponding issue...

1coderookie, 2022-01-23:
..did you read that chapter..? ;)

1coderookie, 2022-01-23:
..but would it be ok for you if I open a thread here in discussions for linking to it, just in case questions/problems arise? @fredlcore

fredlcore, 2022-01-23:
Yes, I read the chapter, my main point is that it will probably never be included into the main repo, so I want to avoid people getting their hopes up or expect all new features from the main repo being included in this repo etc.
Linking to a thread or so would be ok, I guess.

1coderookie, 2022-01-23:
Sure. I made the note about that even more prominent now and put it right at the beginning of that subchapter. But this solution is just too interesting and useful for not being mentioned though.
The belonging discussion thread I opened for it is this one: https://github.com/fredlcore/BSB-LAN/discussions/440

dukess, 2022-01-23:
@1coderookie all looks fine!

> it will probably never be

@fredlcore who knows? :)

The main obstacle that I now see on the way to supporting encrypted data exchange is that I can't think of a beautiful way to set keys for Bluetooth devices.

fredlcore, 2022-01-23:
That's why I said "probably" ;).
Don't get me wrong, I definitely like this feature, but if it's only possible by dropping OTA, my priorities are clear. If at some point we'll have more memory available, then we can add it immediately, but so far it doesn't look like it to happen in the near future...

1coderookie, 2022-01-24:
Just FYI: I just changed the place of that subchapter and put it in the subchapter "Own Hardwaresolutions" (I put it in the "Sensors" chapter before, yesterday). So now already because of the place where it's listed/mentioned it's absolutely clear at first sight that it's an 'own hardwaresolution' / 'unofficial' solution.

@fredlcore If any questions arise within the FHEM forum or in private emails from userside about this topic (even though everything is clearly stated ;) ) we just refer to the mentioned discussion thread..

fredlcore, 2022-01-24:
Yes, that's a much more suitable location in the manual (for now), thanks!