fredlcore/BSB-LAN #428: String won't get populated over MQTT for parameter 8730

olympia, 2021-12-27:
**BSB-LAN Version**
#define MAJOR "2"
#define MINOR "0"
#define PATCH "108"
#define COMPILETIME "20211114123620"

**Architecture**
ESP32 Olimex EVB

**Bus system**
BSB

**Describe the bug**
For parameter 8730 only the value (e.g. 0) gets populated and the associated string is missing (e.g. Off).

**To Reproduce**
Should be visible in MQTT explorer

**Logs**
N/A

**Expected behavior**
String should be next to the value, like for any other parameters. Should be "0 - Off" instead of just "0"

**Screenshots**
[image]

**Desktop (if applicable, please complete the following information):**
N/A

**Smartphone (if applicable, please complete the following information):**
N/A

**Additional context**
N/A

1coderookie, 2021-12-27:
Please make an update of BSB-LAN first and try again. If it's still the same, please report it again here.

fredlcore, 2021-12-27:
Also, please check if this problem also occurs with other parameters or only with this one. And please compare the MQTT output with the output of the Webinterface to see if the problem exists there, too.

olympia, 2021-12-27:
I updated to v2.0.111 and the problem persists (by the way I had to revert the change in line 513 of BSB_LAN.ino otherwise I couldn't compile).

I attached a screenshot to the report where it is visible that from the selected parameters exposed to MQTT only this is having the issue (see last two lines of the screenshot).

Web interface doesn't have this issue:

Screenshots:
[image]

[image]

fredlcore, 2021-12-27:
I think I found the bug, should be fixed in the current master repo. Please try and report back here, thanks!

closed 2021-12-27 by fredlcore

olympia, 2021-12-27:
Hi Frederik,

Great, I confirm the issue is fixed! Many thanks!

Do you have an idea/ hint why I cannot compile with `EEPROMClass EEPROM_ESP((const char *)PSTR("nvs"));` being in line 513 of BSB_LAN.ino and I have to revert it back to `EEPROMClass EEPROM_ESP("nvs", EEPROM_SIZE);` to get a successful compile?
Shall I open a separate issue for this? ...but I guess this is more of a support query :)

I get the following errors:
```
BSB_LAN:513:49: error: invalid conversion from 'const char*' to 'uint32_t {aka unsigned int}' [-fpermissive]
 EEPROMClass EEPROM_ESP((const char *)PSTR("nvs"));
                                                 ^
In file included from C:\Users\olympia\Downloads\BSB_lan\BSB_LAN\BSB_LAN.ino:505:0:
C:\Users\olympia\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\EEPROM\src/EEPROM.h:37:5: note:   initializing argument 1 of 'EEPROMClass::EEPROMClass(uint32_t)'
     EEPROMClass(uint32_t sector);
     ^
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\olympia\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
invalid conversion from 'const char*' to 'uint32_t {aka unsigned int}' [-fpermissive]
```

fredlcore, 2021-12-27:
You haven't installed SDK 2.0.2 completely/properly. As you can see, it is still using the EEPROM library from 1.0.6 whereas it should be using the one from 2.0.2. Either try to update the EEPROM library from the library manager or uninstall and then reinstall the ESP32 board or - if all this fails - uninstall and reinstall the Arduino IDE from scratch. Please let us know which way works in the end so we can add this to our FAQ.

olympia, 2021-12-27:
Spot on.! 
To install SDK 2.0.2 I had to reinstall ESP32 board by adding/ replacing the "Additional board manager URLs" in File/Preferences in Arduino IDE to include:
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
...and then I could update ESP32 board to 2.0.2

fredlcore, 2021-12-28:
Great, that's how it is mentioned in the manual now as well, so we won't have to add anything to the FAQ then, thanks!

olympia, 2021-12-28:
Well, not entirely. The link is different in the manual and using that link v1.0.6 would be the latest.
This is actually what I had as well on my install and I had to replace this with the link above to enable the installation/ update to v2.0.2

fredlcore, 2021-12-28:
Are you referring to the PDF manual or the one on GitHub? On the latter (https://1coderookie.github.io/BSB-LPB-LAN_EN/chap12.html#121-installation) we have updated everything accordingly already. The PDF will take a few days more.

olympia, 2021-12-28:
Right, I was looking at the pdf now. I now see the update on the online
version from 3 days ago. All is fine :)
Thank you!

1coderookie, 2021-12-28:
I updated the PDF at the same time though, so you probably weren't looking at the PDF from the latest download..?
[image]

olympia, 2021-12-28:
yes, yes; sorry for the noise :)