fredlcore/BSB-LAN #432: [FEATURE REQUEST] Add UDP broadcast of log parameters

fredlcore, 2021-12-31:
A year ago or so, we had two users who implemented a UDP transmission of parameters to use with their DIY room units to display various parameters there.
I was thinking whether it would be possible to use our current MQTT functionality and just add a few command to send the same payload via UDP, probably just as a broadcast to keep it simple. The receiver(s) could then parse the payload and display whatever they need. 
A receiving function could also be implemented the same way so that the MQTT processing routines could be used, like for setting a parameter (DHW push after pressing a button on the room unit etc.).

DE-cr, 2022-06-12:
Add one more user who did this just recently (https://github.com/DE-cr/BSBmonCR). :)

I'd be willing to do an implementation proposal - once some design decisions have been made:
1. Shall the parameter list and send interval be shared with logging (less implementation effort) or shall a separate configuration set be implemented (my preference)?
2. What data format shall be sent? The same as in logging (e.g. "364593010;01.05.2022 00:00:15;8314;Kesselrücklauftemperatur Ist;66.7;°C"), or a more concise one (e.g. BSBmonCR would use "8314:667" for the same data)?
3. Or should UDP sending be more like MQTT, as fredlcore suggested? I haven't looked at MQTT, yet, but would propose to keep the UDP data as easily parseable (and concise?) as possible.

Personally, I would
a. definitely use UDP broadcasting, not sending to one (or multiple?) dedicated receivers
b. not allow parameter setting via UDP. This can already be easily done via a simple "HTTP GET /S<x>=<y!z>", and UDP's broadcast capabilities are not required here, but a confirmation would be nice.

@fredlcore, let me know what you think.

fredlcore, 2022-06-24:
For now, I would keep the UDP logging parameters the same as the general logging parameters. I don't think someone will be using SD-card logging and UDP logging **and** use different parameters on both.
The issue I see here is the update interval. If you were to use UDP for updating, say, in a project like the one you mentioned, it makes a difference if you poll 20 parameters within 10 seconds (which will clog the bus and BSB-LAN) or just two or three. So I'd suggest adding a UDP update interval in the configuration.
As for the data format, I think it's easiest to take the format we already use for logging on SD-card or MQTT. That way, the remote device does not need to make a look-up to find out what the parameter number means, what unit it is in etc. and transmission time should not be affected in a noticeable way.

DE-cr, 2022-06-24:
FYI, to not clog the system, my BSBmonCR doesn't fetch+send all (of its fixed six) parameters at once every send interval, but instead rotates through them and only fetches+sends the next one in each send interval (2s for BSBmonCR).

fredlcore, 2022-06-24:
But it would mean that certain more critical parameters would only be sent after a certain amount of time. One of the scenarios I'm thinking of is a room controller/display, maybe it's even similar to what you've been creating. If I use a ePaper display, I can put the ESP to sleep for let's say a minute or so, wake up for max. 10 seconds (if that's the UDP interval) and update the screen, then put it to sleep again. If I have to wait for the whole interval to be completed, that would take quite a bit longer, in addition to the fact that the room controller doesn't know how many parameters to expect (unless you hard-code it). 
Maybe if the room-controller could initiate a request of some sort, it may alleviate some of the issues, because generally I like the suggestion of the rotation...

DE-cr, 2022-06-24:
For critical parameters, I would not rely on a UDP broadcast. For those, I would use the existing "GET /<param>".

How is clogging the system avoided with many parameters in /D logging? (Not at all, iirc.) Maybe this is isn't really a problem then, and the easiest (=least invasive) solution would be to just use the existing logging code and include some lines to also/alternatively send the very same lines that are/could be written to the log file as a UDP broadcast. Added user configuration would then be limited to UDP port number and log data destination choice (file and/or udp).

fredlcore, 2022-06-24:
It's not critical in the sense that the actual parameter is required every x seconds, but the broadcast in itself should be done regularly and in a compact manner, so that the ESP can go to sleep again.
For /D, there is no clogging prevention, only the limiting factor of 20 (IIRC) parameters means that logging would be completed within one minute. Of course if you set the interval to 10 seconds or so, it will surely get clogged.

DE-cr, 2022-06-24:
My proposal would then be to go forward as described in my previous comment here.
Let me know if I should prepare a pull request.

fredlcore, 2022-06-24:
Which previous comment exactly and which way? The cycling variant would not provide one compact UDP transmission, but rather several ones which would make it impossible for the ESP to get all data and go to sleep again.

DE-cr, 2022-06-24:
Use the /D logging code:
add UDP sending to logfile writing,
let the user configure which one (or both) to use.

fredlcore, 2022-06-24:
Ah, ok, yes, fine!
@dukess: Could you add a setting to activate logging via UDP please (similar to MQTT, no extra category, can be added to "Logging")? Thanks!
For UDP port, I'd choose 6502.

fredlcore, 2022-06-24:
;)

DE-cr, 2022-06-24:
https://www.speedguide.net/port.php?port=6502
[image]
Maybe it's better to make the port number user configurable. ;) This would also help to avoid collisions. We're talking about broadcasting, after all ..

dukess, 2022-06-24:
@fredlcore, @DE-cr, did it: https://github.com/fredlcore/BSB-LAN/pull/457/commits/ed5ec260e29cac0d320564a06c2252a1c710c49a

fredlcore, 2022-06-24:
@DE-cr: NetOp is a rather special IT environment application, I don't think that this will collide with our use-case. And no, as mentioned in several issues before, no more special interest configuration options that 99% of the users are never going to use, it just pollutes the configuration screen further. So either people change the port in the code if they really need to or they don't use the feature ;).

fredlcore, 2022-06-24:
@dukess: Thanks a lot, that was really quick! But as mentioned above, please remove the port configuration options for MQTT and UDP broadcast. Thanks!

dukess, 2022-06-24:
> But as mentioned above, please remove the port configuration options for MQTT

Looks like i lost something important :D

fredlcore, 2022-06-24:
Maybe I just thought I had written it in the MQTT issue already, but maybe I didn't - sorry ;)! And thanks for the quick fix!

DE-cr, 2022-06-25:
With just a single on/off value to enable/disable UDP logging: Should UPD logging disable SD card writing, or should it work in parallel? The latter would require an SD card for UDP transmissions, which doesn't make sense to me.

Should UDP logging only use datalog.txt data or also include journal.txt data?

fredlcore, 2022-06-25:
No, UDP and SD card logging are separate of each other. But I think there's no need to differentiate because the part where it writes to SD card checks if an SD card is present. If not, it won't write. 
Since logging applies to the logging parameters, only datalog.txt applies. IIRC (currently away from home), journal.txt logs broadcast messages. We could send these as well, but since they do not happen at the same time as the queried logging parameters, we won't be able to send it in one go. Plus, people interested in these values can simply also query them.

DE-cr, 2022-06-26:
So if I want to have UDP "logging" only, I would have to use this define in the config file on an ESP32 (and not have an SD card attached to it)?:
```
// Use SD card adapter on ESP32-based boards instead of SPIFFS flash-based storage
//#define ESP32_USE_SD
```
Also, an attached SD card might be used for different purposes by custom extensions. Why waste space on it when UDP-only logging is wanted?

I can understand that you don't want to clutter the /C page, but why not use a four-state setting for logging then, as I've suggested some comments ago?: logging = off / SD / UDP / SD+UDP

fredlcore, 2022-06-26:
Yes, that would be the way to go, same as with MQTT. And that would also prevent a four-state setting because with MQTT it would already be eight states.
I agree that it is a bit counter-intuitive for the ESP32 where you have to add another definement in order not to use the internal flash, but other than that, it's quite straightforward: You want logging (of whatever kind), and if you do not want to log on SD card, you simply do not insert one. 
As for "custom extensions" using the SD card. that's a bit far fetched. Not only because I don't know of any except maybe a handful, but what other purpose would someone use the SD card for besides logging?
And even if there were such custom solutions, BSB-LAN is not going to develop into a 'Homer Car' (anymore than it already is ;) )
https://www.youtube.com/watch?v=WFZUB1eJb34

I'm not sure, @dukess, how easy it would be to implement multiple select options. If that would be possible, we could use that and make one configuration option to define the logging option as a bit pattern where users could (un-)select SD card logging, MQTT, UDP and whatever other logging options would come up in the future.

dukess, 2022-06-26:
@fredlcore it is a good question, i need to think how to do it.

fredlcore, 2022-06-26:
I think the actual values would be provided by the HTML code (0, 1, 2, 4, 8 etc. until 128), so storing that in a one byte value should not be an issue. Decoding it and creating the corresponding settings is maybe a bit more work, similar to VT_BIT, I guess...

dukess, 2022-06-28:
Current progress: https://github.com/fredlcore/BSB-LAN/pull/457/commits/9f71bf86009fd3a71b5a36b44a7c802c8ca13f09
[image]

dukess, 2022-07-02:
@fredlcore i implement new switch and it work now.
[image]

fredlcore, 2022-07-03:
Great, thanks a lot! I'm just thinking that IPWE is not really logging, so that may be a bit confusing? Did we have an extra setting for this or was it through the use of 'define'?

dukess, 2022-07-03:
Yes, you're right, IPWE is not a real log. 
My idea was that the "logs" include all the ways to send data outside the program.

fredlcore, 2022-07-03:
I think it's a rather special function that only few people (but including myself ;) ) are using, so it should stay in the "advanced" configuration in order not to confuse people ;)...

fredlcore, 2022-07-03:
@DE-cr here's the code I used when experimenting. I was using a reduced set of data and used pipes as delimiters, but I guess using the output from the logfile is a better option as long as the semicolons are not a problem...
```
// UDP broadcast
#ifdef WIFI
    IPAddress t = WiFi.localIP();
#else
    IPAddress t = Ethernet.localIP();
#endif
    t[3] = 255;
    udp.beginPacket(t,6502);
    udp.print(param);
    udp.print('|');
    udp.print(decodedTelegram.prognrdescaddr);
    udp.print('|');
    udp.print(decodedTelegram.value);
    udp.print('|');
    if (decodedTelegram.unit == NULL) {
      udp.print("NONE");
    } else {
      udp.print(decodedTelegram.unit);
    }
    udp.print('|');
    udp.print(decodedTelegram.enumdescaddr);
    udp.print('|');
    udp.endPacket();
// UDP broadcast end
```

dukess, 2022-07-03:
> I think it's a rather special function that only few people (but including myself ;) ) are using, so it should stay in the "advanced" configuration in order not to confuse people ;)...

No problem :)

DE-cr, 2022-07-03:
Being lazy, and with your suggestion to send the same data over UDP as used in SD card logging, I would have compiled a line to send to both places, then used a single line to send it to the SD card and another single line (ok, plus UDP begin/end) to send it over UDP...

Why should sending semicolons over UDP be a problem?

fredlcore, 2022-07-04:
Sounds good to me, the only problem is that the SD log line is using semicolons to make it easy to import them as a CSV into Excel for example. But there is no real sanity check, so for example if there is a unit that has a semicolon in it (such as &deg; for degree), then this would be a problem. We used to use HTML encoded units before, but no longer, so this _should_ not be a problem at the moment, but I can't guarantee this also to be the case for all text descriptions etc.
Writing sane CSV code would be an option, but that's quite a daunting task because of all the character escapings that might theoretically be necessary (semicolons that would put the value into quotation marks, and if there are semicolons as well as quotation marks ;)...)

DE-cr, 2022-07-05:
I think that right now, replacing semicolons in datalog.txt (!) encoding is more likely to break any user add-ons than also using them in UDP transmissions, but if you're really afraid of what you've said, I, being lazy, would rather move to TSV encoding (tab separated values, should also be easy to import to Excel, etc.) than write "sane" CSV code. Just my 2c...

fredlcore, 2022-07-05:
TSV is actually a good idea, but yes, it might break compatibility and since it's not that likely that we'll have semicolons in the text, we should just keep it as it is for now and act if there are concrete issues...

DE-cr, 2022-08-17:
I see that UDP sending has made it into the 2.1.8-20220731102301 release, but without a comment to that effect in the version history. As it is, there seems to be no documentation regarding the port used for sending, other than the source code itself. How about adding that info directly on the config page?

Personally, I would also find a different order of the config params more helpful:

[image]

...and couldn't this issue here be closed now? :)

DE-cr, 2022-08-17:
Btw, this change seems to have introduced a minor incompatibility to the previous version:
After the update, "Write to SD card" was enabled on my system, although the old "Logging" (?) parameter was set to off before the update.

DE-cr, 2022-08-17:
Ok, seems so far it's only the config page prepared for UDP sending, without the actual implementation, yet...
No worries.
Should I prepare a pull request when I find the time?

dukess, 2022-08-17:
yes, it would be cool!

fredlcore, 2022-08-17:
Yes, I assumed that you'd work on this after @dukess prepared the necessary settings. Part of a good PR is also a single line for the changelog in BSB_LAN.ino ;)...
I wouldn't state the port in the config page, since we also don't do it for MQTT etc. For such a special case, people should refer to the manual, or at least changelog or config...

DE-cr, 2022-08-18:
Could it be that moving configuration parameters around on the /C web page (as I suggested above) would also move their location in the EPROM, rendering its contents incompatible?

dukess, 2022-08-19:
No, option place in eeprom depend from their version in table and their 
place in enum.
So here should be no problems.

DE-cr, 2022-08-19:
Thanks, @dukess!
I've now reordered the parameters on the configuration web page as proposed above:
[image]

DE-cr, 2022-08-20:
to be closed now? :)

fredlcore, 2022-08-20:
It's your issue, so you can close it anytime, but if you want me to do it, that's fine as well...

closed 2022-08-20 by fredlcore