fredlcore/BSB-LAN #435: 2.0 Show "New Vision found"

Saharel001, 2022-01-11:
**BSB-LAN Version**
2.0

**Describe the bug**
In Version 2.0 i get a new version found 2.1.0 massage but in git the last release is 2.0.0

**Screenshots**
[image]

dukess, 2022-01-11:
Current development version is 2.1.0, so it is correct. But last released version is 2.0.0, you right.

Saharel001, 2022-01-11:
@dukess 
Its ok for me, but in my opinion ist a dev version not a stable release. But it´s possible that my interpretation from "New Version Found" ist wrong ;)

1coderookie, 2022-01-23:
We never claimed that a dev version is a stable release - but I'll add a note to the manual, that 'show updates' also includes dev versions..

closed 2022-01-23 by 1coderookie