fredlcore/BSB-LAN #436: [BUG] Brötje EVO 15i parameter 7700 (Relaistest) value description wrong

eikejenn, 2022-01-15:
BSB-LAN Version: Version: 2.1.0-20211231181308
Architecture: Arduino DUE
Bus system: BSB-Bus

### Description:
When i start a relais test directly at the control panel from the LMS15, then i can see the right Value ID in the BSB UI, 
but the test-description is wrong.
example: relais test for "Modul 1 output QX23" is shown as "Heizkreispumpe Q6"
[image]

In the manual from Brötje i can find the following assignment:

>Relaistest (Parameter 7700)
0: Kein Test
1: Alles aus
2: Relaisausgang QX1
3: Relaisausgang QX2
4: Relaisausgang QX3 Relaisausgang QX21 Modul 1
7: Relaisausgang QX22 Modul 1
8: Relaisausgang QX23 Modul 1
9: Relaisausgang QX21 Modul 2
10: Relaisausgang QX22 Modul 2
11: Relaisausgang QX23 Modul 2
12: Relaisausgang QX21 Modul 3
13: Relaisausgang QX22 Modul 3
14: Relaisausgang QX23 Modul 3 

Please check and correct if possible the wrong assignment
Thanks from north germany
Eike

fredlcore, 2022-01-16:
We know for a while that the same parameter number does not have the same meaning across different controllers, let alone the same parameter options (as well as internal option values).
We will have to make a decision if we remove clear-text support for these parameters completely to avoid any misunderstanding or even unintentional use, or if we still try to make a comprehensive list of options for all controllers we know of currently. The latter will take a significant amount of time and it wouldn't help to start here and there wherever we find a matching problem. In the end, it will depend on the question how many different sets we will have to implement for the different controllers and if these all still fit into the memory of the Arduino/ESP32.

closed 2022-01-16 by fredlcore

eikejenn, 2022-01-16:
For me it was a nice Workaround, when globally the Value ID is shown additionally to the text in the drop down menu in the UI.
Then i can double check the ID with my heater manual before click "set". Is this possible?

fredlcore, 2022-01-16:
What do you mean "was" a nice workaround? The ID is still displayed:
[image]
Not in the dropdown, but in the text to the left.
You can still see all the values of the dropdown if you display the source of the webpage.

eikejenn, 2022-01-16:
Sorry for the bad english... ;-)
it would be a nice workaround.. not it was. sorry.
When the value ID is shown additionally in the text field from the dropdown menu, 
then I can save myself the search in the html source code. ;-)

fredlcore, 2022-01-16:
Yes, but it would affect every other user who is importing this in their home automation system. So when they set up "Status Heizkreis" and take the value from the dropdown, it would say "1 - Automatik". For that purpose we removed the ID from the dropdown.
And look at it from that perspective: With the source view, you have alle the parameters in one category with all the option values and don't have to scroll through sometimes long dropdown menus ;)...

eikejenn, 2022-01-17:
I found out the right place in the source code to add the print command for my desired feature. ;-)
[image]
I am happy now 👍

fredlcore, 2022-01-17:
Great, but you won't be able to update unless you re-add this feature every time again. But it's your call.