fredlcore/BSB-LAN #438: [QUESTION] How to match DT_BITS to enum values

escoand, 2022-01-19:
Sorry for not opening feature request or bug report, but hopefully somebody could help me anyway.
I'm using only the BSB_LAN_defs.h file for my own project but this file is already a massive help, thanks

For me it's no clear how the `DT_BITS` units are translated to the actual enum values.
I'm asking for this specific message:
```
src=0x80 dst=0x7f len=14 type=0x02 cmd=0x31000212 data=01:00:45
```
which is in the commands table
```
{0x31000212,  CAT_TW,               VT_BIT,           1602,  STR1602,  sizeof(ENUM1602),     ENUM1602,     DEFAULT_FLAG, DEV_ALL}, // Status Trinkwasserbereitung
```
and in the units table
```
{VT_BIT,            1.0,    1, 1, DT_BITS, 0,  U_NONE, sizeof(U_NONE), STR_BIT},
```

Long story short, the value and bitmask resolves to
```
  00000001
& 01000101
----------
= 00000001
= 0x01
```
but how is this matched to the specific enum value?
```
#define ENUM1602_00_02_TEXT "TWW-Push nicht aktiv"
#define ENUM1602_02_02_TEXT "TWW-Push aktiv"
#define ENUM1602_00_04_TEXT "TWW aus"
#define ENUM1602_04_04_TEXT "TWW an"
#define ENUM1602_00_08_TEXT "TWW Ladung nicht aktiv"
#define ENUM1602_08_08_TEXT "TWW Ladung aktiv"
```

dukess, 2022-01-20:
Just an example on JavaScript
```
                              1*  2* 
#define ENUM1602_00_02_TEXT "TWW-Push nicht aktiv"
#define ENUM1602_02_02_TEXT "TWW-Push aktiv"
#define ENUM1602_00_04_TEXT "TWW aus"
#define ENUM1602_04_04_TEXT "TWW an"
#define ENUM1602_00_08_TEXT "TWW Ladung nicht aktiv"
#define ENUM1602_08_08_TEXT "TWW Ladung aktiv"

1* - value (state)
2* - bit mask

          var bitmask = 0;
          var bitvalue = parseInt(currentValueObject.value, 2); //Value, ex. from program number 1602
          for (var i = 0; i < sel.options.length; i++) { //we should read ENUM list value by value
            sel.options[i].selected = false; //clear selection flag
            var val = parseInt(sel.options[i].value); //value from ENUM
            bitmask = val & 0xff; //got valuable bit from ENUM value
            val = (val >> 8) & 0xff; //got state (zero or non-zero) for this ENUM value
            if((bitvalue & bitmask) == (val & bitmask)){ //comparison bitvalue with current ENUM value
              sel.options[i].selected = true; //set selection flag if it matched
            }
```
i.e., if we don't know the list of values, we won't be able to decode the value correctly.

fredlcore, 2022-01-20:
First of all, maybe you tell us a bit about your project, because with the limited resources I have for the core project, I couldn't afford explaining all sorts of coding background if there is no return benefit for this project itself. Hope you can understand.

escoand, 2022-01-20:
@fredlcore Of course, I do understand. I'm working with ESPHome and try to work with BSB with it. I started with creating something similar as the command table but now just use this one. I would never be able to collect all these commands. Thanks again.

@dukess Thanks for the code example, this helped a lot. I've seen currently these two different message and your are actually doing this, right?
```
### message 1
A*        B*        C*
0000_0001 0000_0001 0100_1101

                 1* 2*                                    1*      2*     C*           
#define ENUM1602_00_02_TEXT "TWW-Push nicht aktiv"     => 0x00 == 0x02 & 0100_1101   => true
#define ENUM1602_02_02_TEXT "TWW-Push aktiv"           => 0x02 == 0x02 & 0100_1101   => false
#define ENUM1602_00_04_TEXT "TWW aus"                  => 0x00 == 0x04 & 0100_1101   => false
#define ENUM1602_04_04_TEXT "TWW an"                   => 0x04 == 0x04 & 0100_1101   => true
#define ENUM1602_00_08_TEXT "TWW Ladung nicht aktiv"   => 0x00 == 0x08 & 0100_1101   => false
#define ENUM1602_08_08_TEXT "TWW Ladung aktiv"         => 0x08 == 0x08 & 0100_1101   => true

### message 2
A*        B*        C*
0000_0001 0000_0000 0100_0101

                 1* 2*                                    1*      2*     C*           
#define ENUM1602_00_02_TEXT "TWW-Push nicht aktiv"     => 0x00 == 0x02 & 0100_0101   => true
#define ENUM1602_02_02_TEXT "TWW-Push aktiv"           => 0x02 == 0x02 & 0100_0101   => false
#define ENUM1602_00_04_TEXT "TWW aus"                  => 0x00 == 0x04 & 0100_0101   => false
#define ENUM1602_04_04_TEXT "TWW an"                   => 0x04 == 0x04 & 0100_0101   => true
#define ENUM1602_00_08_TEXT "TWW Ladung nicht aktiv"   => 0x00 == 0x08 & 0100_0101   => true
#define ENUM1602_08_08_TEXT "TWW Ladung aktiv"         => 0x08 == 0x08 & 0100_0101   => false
```

And `B*` and `C*` value `0x80`, `0x40`, `0x20`, `0x10` and `0x01` is just not known, right?

fredlcore, 2022-01-20:
Interesting, what devices do you use for ESPHome? Does our adapter work with it or do you have to solder your own?

escoand, 2022-01-20:
I'm still in a testing phase and connect without an adapter, and also just reading yet. Anyway I normally use [Wemos D1 mini](https://www.wemos.cc/en/latest/d1/d1_mini.html) but it is too limited for the huge def file. So I switched to [LOLIN32](https://www.az-delivery.de/products/esp32-lolin-lolin32).

dukess, 2022-01-20:
@escoand yep.

closed 2022-01-20 by escoand

fredlcore, 2022-01-20:
Ok, just FYI: Have a look at our license when using our code in case you want to use it in a commercial context (which would be prohibited).