fredlcore/BSB-LAN #441: [BUG]

proasnet, 2022-01-27:
**Architecture**
due

**Bus system**
bsb

**Describe the bug**
In the manual, chapter 8.11 is description abut home assistant. In this chapter is link
`https://www.home-assistant.io/integrations/bsblan/`
In this site is about compatibility
The integration is tested with the stable firmware version 1.00. A newer firmware versions will not work, because the parameters are changed of the specific info that is needed. Please use the latest release. release 1.0

I tried this, still not working.....

fredlcore, 2022-01-27:
What did you try exactly? The BSB-LAN integration for HA only works for BSB-LAN software versions 1.0 so far because the integration didn't apply the changes in the API that have happened over the last year or so. There is nothing we can do on our side except to recommend using BSB-LAN version 1.0 if the integration for HA is necessary.
However, we have shown alternative ways of using BSB-LAN with HA which are described in our manual.

1coderookie, 2022-01-27:
Please read the note at the very first beginning of that chapter 8:
(..) If you should discover a code example or a module, which is not executable in the represented form with the current BSB-LAN version v2.x, then please inform the author of the example/module (..) 

About the HA implementation itself: as already mentioned in the manual also, that one was developed by Willem-Jan aka @liudger - so if you want to contact him, feel free to do that (but he knows about this circumstance, just to already let you know that). 

As fredlcore already mentioned, for the recent BSB-LAN version 2.x you can use the alterantive methods of implementing BSB-LAN, I'd suggest to follow the example from user tiger42 which you find when you scroll a bit down within the HA chapter.

liudger, 2022-01-27:
Yeah I need to find some time to make it compatible with newer versions. Still not much time. Mqtt or other options are still a good way to integrate bsblan. You can find it in the forum of homeassistant.

liudger, 2022-01-28:
There is a small bug in the current version that the thermostat becomes unavailable. This is solved in next release. You can manually reload the integration and the device will become available

proasnet, 2022-01-28:
In some moments, integration was connected to BSB, but a numbers has bad calibration... I wrote this to liudger. But in this moment, I have this integration removed from HA. But, in time, where was working, was easy and good.
If I will have some time, I try another methods how integrate BSB to my HA.
Thank you.

closed 2022-01-30 by 1coderookie

liudger, 2022-01-31:
Please fill a bug on the component or python lib. I am working now on 2.0 support and if it is another issue than the software I could have a look.