fredlcore/BSB-LAN #443: [BUG]

ivandompe, 2022-02-09:
Hello,

I just received my BSB-LAN adapter and try to set-it up... Everything looks fine but not my MQTT config.
From the Broker nothing is structured in terms of topics... I tried all the mqtt-mode (1-2-3).. but still the same. Here below my conf.

**BSB-LAN Version**
2.1.0-20211231181308

**Architecture**
ESP32 NodeMCU

**Bus system**
BSB

**Describe the bug**
Incorrect (unexpected) hierarchy applied on the MQTT Broker side?

#define MQTT
byte mqtt_mode = 1;   ******* I TRIED 2 & 3 option. *******
byte mqtt_broker_ip_addr[4] = {192,168,0,201};
char MQTTUsername[32] = "mqtt@user";                 
char MQTTPassword[32] = "mqtt@pwd";                 
char MQTTTopicPrefix[32] = "BSB-LAN"; 	    
char MQTTDeviceID[32] = "BSB-LAN";

**Expected behavior**
All parameters from BSB must be displayed as topics.

**Screenshots**
Hierarchy from BSB-LAN:
[image]

Hierarchy from others clients:
[image]

As you can see, instead of having "BSB-LAN" as prefix, I have $SYS... and no devices.. and parameters as topics. Strange.
Maybe I missed something?

THX for your help.

Ivan

fredlcore, 2022-02-09:
@Luposoft63 and @dukess, do you have any idea what could have gone wrong? @ivandompe, could you also please attach your BSB-LAN_config.h for us to check?

dukess, 2022-02-10:
$SYS is used for system information.

I never used mqtt so these ideas can looks naive:
root topic BSB-LAN should be created on mqtt broker side?
Firewall?
Try to run wireshark/tcpdump for monitoring network connections

ivandompe, 2022-02-10:
Hi,

Herewith attache the config.h file.

@dukess , no the topics are written by the client. That is strange because even the MQTT DEVICE prefix is not taken in account. I have another project running on another ESP32, using the same PubSubClient.h library, and works perfect.
I tried to flash again the ESP, reboot it, test several configs.. nothing better.

Meanwhile, as I also have Home Assitant installed on a server, I saw that there is an integration for BSB-LAN.. nice.
So I installed this https://www.home-assistant.io/integrations/bsblan, and after setting the requiered parameters, I have a successful registration in HA, but I don't see any device... nothing. mmmmh
[image]
Maybe it has nothing to do with my MQTT issue? Don't know.

I didn't tried yet some http GET/POST request with Postman... maybe here also I'll have issue. Have to test.
I maybe have a general issue, or I made a bad mistake on the config file?

[BSB_LAN_config.h_investigation.zip](https://github.com/fredlcore/BSB-LAN/files/8039728/BSB_LAN_config.h_investigation.zip)

Thx for your support.
Ivan

1coderookie, 2022-02-10:
Hi,
first of all, please read the manual, especially the chapters [configuration](https://1coderookie.github.io/BSB-LPB-LAN_EN/chap02.html#22-configuration) and [MQTT](https://1coderookie.github.io/BSB-LPB-LAN_EN/chap05.html#52-mqtt).

2nd: I don't know if you changed anything within the webconfig (you could send the output from url/CO so we could check), but at least within the config.h you have the loginterval set to default = 3600, which means that the parameters will be sent once in an HOUR (3600 seconds = 60 minutes). So you maybe want to change that, to have the values coming in more often, which would make it easier for you to check if it works or not.  

3rd: About your HA integration:
> Meanwhile, as I also have Home Assitant installed on a server, I saw that there is an integration for BSB-LAN.. nice. So I installed this https://www.home-assistant.io/integrations/bsblan, and after [...]

In the chapter about the [HA integration](https://1coderookie.github.io/BSB-LPB-LAN_EN/chap08.html#811-home-assistant) of the manual, you see the note about the HA integration and the software version. The HA integration _does not work_ with BSB-LAN version 2.x.
If you scroll down a bit, you see the example from user "tiger42" - there he explained how to integrate BSB-LAN into HA using MQTT sensor.

ivandompe, 2022-02-10:
Hi again,

I'm so confused, it was my bad. But I read the manual, and I thought that I made everything correct.
Don't laughing please, the non-error was due to the 3600s = 1 hour updating.... grrrrrr!
I didn't understood that these logging params will affect the MQTT ones. And I was confused because on the Broker side I was able to see status topic and only $SYS with no device/param topics. Thus I was focused on that.

[image]

Thanks all of you for your great support.
You can close this ticket, apologies.

1coderookie, 2022-02-10:
No problem, glad your problem was solved ;)
This is in fact a pretty common problem and that's also the reason I wrote to please read the manual first, because everything is mentioned there. But I can understand that in the 'heat of the project' some informations get lost during reading. 
So, have fun using BSB-LAN!

closed 2022-02-10 by 1coderookie