fredlcore/BSB-LAN #450: [BUG] 'numAverages' was not declared in this scope

DonFalma, 2022-04-01:
**BSB-LAN Version**
master from github

**Architecture**
ESP32

**Describe the bug**
Compiling the Code with `#define AVERAGES` uncommented leads to error `BSB_LAN/BSB_LAN.ino:3815:91: error: 'numAverages' was not declared in this scope`

**Expected behavior**
Compiles without errors

**Additional context**
I think a possible fix for the problem is to move line 765 `#ifdef AVERAGES` to 766 in BSB_LAN.ino. 

Fixed:
```
static const int numAverages = sizeof(avg_parameters) / sizeof(avg_parameters[0]);
#ifdef AVERAGES
float avgValues_Old[numAverages] = {0};
float avgValues[numAverages] = {0};
float avgValues_Current[numAverages] = {0};
int avgCounter = 1;
#endif
```

fredlcore, 2022-04-04:
@dukess was kind enough to fix this quickly, so the current master repository should no longer have this error.

closed 2022-04-04 by fredlcore

dukess, 2022-04-04:
Oh, I didn't notice this problem, I discovered it independently. Yes, this correction option is also suitable, but my program size will be slightly smaller.
:)