fredlcore/BSB-LAN #452: [FEATURE REQUEST] MQTT port

tobyDoh, 2022-04-12:
**Is your feature request related to a problem? Please describe.**
MQTT Port is currently hard-coded in Programm

**Describe the solution you'd like**
Have the MQTT Port avaiable as Option in header as variable or better as option in webinterface

**Describe alternatives you've considered**
changing the coding prior to compiling process

**Describe your own contribution**
i will test the changes

fredlcore, 2022-06-24:
Since the webinterface is already quite "full", I'd prefer only to add necessary configuration options. What is the benefit of adding a configuration option to configure the MQTT port? Especially in relation to all other network ports we are using? If we start making one of them configurable, we would have to do the same with all the others as well. 
If you are looking for security through obfuscation, it won't bring much in a local network where local users connected to the network can easily monitor the unencrypted communication on all ports.

fredlcore, 2022-06-27:
Ok, thanks.