fredlcore/BSB-LAN #456: falsche Werte bei Parameter 1620

Transalpler71, 2022-05-28:
**BSB-LAN Version**
As displayed in the web-configuration or copy from `bsb-version.h` file. 2.1.3-20220209235153

**Architecture**
The architecture BSB-LAN is running on (Arduino Mega, Arduino Due, ESP32 NodeMCU, ESP32 Olimex EVB etc.) SP32 NodeMCU

**Bus system**
Which bus BSB-LAN is running on (BSB, LPB, PPS)? BSB

**Describe the bug**
A clear and concise description of what the bug is.
Der Parameter 1620 wird falsch ausgewertet - in der Heizungsparameteransicht erhalte ich 1620 Trinkwasser - TWW Freigabe: 2 - not found. Wenn ich den Parameter mit dem QAA75 Bedienteil abfrage, so erhalte ich: Zeitprogramm 4 / TWW.
Falls möglich bitte in den Heizungsfunktionen bei "Zeitprogramm 4" ergänzen auf "Zeitprogramm 4 / TWW"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Logs**
If possible, attach or copy/paste a log of the Arduino IDE's serial monitor when performing the above actions. If the error is reproducible, do so after pressing the reset button on the board.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (if applicable, please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
[image]

**Smartphone (if applicable, please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

1coderookie, 2022-05-29:
Bitte 
1) nenne uns dein genaues Heizungsmodell,
2) führe /Q aus (https://1coderookie.github.io/BSB-LPB-LAN/kap03.html#33-%C3%BCberpr%C3%BCfen-auf-nicht-freigegebene-reglerspezifische-command-ids) und 
3) melde uns den kompletten Output davon.
4) Logge sicherheitshalber bitte außerdem nochmal den Aufruf und die Antwort von 1620 mit dem Seriellen Monitor (https://1coderookie.github.io/BSB-LPB-LAN/kap12.html#122-serieller-monitor) und melde auch das.

Transalpler71, 2022-05-29:
Anbei die Ausgabe von /Q
[BSB-Lan_q.txt](https://github.com/fredlcore/BSB-LAN/files/8793507/BSB-Lan_q.txt)
und aus der Seriellen Konsole:
[BSB-Lan_SerMon.txt](https://github.com/fredlcore/BSB-LAN/files/8793510/BSB-Lan_SerMon.txt)

Es handelt sich um eine Wärmepumpe Herz Commotherm SW

1coderookie, 2022-05-30:
Super, danke - cool, ein neues Modell/Hersteller, dann trage ich das auch demnächst noch in der Kompatibilitätsliste im Handbuch nach :)

Bei der Ausgabe von Q siehst du, dass bei dir noch ein paar Parameter freizugeben sind, das wird dann irgendwann erfolgen und dann kannst du auch die via BSB-LAN bedienen (wird aber etwas dauern, derzeit ists zeitlich bei uns schlecht..). 

Was 1620 angeht: Hmm, komisch - es scheint, als hättest du evtl ne andere Command ID. Alle anderen Parameteroptionen werden aber korrekt dargestellt?
Lies bitte mal Kap. 09 (https://1coderookie.github.io/BSB-LPB-LAN/kap09.html) und folge dem Procedere für die Optionseinstellung 2 "Zeitprogramm 4 / TWW" von 1620, das Logfile dann posten. Falls 1620 auch bei anderen Parameteroptionen Probleme macht, dann bitte einmal komplett alle Einstellungen dekodieren und posten.
 
Btw: Du hattest ja noch dieses hier geschrieben
> Falls möglich bitte in den Heizungsfunktionen bei "Zeitprogramm 4" ergänzen auf "Zeitprogramm 4 / TWW"

Die Optionsbezeichnung ist nach wie vor "Zeitprogramm 4 / TWW". Insofern irritiert mich das etwas, dass du a) die Parameteroption einerseits nicht sehen kannst und b) dann aber wiederum sagst, dass dort nur "Zeitprogramm 4" steht. Verlief das Flashen problemlos? Hast du schonmal komplett neu installiert?

Kannst du bitte auch nochmal den Output von /CO posten?

Transalpler71, 2022-05-30:
Hallo,

Es handelt sich um eine WP aus 2010 - Herz hat glaub ich 2012 den controller durch etwas anderes ersetzt - sprich danach gelieferte Commotherm sind wohl nicht mehr per BSB-LAN anbindbar bzw. ab Werk mit NW Anschluss

Zu den Optionsbezeichungen ein kurzer Auszug aus der LANG_DE.h im Ordner Localization
#define ENUM_CAT_00_TEXT "Uhrzeit und Datum"
#define ENUM_CAT_01_TEXT "Bedieneinheit"
#define ENUM_CAT_02_TEXT "Funk"
#define ENUM_CAT_03_TEXT "Zeitprogramm Kühlkreis 1"
#define ENUM_CAT_04_TEXT "Zeitprogramm Kühlkreis 2"
#define ENUM_CAT_05_TEXT "Zeitprogramm Kühlkreis 3"
#define ENUM_CAT_06_TEXT "Zeitprogramm Heizkreis 1"
#define ENUM_CAT_07_TEXT "Zeitprogramm Heizkreis 2"
#define ENUM_CAT_08_TEXT "Zeitprogramm 3/HKP"
#define ENUM_CAT_09_TEXT "Zeitprogramm 4"
#define ENUM_CAT_0a_TEXT "Zeitprogramm Lüftung 1"
#define ENUM_CAT_0b_TEXT "Zeitprogramm Lüftung 2"
#define ENUM_CAT_0c_TEXT "Zeitprogramm 5"
hier sollte die Zeile 37 durch 
#define ENUM_CAT_09_TEXT "Zeitprogramm 4/TWW" 

ersetzt werden

In dem File fand ich folgende Einträge:
#define ENUM1620_00_TEXT "24h/Tag"
#define ENUM1620_01_TEXT "Zeitprogramme Heizkreise"
#define ENUM1620_02_TEXT "Zeitprogramm 4/TWW"
#define ENUM1620_3_01_TEXT "Alle Zeitprogramme HK/KK"
#define ENUM1620_3_03_TEXT "Niedertarif"
#define ENUM1620_3_04_TEXT "Zeitprog 4/TWW oder NT"

ich denke da sollte folgendes stehen:
#define ENUM1620_2_TEXT "Zeitprogramm 4/TWW"
oder liege ich da falsch - ist schon eine Zeit her, dass ich mich mich mit C/C++ auseinandergesetzt habe..

Anbei noch der Output von /CO

[BSB-Lan_CO.txt](https://github.com/fredlcore/BSB-LAN/files/8800119/BSB-Lan_CO.txt)

und das Serielle Log - im ersten Durchgang hab ich mal alle 3 Möglichkeiten durchgeschaltet, die mir die Bedieneinheit in der Fachmann Ebene angeboten hat:
[BSB-Lan_Verbose.txt](https://github.com/fredlcore/BSB-LAN/files/8800121/BSB-Lan_Verbose.txt)

Ich hab dann noch mal kompiliert und hochgeladen - beides ohne erkennbare Fehlermeldung..

Grüße - und danke für das Projekt!!

Transalpler71, 2022-05-31:
Hallo,

Leider kann ich die hochgeladenen Files nicht löschen...
https://github.com/go-gitea/gitea/issues/677

so sieht es per Telnet aus, wenn man die Einstellungen durchspielt am Bedienteil:
RGT1->HEIZ QUR 1610 Trinkwasser - TWW Nennsollwert:
DC 86 00 0B 06 3D 31 06 B9 0E 46
HEIZ->RGT1 ANS 1610 Trinkwasser - TWW Nennsollwert: 46.0 ┬░C
DC 80 06 0E 07 31 3D 06 B9 00 0B 80 1A C5
RGT1->HEIZ QUR 1612 Trinkwasser - TWW Reduziertsollwert:
DC 86 00 0B 06 3D 31 06 BA 3E 25
HEIZ->RGT1 ANS 1612 Trinkwasser - TWW Reduziertsollwert: 42.0 ┬░C
DC 80 06 0E 07 31 3D 06 BA 00 0A 80 B2 28
RGT1->HEIZ QUR 1620 Trinkwasser - TWW Freigabe:
DC 86 00 0B 06 3D 25 07 22 90 06
HEIZ->RGT1 ANS 1620 Trinkwasser - TWW Freigabe: 2
DC 80 06 0D 07 25 3D 07 22 00 02 58 43
DSP1->HEIZ QUR      053D07B0
DC 8A 00 0B 06 3D 05 07 B0 A4 8C
HEIZ->DSP1 ANS      053D07B0 00 19
DC 80 0A 0D 07 05 3D 07 B0 00 19 6D 58
RGT1->HEIZ SET 1620 Trinkwasser - TWW Freigabe: 65535
DC 86 00 0D 03 3D 25 07 22 01 01 A2 51
HEIZ->RGT1 ACK 1620 Trinkwasser - TWW Freigabe:
DC 80 06 0B 04 25 3D 07 22 81 CD
RGT1->HEIZ QUR 1620 Trinkwasser - TWW Freigabe:
DC 86 00 0B 06 3D 25 07 22 90 06
HEIZ->RGT1 ANS 1620 Trinkwasser - TWW Freigabe: 1 - Zeitprogramme Heizkreise
DC 80 06 0D 07 25 3D 07 22 00 01 68 20
RGT1->HEIZ SET 1620 Trinkwasser - TWW Freigabe: 65535
DC 86 00 0D 03 3D 25 07 22 01 00 B2 70
HEIZ->RGT1 ACK 1620 Trinkwasser - TWW Freigabe:
DC 80 06 0B 04 25 3D 07 22 81 CD
RGT1->HEIZ QUR 1620 Trinkwasser - TWW Freigabe:
DC 86 00 0B 06 3D 25 07 22 90 06
HEIZ->RGT1 ANS 1620 Trinkwasser - TWW Freigabe: 0 - 24h/Tag
DC 80 06 0D 07 25 3D 07 22 00 00 78 01
DSP1->HEIZ QUR      053D07B0
DC 8A 00 0B 06 3D 05 07 B0 A4 8C
HEIZ->DSP1 ANS      053D07B0 00 19
DC 80 0A 0D 07 05 3D 07 B0 00 19 6D 58
RGT1->HEIZ SET 1620 Trinkwasser - TWW Freigabe: 65535
DC 86 00 0D 03 3D 25 07 22 01 02 92 32
HEIZ->RGT1 ACK 1620 Trinkwasser - TWW Freigabe:
DC 80 06 0B 04 25 3D 07 22 81 CD
RGT1->HEIZ QUR 1620 Trinkwasser - TWW Freigabe:
DC 86 00 0B 06 3D 25 07 22 90 06
HEIZ->RGT1 ANS 1620 Trinkwasser - TWW Freigabe: 2
DC 80 06 0D 07 25 3D 07 22 00 02 58 43
DSP1->HEIZ QUR      053D07B0
DC 8A 00 0B 06 3D 05 07 B0 A4 8C
HEIZ->DSP1 ANS      053D07B0 00 19
DC 80 0A 0D 07 05 3D 07 B0 00 19 6D 58

0 und 1 werden richtig interpretiert und beschriftet - 2 gleichlautend mit Zeitprogramm 4/TWW entspricht leider nicht

Transalpler71, 2022-06-01:
Bezüglich der Timeouts - ich hab jetzt  fast 12 h durchgehend per telnet mitgesehen - bislang keine weiteren Timeouts gesehen

1coderookie, 2022-06-02:
Deine o.g. Anmerkungen zu den Telegrammen sind m.E. (nach einem kurzen Blick) nicht ganz eindeutig, aber ich _vermute_, dass das "TWW Freigabe: 65535 #OK"-Telegramm das ist, was wir suchen. Ich gehe davon aus, dass @fredlcore (der die Änderungen vornimmt) da durchsteigt - ansonsten wird er sich dann nochmal melden und nachfragen.

Wie gesagt, derzeit ist es zeitlich _sehr_ eng bei uns und es stehen noch etliche andere Dinge auf der ToDo-Liste, insofern wird es sicherlich noch eine Weile dauern, bis die Änderung erfolgt, aber sie wird kommen. Ich lasse das Issue hier deshalb auch noch offen, damit es nicht in Vergessenheit gerät.. ;)
 
Da es sich hier ja für dich grundsätzlich nicht um ein Funktionsproblem im Sinne von 'System ist nicht bedienbar' handelt, sondern 'nur' um eine Optionseinstellung, die man i.d.R. eh nicht großartig verstellt, sollte das Warten hoffentlich unproblematisch für dich sein. Wenn du es in der Zwischenzeit umstellen musst, mach es bitte per Raumgerät/Bedieneinheit.

Transalpler71, 2022-06-06:
ja das ist absolut kein Showstopper - somit nicht oberste Prio.

fredlcore, 2022-10-12:
Mit der Version 3.0 und der reglerspezifischen BSB_LAN_custom_defs.h wird sich dieses Problem auflösen.

closed 2022-10-12 by fredlcore