fredlcore/BSB-LAN #463: [BUG] Unknown telegram: 1D 6A FF FF FF   FF 00 90 ED - PPS- Brötje - Mannesmann

zonfacter, 2022-06-29:
BSB-LAN Version
As displayed in the web-configuration or copy from bsb-version.h file.
Version: **2.1.0-20211231181308**

Architecture
The architecture BSB-LAN is running on (Arduino Mega, Arduino Due, ESP32 NodeMCU, **ESP32 Olimex EVB** etc.)

Bus system
Which bus BSB-LAN is running on (BSB, LPB, **PPS**)?
maprotherm MGA 16
MRK K ( Baugleich:= Brötje Eurocontrol K)
MRK M ( Baugleich:= Brötje Eurocontrol M)
Keine Raumgeräte nur per BSB-LAN QAA 70

Beim beobachten per Telnet ist mir folgendes Telegramm aufgefallen:
Es kommt sehr häufig nachdem der Status der Präsenztaste abgerufen wird.

Gibt es irgendwo auch noch mehr Dokumentation über den PPS Bus und deren Telegramme ?

PPS   RTS
17 00 00 00 00 00 00 00 00
PPS ANS   15001 PPS-Bus - Präsenztaste: 0 - Aus
FD 4C FF FF FF FF FF 00 BC
**Unknown telegram: 1D 6A FF FF FF   FF 00 90 ED**
PPS INF
**1D 6A FF FF FF FF 00 90 ED**

PPS REQ 15001 PPS-Bus - Präsenztaste:
1E 4C FF FF FF FF FF FF 9C
PPS RTS
17 00 00 00 00 00 00 00 00
PPS ANS 15001 PPS-Bus - Präsenztaste: 0 - Aus
FD 4C FF FF FF FF FF 00 BC
**Unknown telegram: 1D 6A FF FF FF FF 00 90 ED**
PPS INF
**1D 6A FF FF FF FF 00 90 ED**
PPS RTS

Offtopic P.s: Spitzenklasse was ihr da auf die Beine gestellt habt !

[MobaXterm_192.168.2.136_20220629_215014.txt](https://github.com/fredlcore/BSB-LAN/files/9014277/MobaXterm_192.168.2.136_20220629_215014.txt)

fredlcore, 2022-06-29:
Danke, das 1D 6A-Telegramm gibt es auf den Steuerungen, die ich hier habe, bisher noch nicht. Da es sich um ein Telegramm handelt, das die Steuerung an das Raumgerät schickt, müsste man schauen, welche Bedeutung es haben könnte. 0x90 deutet auf eine Zeitangabe hin, denn die steht meist für "24h". Die 00 davor kann für etwas anderes stehen. Es könnte sein, dass die 24h für z.B. das Trinkwasserfenster stehen oder irgendetwas anderes, was in dem Fall den ganzen Tag über aktiv ist. Da müsstest Du mal schauen.

zonfacter, 2022-07-02:
Ok...ich werde mal schauen ob ich da hinter komme...

zonfacter, 2022-07-03:
Veränderung der folgenden Einstellungen hat keine Veränderung auf das Telegram gehabt:
polo.broetje.de/pdf/zu0035.pdf

12 Warmwasser-Betriebsart (0 = aus; 1 = ein)
16 Sommer/Winter-Umschalttemperatur
17 Heizkennlinien-Steilheit (ohne Raumeinfluß)
64 Heizkennlinien-Parallelverschiebung
65 Raumtemperatur-Einfluß (bei Witterungsführung)
67 Schaltdifferenz Raum
68 Vorlauftemperatur-Minimalbegrenzung
69 Vorlauftemperatur-Maximalbegrenzung
70 Gebäudebauweise (1 = leicht; 0 = schwer)
71 Heizkennlinienadaption (1 = wirksam; 0 = unwirksam)
78 Konstante für Schnellabsenkung (KON)
81 Warmwasserprogramm (0,1,2,3)
87 Uhrbetrieb; 0 = externe Uhr; 1 = interne Uhr; 2 = interne Uhr mit Verstellung; 3 = interne Uhr (Master)
88 Sommer/Winter-Umschaltautomatik 0 = lokale Umschaltung; 1 = zentrale Umschaltung aller Heizkreise
89 Wirkung des Standby-Schalter 0 = lokal; 1 = zentral

fredlcore, 2022-07-04:
Danke für's Ausprobieren - dann hilft wohl nur im Auge behalten und schauen, ob er sich doch noch mal anderweitig verändert und dann den Auslöser isolieren...

zonfacter, 2022-07-07:
> Danke für's Ausprobieren - dann hilft wohl nur im Auge behalten und schauen, ob er sich doch noch mal anderweitig verändert und dann den Auslöser isolieren...

Wird gemacht

fredlcore, 2022-12-09:
Ich schließe dieses Issue dann mal für den Moment, wenn es Klarheit über das Telegramm gibt, dann gerne hier oder bei mir direkt wieder melden.

closed 2022-12-09 by fredlcore