fredlcore/BSB-LAN #469: [BUG] Number type returns string '---'

hacki11, 2022-08-07:
**BSB-LAN Version**
2.1.8-20220731102301

**Architecture**
Due

**Bus system**
BSB

**Describe the bug**
Despite having a number (VALS/dataType 0) i receive a string for value '---' which can not be mapped to a number.
```
{
  "8326": {
    "name": "Brennermodulation",
    "dataType_name": "PERCENT",
    "dataType_family": "VALS",
    "error": 0,
    "value": "---",
    "desc": "",
    "precision": 1,
    "dataType": 0,
    "readonly": 1,
    "readwrite": 1,
    "unit": "%"
  }
}
```
**To Reproduce**
Steps to reproduce the behavior:
1. GET /JQ=8326

**Logs**
see json output

**Expected behavior**
`value: 0`

**Desktop (if applicable, please complete the following information):**
 - OS: Windows
 - Browser Chrome

fredlcore, 2022-08-07:
`---` is not the same as 0%. `---` means that this parameter is disabled, i.e. it has no function. Returning 0 for a disabled parameter would give the impression that the parameter is active.
For ENUMs we have assigned the value 65535 if you want to disable an ENUM-based parameter. This works because we know that ENUMs are always 16-bit wide. To do the same with numerical parameters won't work because 65535 could be a valid result.
Would it help if we added a property `isactive` or `isdisabled` and set it accordingly? (if that's possible, @dukess)

dukess, 2022-08-08:
Following Occam's Razor, I have to ask: do we need it? :)
It possible, but in this case we should to add new field for /JS command (is active or is disabled) and rewrite parser. Moreover, /S and /JS commands are similar in value syntax. If we will changing it then user will can use "---" in /S and cannot use it in /JS. Maybe we can replace "---" with 'null' value but this variant can confuse users.

hacki11, 2022-08-08:
Ok, i didn't knew it means disabled. I don't need the additional attribute, i just wanted to tell that it could lead to issues if a user want to parse it to a number. For my part i handle the '---' before parsing it so i worked around. Setting it to `null` (not `"null"`) would be another solution. Feel free to close.

fredlcore, 2022-10-12:
Do we still have something to do on this matter? Otherwise I'd close this issue.

mirkolenz, 2022-11-01:
Just to add another perspective on this conversation: My boiler also returns the string `---` for any number that is currently zero, for instance the modulation. That confused me initially, but I have since updated my scripts to handle this value as zero where it makes sense. I am not saying that the behavior needs to be changed, just wanted to add that initially this also confused me.

fredlcore, 2022-11-01:
Thanks, but actually there is a difference between 0% and "---". For example, you could have a temperature sensor that reads 0 degrees (outside temperature for example) or "---" if the sensor is not connected or disabled. For modulation it should be similar in the sense that "---" indicates that the burner is off/disabled/not present and 0% is the burner running at less than 1% (where the percentage actually means "above minimum modulation", so on my Thision which can modulate down to 10%, 1% would effectively mean 11% modulation).

mirkolenz, 2022-11-09:
That makes sense, thank you for the clarification!

closed 2022-11-09 by fredlcore