fredlcore/BSB-LAN #472: [BUG]?: Menu entry to display log file is disabled when logging is turned off - although "old" log file exists

DE-cr, 2022-08-18:
The behavior described in this issue's subject was unexpected to me.
However, it may have been an intentional design decision, i.e. not a bug.

DE-cr, 2022-08-18:
...the "old" log file can still be accessed by directly entering the corresponding url commands, though (e.g. /DG).

fredlcore, 2022-08-18:
Logging and log file are two different things, so there is no contradiction.

closed 2022-08-18 by fredlcore

DE-cr, 2022-08-19:
Just to better explain what I mean:
Having this configuration option unchecked:
[image]
...disables the "Ausgabe Logdatei" menu entry - even when such a "Logdatei" does exist:
[image]

With the above mentioned configuration option enabled, the menu entry is enabled (and now reads "Zeichne Logdatei", calling /DG).

fredlcore, 2022-08-19:
Yes, I understood. The alternative would be to either delete the log file when turning off logging (not a good idea if done unintentionally) or to keep the menu item active even when logging is turned off (which could give the impression that logging is still active). Both is not ideal and the current way is both non-destructive and shows the logging status at first glance.