fredlcore/BSB-LAN #475: Loop beim Klick auf Heizkreis 2

cpohle1983, 2022-08-30:
Hallo,

in der aktuellsten Version 2.1 begibt sich das Board in einen Loop beim Klick auf Heizkreis 2!
Anbei mal ein serielles Log. Ich habe im Anschluss version 2.0 geflasht und da klappt es problemlos!

Sagt bescheid, wenn noch was fehlt.

Danke!

[seriallog.txt](https://github.com/fredlcore/BSB-LAN/files/9450837/seriallog.txt)

fredlcore, 2022-08-30:
@dukess, it seems like this is related with your recent changes? The strange thing is that the HC1 category works fine and in HC2 it also seems to work allright until parameter 1042 from where it jumps back to 1040 and then does so indefinitely.  Could you have a look at this? Thanks!

dukess, 2022-08-30:
```
{0x223D0A88,  CAT_HK2,              VT_TEMP,          1042,  STR1042,  0,                    NULL,         DEFAULT_FLAG, DEV_ALL}, // [°C ] - Heizkreis 2 (nur wenn aktiviert) - Vorlaufsollwert Raumthermostat // Command ID tbc
{0x063D1214,  CAT_HK2,              VT_YESNO,         1033,  STR1033,  sizeof(ENUM_YESNO),   ENUM_YESNO,   DEFAULT_FLAG, DEV_ALL}, // Verlängerung Tagesheizgrenze HK1
{0x2E3D0D85,  CAT_HK2,              VT_PERCENT,       1044,  STR1044,  0,                    NULL,         DEFAULT_FLAG, DEV_ALL},  // Logged from LMS15

```
Hi, here is error in program numbers order. Just move 1033 upper.

fredlcore, 2022-08-30:
Ah, well spotted :)!
@cpohle1983, could you test and move the second line (1033) a few lines up so that it is in order regarding the parameter number (column 4) and test again?

cpohle1983, 2022-08-30:
I can confirm that with moving this line up (in order) has fixed the issue!

THX

closed 2022-08-31 by fredlcore