fredlcore/BSB-LAN #477: [BUG] Compile error Arduino Due

chalojak, 2022-08-30:
**BSB-LAN Version**
2.1   - 30.07.2022

**Architecture**
Arduino Due

**Bus system**
NA

**Describe the bug**
Arduino does compile the code and returns an error, see below.

**Logs**
`Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "Arduino Due (Programming Port)"

In file included from C:\Users\jakub.chaloupka\Downloads\BSB-LAN-master\BSB_LAN\BSB_LAN.ino:1202:0:

C:\Users\jakub.chaloupka\Downloads\BSB-LAN-master\BSB_LAN\include\get_cmdtbl_values.h: In function 'float get_next_prognr(float, int)':

C:\Users\jakub.chaloupka\Downloads\BSB-LAN-master\BSB_LAN\include\get_cmdtbl_values.h:59:33: error: cannot convert 'float*' to 'double*' for argument '2' to 'double modf(double, double*)'

     modf(currentProgNR, &intpart);

                                 ^

In file included from C:\Users\jakub.chaloupka\Downloads\BSB-LAN-master\BSB_LAN\BSB_LAN.ino:1202:0:

C:\Users\jakub.chaloupka\Downloads\BSB-LAN-master\BSB_LAN\include\get_cmdtbl_values.h:78:27: error: cannot convert 'float*' to 'double*' for argument '2' to 'double modf(double, double*)'

     modf(prognr, &intpart1);

                           ^

C:\Users\jakub.chaloupka\Downloads\BSB-LAN-master\BSB_LAN\include\get_cmdtbl_values.h:79:31: error: cannot convert 'float*' to 'double*' for argument '2' to 'double modf(double, double*)'

     modf(nextprognr, &intpart2);

                               ^

C:\Users\jakub.chaloupka\Downloads\BSB-LAN-master\BSB_LAN\include\get_cmdtbl_values.h:85:37: error: cannot convert 'float*' to 'double*' for argument '2' to 'double modf(double, double*)'

         modf(currentProgNR, &intpart);

                                     ^

C:\Users\jakub.chaloupka\Downloads\BSB-LAN-master\BSB_LAN\BSB_LAN.ino: In function 'int findLine(float, uint16_t, uint32_t*)':

BSB_LAN:1280:49: error: cannot convert 'float*' to 'double*' for argument '2' to 'double modf(double, double*)'

           line = BSP_DHT22 + modf(line, &intpart);

                                                 ^

BSB_LAN:1284:71: error: cannot convert 'float*' to 'double*' for argument '2' to 'double modf(double, double*)'

       case 4: {float intpart; line = BSP_ONEWIRE + modf(line, &intpart); break;}

                                                                       ^

BSB_LAN:1290:47: error: cannot convert 'float*' to 'double*' for argument '2' to 'double modf(double, double*)'

           line = BSP_MAX + modf(line, &intpart);

                                               ^

BSB_LAN:1300:50: error: cannot convert 'float*' to 'double*' for argument '2' to 'double modf(double, double*)'

           line = BSP_BME280 + modf(line, &intpart);

                                                  ^

exit status 1

cannot convert 'float*' to 'double*' for argument '2' to 'double modf(double, double*)'`

**Additional context**
Add any other context about the problem here.

dukess, 2022-08-30:
As a temporary workaround you can change type of intpart, intpart1 and intpart2 from float to double.

fredlcore, 2022-08-30:
Sorry, this is a known bug currently, and we're working on it to fix it. Please use the 2.0 release version for the moment.

fredlcore, 2022-08-30:
Hm, strange, I just tried to compile for the Due (with VSCode) and standard _config.h and it compiled without errors. Did you change something already, @dukess?

dukess, 2022-08-30:
Nope. This may be a specificity of Arduino Studio compiler/library.

chalojak, 2022-08-30:
I confirm that I could compile the v2.0 without an error, but
it occurred in the v2.1 in the same compiler.

fredlcore, 2022-08-30:
Yes, we have added a function call that works perfectly fine in our development environment, but we're not using Arduino IDE. The cmath library looks the same on both environments, so it must be compiler specific.
@dukess: Is it safe to just change all intpart* variables from float to double? Then I can do that quickly.

dukess, 2022-08-30:
We can add definitions:
```
  #if defined(__SAM3X8E__)
    double intpart;
  #else
    float intpart;
  #endif
```

fredlcore, 2022-08-31:
Should be fixed now with @dukess' latest changes in code. Please check if it now works, @chalojak and reopen the issue if the problem still persists.

closed 2022-08-31 by fredlcore

1coderookie, 2022-08-31:
Yepp, now it works! :)
Great - thanks!

chalojak, 2022-08-31:
Yes, I confirm that it works too. Thanks for the prompt support and amazing project!