fredlcore/BSB-LAN #485: [BUG] Multiple build issues esp32 + arduino-cli

flohoff, 2022-09-14:
**BSB-LAN Version**
#define MAJOR "2"
#define MINOR "1"
#define PATCH "8"
#define COMPILETIME "20220731102301"

flo@p5:~/projects/BSB/BSB-LAN/BSB_LAN$ git describe --long --all
tags/v2.1-0-g4c932876

I had some compile issues which seem to be incompatibilities in the BSB-LAN and the respective libraries.

I am building for esp32 with arduino-cli

arduino-cli compile -b esp32:esp32:esp32  -v

**Issue 1**
```

Using previously compiled file: /tmp/arduino-sketch-5E3C761432E44D618AB643A1D45E0C98/sketch/src/WiFiSpi/src/utility/srvspi_drv.cpp.o
In file included from /home/flo/projects/BSB/BSB-LAN/BSB_LAN/src/WiFiSpi/src/utility/espspi_proxy.cpp:21:0:
/home/flo/projects/BSB/BSB-LAN/BSB_LAN/src/WiFiSpi/src/utility/espspi_proxy.h: In member function 'void EspSpiProxy::_pulseSS(bool)':
/home/flo/projects/BSB/BSB-LAN/BSB_LAN/src/WiFiSpi/src/utility/espspi_proxy.h:95:39: error: 'LOW' was not declared in this scope

```

espspi_proxy.cpp uses LOW and HIGH without including Arduino.h
```

diff --git a/BSB_LAN/src/WiFiSpi/src/utility/espspi_proxy.h b/BSB_LAN/src/WiFiSpi/src/utility/espspi_proxy.h
index 46c70aa3..ef9e3b9b 100755
--- a/BSB_LAN/src/WiFiSpi/src/utility/espspi_proxy.h
+++ b/BSB_LAN/src/WiFiSpi/src/utility/espspi_proxy.h
@@ -41,6 +41,7 @@
 
 #include "../config.h"
 #include <SPI.h>
+#include <Arduino.h>
 #include "debug.h"
 
 // The command codes are fixed by ESP8266 hardware

```

**Issue 2**
```

Using previously compiled file: /tmp/arduino-sketch-5E3C761432E44D618AB643A1D45E0C98/sketch/src/WiFiSpi/src/utility/wifispi_drv.cpp.o
/home/flo/projects/BSB/BSB-LAN/BSB_LAN/src/WiFiSpi/src/WiFiSpiClient.cpp: In member function 'IPAddress WiFiSpiClient::remoteIP()':
/home/flo/projects/BSB/BSB-LAN/BSB_LAN/src/WiFiSpi/src/WiFiSpiClient.cpp:249:29: error: call of overloaded 'IPAddress(long unsigned int)' is ambiguous
         return IPAddress(0UL);
```

Initializing with 0UL (unsigned long) does not work as there is only a method for uint32_t. 

```
diff --git a/BSB_LAN/src/WiFiSpi/src/WiFiSpiClient.cpp b/BSB_LAN/src/WiFiSpi/src/WiFiSpiClient.cpp
index 4eee7f62..7c048f9a 100755
--- a/BSB_LAN/src/WiFiSpi/src/WiFiSpiClient.cpp
+++ b/BSB_LAN/src/WiFiSpi/src/WiFiSpiClient.cpp
@@ -246,7 +246,7 @@ IPAddress WiFiSpiClient::remoteIP()
     if (WiFiSpiDrv::getRemoteData(_sock, _remoteIp, &_remotePort))
         return IPAddress(_remoteIp);
     else
-        return IPAddress(0UL);
+        return IPAddress((uint32_t) 0);
 }
 
 /*
diff --git a/BSB_LAN/src/WiFiSpi/src/WiFiSpiUdp.cpp b/BSB_LAN/src/WiFiSpi/src/WiFiSpiUdp.cpp
index a4ff26f0..779eeb38 100755
--- a/BSB_LAN/src/WiFiSpi/src/WiFiSpiUdp.cpp
+++ b/BSB_LAN/src/WiFiSpi/src/WiFiSpiUdp.cpp
@@ -216,7 +216,7 @@ IPAddress WiFiSpiUdp::remoteIP()
     if (WiFiSpiDrv::getRemoteData(_sock, _remoteIp, &_remotePort))
         return IPAddress(_remoteIp);
     else
-        return IPAddress(0UL);
+        return IPAddress((uint32_t) 0);
 }
 
 /*
```

**Issue 3**
```

/home/flo/projects/BSB/BSB-LAN/BSB_LAN/src/Time/examples/Processing/SyncArduinoClock/SyncArduinoClock.pde
/home/flo/projects/BSB/BSB-LAN/BSB_LAN/BSB_LAN.ino:520:49: error: invalid conversion from 'const char*' to 'uint32_t {aka unsigned int}' [-fpermissive]
 EEPROMClass EEPROM_ESP((const char *)PSTR("nvs"));
                                                 ^
In file included from /home/flo/projects/BSB/BSB-LAN/BSB_LAN/BSB_LAN.ino:511:0:
/home/flo/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/EEPROM/src/EEPROM.h:37:5: note:   initializing argument 1 of 'EEPROMClass::EEPROMClass(uint32_t)'
     EEPROMClass(uint32_t sector);
```

For obvious reasons this does not compile. But i have no clue what the correct argument would be. I need to dig in further. Just getting past this by passing a 1 results in an image, although the EEPROM issue might break (Not tested)

1coderookie, 2022-09-15:
Hi,
I can't say anything about the third issue, but the first two issues _may_ be caused by not (re)moving the WiFiSpi folder from src before compiling for ESP32..!?! 
See step 5 in the manual:
`5. Remove (or move) the two folders “ArduinoMDNS” and “WiFiSpi” from the BSB-LAN subfolder “src” - these must not be present in the “BSB-LAN” or “src” folder!`

fredlcore, 2022-09-15:
Exactly. Reading the manual would have helped in all three cases, the third one is caused by not using the ESP framwork 2.0 or above, because it says 1.0.6 here.

closed 2022-09-15 by fredlcore

fredlcore, 2022-09-15:
@1coderookie: I know it's even mentioned in the Quick Start section of the manual, but maybe we should add as first entry in the FAQ the entry "Compilation fails" and list a few of the most common problems there (like the two that occurred here)?

For example like this: 
The error mentions something about "WiFiSPI"? If on ESP32, remove the WiFiSPI folder.
The error mentions something about "ArduinoMDNS"? If on ESP32, remove the ArduinoMDNS folder.
The error mentions something about "EEPROMClass"? Make sure you have the correct ESP32 framework installed (1.0.6 is too old).

1coderookie, 2022-09-15:
Yepp, I just thought the same, I'll send you a mail.. ;)