fredlcore/BSB-LAN #493: [FEATURE REQUEST] MQTT: port setting, username/password length, domain name as server address

jinx17, 2022-10-07:
Hi, 
I think these additional settings should be implemented for MQTT

- MQTT TCP port setting, should also be available (like "7883")
- username and password for MQTT should be larger. Some brokers (commercial) provide tokens that are more than 80 characters long. But probably there is also a request to be able to establish secure (SSL/TLS) connection. Maybe / Probably this is out of scope for this device.
- instead of IP address for MQTT broker there should also be an option to insert server by domain name, like for instance: "http://mqtt.flespi.io/"

Maybe all of these parameters could be settable from WEB?

Best regards,
Gregor

fredlcore, 2022-12-09:
I think setting the port could be done in a way like web browsers do it, i.e. with a colon after the IP.
Longer variables can simply done by the user by changing the respective variable in BSB_LAN_config.h.
DNS resolution would certainly be necessary if users want to use MQTT broker outside the home network. Will have to look into that if that's easily doable.

juergen852, 2023-12-05:
I would also appreciate DNS resolution.

fredlcore, 2024-02-26:
I have just added functionality for using hostnames (IP addresses are still possible, too) and an optional port number that can be added after a trailing colon. For example:
192.168.1.20:1884
broker.local:1884
work now. If you omit the port number and the colon, the default port 1883 will be used.

closed 2024-02-26 by fredlcore