fredlcore/BSB-LAN #494: [FEATURE REQUEST] MQTT - Enable secure (SSL/TLS) connection

jinx17, 2022-10-07:
For commercial or professional MQTT brokers, a user has to connect with secure (SSL/TLS) connection.
Maybe this should be implemented.
Best regards,
Gregor

fredlcore, 2022-12-09:
I think that's a good idea and using [WifiClientSecure](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/examples/WiFiClientInsecure/WiFiClientInsecure.ino) it would be possible even without installing certificates on the ESP32 (albeit with losing CA certificate checks). However, it's still a complex matter because local installations of MQTT brokers usually do not use SSL, so we would have to differentiate (or make it configurable) which service should use SSL or not. 
Any ideas, @dukess?

dukess, 2022-12-18:
@fredlcore we need man with skills for MQTT.
In theory we can allcate memory for certificate and add configuration options (cert, key and ca cert), but it would work on ESP32 only?

fredlcore, 2022-12-18:
Yes, it would work on ESP32 only, but that would be ok for me. We shouldn't deny security to those users where this would be possible, I think. But the problem I see is more about the cert management. In case someone wanted to send their MQTT requests to some external site, we would have to provide the option to store the root certificate of that site on the ESP32, but then while we're at it, we should also provide the functionality for users to store their own certificate on the ESP32 to access the ESP32 via SSL. Or is this something completely different (or impossible)?

fredlcore, 2022-12-29:
@jinx17: Would you be willing to do some testing before we start implementing this any further?
Which cloud MQTT broker do you use? 
Taking HiveMQ as an example, please try to make these changes and see if it works:
In BSB_LAN.ino, change this line
```
ComClient *mqtt_client;   //Luposoft: own instance
```
to these lines:
```
#include <WiFiClientSecure.h>
WiFiClientSecure *mqtt_client;   //Luposoft: own instance
```
In mqtt_handler.h, change this line
```
    MQTTPubSubClient->setServer("MQTTBroker", 1883);
```
to this (assuming you are using HiveMQ on port 8883:
```
    MQTTPubSubClient->setServer("SOME_LONG_STRING.hivemq.cloud", 8883);
```
and this line
```
    mqtt_client= new ComClient();
```
to these lines:
```
    mqtt_client= new WiFiClientSecure();
    mqtt_client.setInsecure();//skip verification
```

Let me know if you have any luck with that. If it works, it would still only be working using WiFi because the WiFiClientSecure library only works on WiFi (and only on ESP32). We would then have to think about how to make this available for Ethernet users - or not. But first things first...

jinx17, 2022-12-30:
Hi,
I am willing to test, but at the momemt I am using board "in production" so I can not test it. I am not able to do this until the end of heating season.
Othervise, I am using flespi mqtt broker (free option), with TLS ...

fredlcore, 2022-12-30:
Well, this shouldn't hurt your production system, worst case is you lose a few logging entries as you can always revert back to the previous version and everything should be as it was before...

fredlcore, 2024-02-26:
No feedback for over a year, so I'm closing this for now...

closed 2024-02-26 by fredlcore