fredlcore/BSB-LAN #495: [BUG] compilation stopps with error: expected unqualified-id before '{' token

jinx17, 2022-10-07:
**Before submitting a bug report, please make sure that you have checked [chapter 14](https://1coderookie.github.io/BSB-LPB-LAN_EN/chap14.html) of the manual** ("Problems and their Possible Causes"). Most problems that are reported to us can be solved by going through the solutions provided there.

**BSB-LAN Version**
2.1.8

**Architecture**
ESP32 Olimex EVB

**Bus system**
Which bus BSB-LAN is running on: BSB

**Describe the bug**
Trying to compile - gives me error:
In file included from C:\BSB-LAN-master 2.1.8\BSB_LAN\BSB_LAN.ino:482:
C:\BSB-LAN-master 2.1.8\BSB_LAN\BSB_LAN_defs.h:1131:1: error: expected unqualified-id before '{' token
C:\BSB-LAN-master 2.1.8\BSB_LAN\BSB_LAN_defs.h:1131:118: error: expected unqualified-id before ',' token
...
C:\BSB-LAN-master 2.1.8\BSB_LAN\BSB_LAN_defs.h:1251:1: error: expected declaration before '}' token
exit status 1
Compilation error: expected unqualified-id before '{' token

**To Reproduce**
compile BSB-LAN

**Logs**

**Expected behavior**
normal compilation withou error

**Additional context**

LOOKS LIKE SOME CONSTANT DEFINITION IS MISSING IN FILE  
**BSB_LAN_defs.h** 

(in line 1130) - before JSON definitions:
**{0x2D490000,  VT_ENUM,          15000, STR15000, sizeof(ENUM15000),    ENUM15000,    DEFAULT_FLAG+FL_EEPROM, DEV_ALL}, // Betriebsart**
...

fredlcore, 2022-10-07:
Have you renamed "BSB_LAN_custom_defs.h.default" to  "BSB_LAN_custom_defs.h"?

jinx17, 2022-10-07:
Hi,
I just solved a mistake I made. Everything is working OK.
1st I didn't rename custom.defs and got some errors, therefore I simply commented out   #include "BSB_LAN_custom_defs.h". 
After that I did rename custom.defs, but forgot to uncomment #include "BSB_LAN_custom_defs.h", which caused this problem.
Thank you, and I am sorry for inconvenience.
Best regards.

closed 2022-10-07 by jinx17