fredlcore/BSB-LAN #506: [BUG] MQTT sensor publish dont work

nolsnols, 2022-10-20:
**Before submitting a bug report, please make sure that you have checked [chapter 14](https://1coderookie.github.io/BSB-LPB-LAN_EN/chap14.html) of the manual** ("Problems and their Possible Causes"). Most problems that are reported to us can be solved by going through the solutions provided there.

**BSB-LAN Version**
BSB-LAN, Version 2.1.8-20220731102301

**Architecture**
ESP32 Olimex EVB

**Bus system**
LPB

**Describe the bug**
I cant publish the sensor numbers with . (point) in the name. If I edit it i the BSB_LAN_config.h i get this error while compiliing:

[code]In file included from /Users/nols/Documents/BSB-Lan/2.1.8/BSB-LAN-master/BSB_LAN/BSB_LAN.ino:477:
BSB_LAN_config.h:192:1: error: narrowing conversion of '2.0200099999999999e+4' from 'double' to 'int' inside { } [-Wnarrowing]
 };
 ^
BSB_LAN_config.h:192:1: error: narrowing conversion of '2.0200200000000001e+4' from 'double' to 'int' inside { } [-Wnarrowing]
BSB_LAN_config.h:192:1: error: narrowing conversion of '2.0200299999999999e+4' from 'double' to 'int' inside { } [-Wnarrowing]
BSB_LAN_config.h:192:1: error: narrowing conversion of '2.0200400000000001e+4' from 'double' to 'int' inside { } [-Wnarrowing]
Mehrere Bibliotheken wurden für "WiFiUdp.h" gefunden
 Benutzt: /Users/nols/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/libraries/WiFi
 Nicht benutzt: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
narrowing conversion of '2.0200099999999999e+4' from 'double' to 'int' inside { } [-Wnarrowing][/code]

In the webinterface the sensor parameters are numbered like this:
20200.0 One Wire, DHT & MAX! Sensors - BME280 Sensor ID #1: 77		
20200.1 One Wire, DHT & MAX! Sensors - BME280 Sensor Temperatur #1: 21.28 °C		
20200.2 One Wire, DHT & MAX! Sensors - BME280 Sensor Luftfeuchtigkeit #1: 59.33 %		
20200.3 One Wire, DHT & MAX! Sensors - BME280 Sensor Pressure #1: 1014.01 hPa		
20200.4 One Wire, DHT & MAX! Sensors - BME280 Sensor Altitude #1: -8.41 m

fredlcore, 2022-10-20:
@dukess, do you have any idea why this happns?

fredlcore, 2022-10-20:
@nolsnols, please post your BSB_LAN_config.h that causes this error here (using code tags "<>" above)

dukess, 2022-10-20:
Hi Frederik,
First though: nolsnols use old config file with int definitions.

nolsnols, 2022-10-21:
> […](#)

Thts it. Thanks, i create new config file and now it works. Thanks!

closed 2022-10-21 by nolsnols