fredlcore/BSB-LAN #507: [BUG] logging unknown parameter crashes system

DE-cr, 2022-10-30:
**BSB-LAN Version**
Current (30Oct2022) GitHub master sources

**Architecture**
ESP32 NodeMCU

**Bus system**
BSB

**Describe the bug**
Trying to log unknown parameters resets the system (and does not log values).

**To Reproduce**
I re-enabled "SD card" logging of parameters I had logged before. When the time came to log their values, the system went through a reset.
The problem seems to be that I'm now running BSB-LAN with a tailored BSB_LAN_custom_defs.h, which does not contain parameter 8760 anymore (which did work before the custom_defs). Including this parameter in logging leads to the reset on my system.

fredlcore, 2022-10-31:
Thanks for reporting this, I had this problem elsewhere and fixed it there, but I'm currently away from home, so I can't test this:
Could you please make these changes at around line 6646 in BSB_LAN.ino:
Instead of 
```
          query(log_parameters[i]);
          outBufLen = 0;
```
if should be 
```
          query(log_parameters[i]);
          if (decodedTelegram.prognr < 0) continue;
          outBufLen = 0;
```
This should skip non-existing parameters.

fredlcore, 2022-11-01:
Thanks, I've taken up your fix as I still haven't decided about your SD card PR, so the others don't have to live with this error until I've had time to deal with this.