fredlcore/BSB-LAN #508: Pin assignment on Olimex ESP32-EVB-EA

todor-dk, 2022-10-31:
Can somebody please confirm that the adapter board uses the following pins on the **Olimex ESP32-EVB-EA** boards:
- UEXT1 Pin 1 - 3,3V 
- UEXT1 Pin 2 - Gnd
- UEXT1 Pin 3 - UART1 TXD - GPIO4
- UEXT1 Pin 4 - UART1 RXD - GPIO36

See: https://i.stack.imgur.com/z0owu.png

The reason I am asking; I am looking for the best available/free pin on the Olimex to connect OneWire DS18B20 probes.

fredlcore, 2022-11-01:
This is true for the ESP32 BSB-LAN board version 4.1. In 4.2 we changed that due to the fact that pulling the RX pin high (or low?) caused boot problems of the Olimex EVB. It also turned out that the Olimex PoE's UEXT pin 10 is connected to a different GPIO (17 on the EVB, 5 on the PoE).
So these are the GPIOs used in the different board versions used for RX: 
4.1: GPIO 36 / UEXT1 pin 4
4.2 EVB: 17 / UEXT1 pin 10
4.2 PoE: 5 / UEXT1 pin 5

So on BSB-LAN board version 4.2, the following UEXT1 pins should be usable on an Olimex EVB:
UEXT1 pin 4 / GPIO36 (readonly)
UEXT1 pin 5 / GPIO16 (usually used for I2C)
UEXT1 pin 6 / GPIO13 (usually used for I2C)

UEXT1 pins 7-9 are connected to the SD card reader on the EVB, so it may be possible to use them if you don't use the SD card.

Since none of us has really tried using external sensors on the EVB, it would be great if you could let us know which pins can safely be used for sensors such as the DS18B20, then we could publish that in the manual as well.

fredlcore, 2022-11-01:
As for which pins need to be or are pulled high/low during boot, see the lower right corner of this schematic:
https://raw.githubusercontent.com/OLIMEX/ESP32-EVB/master/HARDWARE/REV-I/ESP32-EVB_Rev_I.pdf

closed 2022-11-01 by fredlcore