fredlcore/BSB-LAN #509: [BUG] Update from 2.1.0 to 2.1.8-20220731102301

proasnet, 2022-11-01:
**Before submitting a bug report, please make sure that you have checked [chapter 14](https://1coderookie.github.io/BSB-LPB-LAN_EN/chap14.html) of the manual** ("Problems and their Possible Causes"). Most problems that are reported to us can be solved by going through the solutions provided there.

**BSB-LAN Version**
2.1.8-20220731102301

**Architecture**
Arduino Due

**Bus system**
BSB

**Describe the bug**
Today after update, BSB not starting. In a serial log ending by this sample:
```
Size of cmdtbl: 126728
free RAM: 85403
BSB-LAN version: 2.1.8-20220731102301
2.1.8-20220731102301
Using RX/TX pins 18, 19
PPS settings:
Starting SD..ok
Waiting for DHCP address
192.168.0.150
255.255.255.0
192.168.0.254
Waiting 3 seconds to give Ethernet shield time to get ready...
Calculating free space on SD...14889 MB free
Start network services
```
In web viewer is not site loaded.
BSB can ping

**Logs**
```
READY
EEPROM not ready
Reading EEPROM
Using settings from config file
Reading done.
EEPROM schema v.0 Program schema v.10
Update EEPROM schema
Address EEPROM option 0: 0
Address EEPROM option 1: 1
Address EEPROM option 2: 2
Address EEPROM option 3: 6
Address EEPROM option 4: 226
Address EEPROM option 5: 306
Address EEPROM option 6: 490
Address EEPROM option 7: 491
Address EEPROM option 8: 492
Address EEPROM option 9: 493
Address EEPROM option 10: 494
Address EEPROM option 11: 495
Address EEPROM option 12: 499
Address EEPROM option 13: 659
Address EEPROM option 14: 663
Address EEPROM option 15: 823
Address EEPROM option 16: 829
Address EEPROM option 17: 830
Address EEPROM option 18: 834
Address EEPROM option 19: 838
Address EEPROM option 20: 842
Address EEPROM option 21: 846
Address EEPROM option 22: 848
Address EEPROM option 23: 852
Address EEPROM option 24: 856
Address EEPROM option 25: 920
Address EEPROM option 26: 984
Address EEPROM option 27: 985
Address EEPROM option 28: 986
Address EEPROM option 29: 996
Address EEPROM option 30: 997
Address EEPROM option 31: 1157
Address EEPROM option 32: 1158
Address EEPROM option 33: 1162
Address EEPROM option 34: 1163
Address EEPROM option 35: 1164
Address EEPROM option 36: 1165
Address EEPROM option 37: 1169
Address EEPROM option 38: 1201
Address EEPROM option 39: 1233
Address EEPROM option 40: 1265
Address EEPROM option 41: 1297
Address EEPROM option 42: 1299
Address EEPROM option 43: 1300
Address EEPROM option 44: 1301
Address EEPROM option 45: 1302
Address EEPROM option 46: 1334
Address EEPROM option 47: 1398
Address EEPROM option 48: 1418
Address EEPROM option 49: 1438
Address EEPROM option 50: 1458
Address EEPROM option 51: 1459
Address EEPROM option 52: 1460
Address EEPROM option 53: 1461
Address EEPROM option 54: 1462
Address EEPROM option 55: 1463
Address EEPROM option 56: 1464
Address EEPROM option 57: 1496
Address EEPROM option 58: 1497
Address EEPROM option 59: 1498
Address EEPROM option 60: 1499
Address EEPROM option 61: 1501
Address EEPROM option 62: 1503
Address EEPROM option 63: 1504
Size of cmdtbl: 126728
free RAM: 85403
BSB-LAN version: 2.1.8-20220731102301
2.1.8-20220731102301
Using RX/TX pins 18, 19
PPS settings:
Starting SD..ok
Waiting for DHCP address
192.168.0.150
255.255.255.0
192.168.0.254
Waiting 3 seconds to give Ethernet shield time to get ready...
Calculating free space on SD...14889 MB free
Start network services
```

fredlcore, 2022-11-01:
Please also post your BSB_LAN_config.h file contents here (remove credentials if necessary)

proasnet, 2022-11-01:
After change pins rx tx back to 0,0, from 19,18, device working.
In previous version, I had 19,18.

fredlcore, 2022-11-01:
Great to hear it works then.

closed 2022-11-01 by fredlcore