fredlcore/BSB-LAN #511: [BUG] MicroSD format

proasnet, 2022-11-07:
Please, may be specified a format of a MicroSD card for version 3?
In my Due, format of previous version 2.8 not working.
My SD card is hard to access and I want to avoid experimenting.
After upgrading to 3.0, when commanding D0, BSB LAN displays the following message:
, Failed to create datalog.txt!

Thank you very much.

fredlcore, 2022-11-07:
Hm, we had to change the SD card library from version 1 to version 2 because the old one was no longer compiling properly with recent SDK, but it should be mostly compatible. Especially the format should be the same (FAT, exFAT).
@dukess, do you still have a Due with an SD card running and could check on this? That would be great!

proasnet, 2022-11-07:
Now, I removed card, slow format as FAT32, but card not working in BSB LAN.
Before formating, I found on card both txt files, but end time of write was from my previous version 2.8.1
I dont know, where is problem.
Card has size 16GB.

proasnet, 2022-11-07:
Another 16GB card has the same problem.

fredlcore, 2022-11-07:
No need to test further, there must be an issue in the code.

fredlcore, 2022-11-07:
No.

fredlcore, 2022-11-07:
@dukess: No need to test anymore, I have found and connected my Due including an SD card and it works perfectly fine - at least with my SD card.
@proasnet: The problem seems to be in your setup. Can you downgrade to release version 2.2 and check if the problem is solved then? If not, then it's obvious that it's your card or something related to the hardware because we only changed to the new library a few days ago. If it does work with version 2.2, then I'll have to look deeper...

proasnet, 2022-11-08:
Okay, today after I return from work.
Another anomaly in 3.0 is if I connect to BSB LAN over web, is this first page:
```
Error opening datalog.txt!
Error opening datalog.txt!
Error opening datalog.txt!
Error opening datalog.txt!
Error opening datalog.txt!
Error opening datalog.txt!
Error opening datalog.txt!
Error opening datalog.txt!
Error opening datalog.txt!
Error opening datalog.txt!
Error opening datalog.txt!
HTTP/1.1 200 OK
Content-Type: text/html; charset=utf-8
Cache-Control: no-cache, no-store, must-revalidate
Pragma: no-cache
Expires: 0

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01//EN"><html><head><meta http-equiv="content-type" content="text/html; charset=UTF-8"><meta name="viewport" content="width=device-width, initial-scale=1.0,  minimum-scale=1.0"><link rel="icon" type="image/svg+xml" href="/favicon.svg"><title>BSB-LAN</title><style>A:link  {color:blue;text-decoration: none;} A:visited {color:blue;text-decoration: none;} A:hover {color:red;text-decoration: none;background-color:yellow} A:active {color:blue;text-decoration: none;} A:focus {color:red;text-decoration: none;}a.logo:hover {background-color: initial; text-decoration: initial;}table {width: 100%;  max-width: 1024px;  margin: auto;}td {vertical-align: top;}td.header {vertical-align: middle;}svg.logo {width: 10%; height: 10%}input {width: 100%; box-sizing: border-box;} select {width: 100%;}</style></head><body><script>function set(line){var value = document.getElementById('value'+line).value.replace(/\.$/, '');value = value.replaceAll(' ', '_');value = value.replace('---', '');window.open(document.getElementById('main_link').href+'S'+line+'='+value,'_self');}function setbit(line){var x=document.getElementById('value'+line); var value=0;for (var i=0; i<x.options.length; i++) {if(x.options[i].selected){value=value+eval(x.options[i].value);}}window.open(document.getElementById('main_link').href+'S'+line+'='+value,'_self');}</script><font face='Arial'><center><a href='/' ID=main_link class='logo'><svg class="logo" version="1.0" xmlns="http://www.w3.org/2000/svg" width="400pt" height="400pt" viewBox="0 0 400 400" preserveAspectRatio="xMidYMid meet"><g transform="translate(0,400) scale(0.1,-0.1)"
fill="#000000" stroke="none"><path d="M292 3510 c-61 -38 -62 -42 -62 -537 0 -501 0 -500 70 -536 32 -16
69 -17 445 -15 399 3 411 4 451 25 102 54 151 156 141 293 -6 92 -35 164 -95
238 -37 46 -41 55 -31 74 23 43 39 124 39 203 0 97 -27 165 -86 213 -70 58
-100 62 -488 62 -334 0 -353 -1 -384 -20z m708 -275 c0 -60 -21 -98 -64 -119
-28 -13 -70 -16 -235 -16 l-201 0 0 -125 0 -125 246 0 c203 0 252 -3 279 -16
40 -19 65 -66 65 -123 l0 -41 -305 0 -305 0 0 305 0 305 260 0 260 0 0 -45z"/><path d="M1705 3516 c-99 -31 -171 -91 -210 -172 -73 -155 -17 -359 122 -441
77 -45 136 -53 406 -53 l248 0 24 -25 c35 -34 35 -96 0 -130 l-24 -25 -401 0
-400 0 0 -126 0 -125 418 3 c407 3 418 4 473 26 77 31 142 90 176 160 24 50
27 69 28 152 0 84 -3 102 -28 152 -33 67 -90 121 -166 157 -55 26 -59 26 -338
31 -325 6 -318 4 -318 91 0 41 4 54 24 70 22 18 44 19 377 19 l354 0 0 125 0
125 -362 -1 c-264 0 -374 -4 -403 -13z"/><path d="M2792 3510 c-61 -38 -62 -42 -62 -537 0 -501 0 -500 70 -536 32 -16
69 -17 445 -15 399 3 411 4 451 25 102 54 151 156 141 293 -6 92 -35 164 -95
238 -37 46 -41 55 -31 74 23 43 39 124 39 203 0 97 -27 165 -86 213 -70 58
-100 62 -488 62 -334 0 -353 -1 -384 -20z m708 -275 c0 -60 -21 -98 -64 -119
-28 -13 -70 -16 -235 -16 l-201 0 0 -125 0 -125 246 0 c203 0 252 -3 279 -16
40 -19 65 -66 65 -123 l0 -41 -305 0 -305 0 0 305 0 305 260 0 260 0 0 -45z"/><path d="M1932 2152 c-138 -127 -300 -298 -376 -400 -82 -108 -178 -298 -211
-416 -21 -73 -25 -110 -25 -215 0 -201 50 -337 170 -465 59 -63 198 -172 207
-162 2 2 -7 28 -20 59 -34 75 -47 143 -47 242 l0 85 300 0 300 0 0 -82 c0 -99
-21 -206 -53 -270 l-24 -47 49 26 c105 57 218 175 275 287 86 171 84 414 -7
585 -47 88 -132 195 -255 321 -62 63 -122 133 -133 155 -47 94 -70 236 -54
333 5 28 7 52 5 52 -2 0 -47 -40 -101 -88z m45 -772 c46 -34 139 -152 172
-217 l21 -43 -240 0 -240 0 11 27 c30 80 185 263 223 263 7 0 31 -14 53 -30z"/><path d="M2735 1598 c-55 -43 -55 -44 -55 -593 l0 -505 125 0 125 0 0 348 0
347 123 -125 c67 -69 219 -226 337 -349 l215 -223 51 -5 c48 -5 54 -3 90 30
l39 35 3 526 3 526 -126 0 -125 0 -2 -345 -3 -345 -237 247 c-130 137 -281
294 -335 351 l-99 102 -51 0 c-36 0 -58 -6 -78 -22z"/><path d="M212 1098 c3 -459 5 -515 20 -536 41 -59 22 -57 531 -60 l467 -3 0
126 0 125 -385 0 -385 0 0 430 0 430 -125 0 -126 0 3 -512z"/></g></svg></A></center>
<table align=center><tr bgcolor=#f0f0f0><td class="header" width=20% align=center><a href='/K'>Heater functions</a></td><td class="header" width=20% align=center><a href='/K8'>Sensors</a></td><td class="header" width=20% align=center><a href='/DG'>Plot log file</a></td><td class="header" width=20% align=center><a href='/Q' TARGET='_new'>Device-specific parameter list</a></td></tr>
<tr bgcolor=#f0f0f0><td class="header" width=20% align=center><a href='/C'>Settings</a></td><td class="header" width=20% align=center><a href='https://1coderookie.github.io/BSB-LPB-LAN_EN/chap05.html#51-url-commands' target='_new'>URL commands</a></td><td class="header" width=20% align=center><a href='https://1coderookie.github.io/BSB-LPB-LAN_EN/toc.html' target='new'>Manual</a></td><td class="header" width=20% align=center><a href='https://1coderookie.github.io/BSB-LPB-LAN_EN/chap15.html' target='_new'>FAQ</a></td></tr></table><p></p><table align=center><tr><td class="header">
<p>
[... cut ...]
<p><b>Settings:</b> Displays a list of configuration options. You can change most of these by using the extended URL commands.
<p><b>URL commands:</b> Displays a list of extended commands which you can access by directly entering them in the browser's address line. These commands are also necessary to link up BSB-LAN to any home automation system such as FHEM.
<BR><BR>Checking for newer version...<BR>
Current version is up to date
</td></tr></table>
</body>
</html>
Then after refresh, I see page normally.
```

proasnet, 2022-11-08:
May be, that this text
```
Error opening datalog.txt!
Error opening datalog.txt!
Error opening datalog.txt!
Error opening datalog.txt!
Error opening datalog.txt!
Error opening datalog.txt!
Error opening datalog.txt!
Error opening datalog.txt!
Error opening datalog.txt!
Error opening datalog.txt!
Error opening datalog.txt!
```
Is pasted to html code and page is not displayed in graphic

1coderookie, 2022-11-08:
> In my Due, format of previous version **2.8** not working.

and

> Before formating, I found on card both txt files, but end time of write was from my previous version **2.8.1** 

I'm just curious: the last version of BSB-LAN before v3.x was v2.2 - where does that 2.8/2.8.1 come from..?

And: did you use the _complete_ version of 2.2/3.x when down-/upgrading (including the belonging config.h, librarires and so on), or did you just use certain files of that version..?

proasnet, 2022-11-08:
I am sorry, I tried some upgrades with their edited config files.
My last long working version was 2.1.0
Then I upgrade to 2.1.8, where I had problems with start after edit config to pins 19 and 18. But card work. If I had 0,0 pins, then this work about one week. Then after load web setting, and add some new parameters for logging ( from default ), my device stop working, and I cant start it. Every reboot waited for 
`Start network services`
At this weekend, I tried reflash with the same 2.1.8 but still not start setup.
My next step was upgrade to 3.0 and device starting normally.
But not with SD card.

1coderookie, 2022-11-08:
> If I had 0,0 pins, then this work about one week. Then after load web setting, and **add some new parameters for logging ( from default ), my device stop working, and I cant start it.** Every reboot [...]

Did you use the controller specific custom_defs.h for that which you had to ask @fredlcore for?
And did you check before adding those parameters to the logging function, if they were available using the webinterface _within that version_?
 
There was some kind of a bug about logging unsupported parameters and with the customized custom_defs.h some parameters aren't available anymore which where available with the 'complete' defs.h, so maybe it has something to do with that..? 

Cuz usually there's no need to change the pins anyway..

1coderookie, 2022-11-08:
This was the bugreport I meant: https://github.com/fredlcore/BSB-LAN/issues/507

proasnet, 2022-11-08:
custom_defs.h is empty. Without some code.
I added this parameters: 8740,8310 and I remove 8743
8700 and 8314 I left
This editing only by web UI

1coderookie, 2022-11-08:
> custom_defs.h is empty. Without some code.

custom_DEFS.h, not custom_code.h! The custom_defs.h has to be generated by @fredlcore for your specific controller after you did /Q again: https://1coderookie.github.io/BSB-LPB-LAN_EN/chap03.html#33-create-device-specific-parameter-list
That's a new function!

> I added this parameters: 8740,8310 and I remove 8743 8700 and 8314 I left This editing only by web UI

So, my advice would be: Remove the card, completey flash the recent version of BSB-LAN with RX/TX set as default (0,0), get your specific custom_defs.h, flash that one again and the do the query of the mentioned parameters using the webUI. If they are available, the you can add them to your logging list.

For the problem with the card, I'd try it again after the above mentioned steps and if the system is running then. If that still doesn't work, please report the errors&behaviour with the logs from the serial monitor again. If you do have another smaller card, maybe also try that (like 4GB or so).

1coderookie, 2022-11-08:
..oh, and maybe delete all parameters listed in the config.g besides 8700 outside temperature before you're flashing, just to make sure that there's no unsupported parameter listed for logging..

proasnet, 2022-11-08:
I am edit only BSB_LAN_config.h.
You write "get your specific custom_defs.h", where? this file has only removed "default" from end of name.
"do the query of the mentioned parameters using the webUI" - where is this function?
My mentioned parameters worked in version 2.1.0

I used to use a smaller card but it didn't work with that one
https://github.com/fredlcore/BSB-LAN/issues/336

proasnet, 2022-11-08:
I am sorry, I found this in folder scripts.
I do it in saturday morning.

1coderookie, 2022-11-08:
Please read the link to the belonging chapter of the manual in my post above about the new procedure of getting your device specific custom_defs.h.

In v2.1 we delivered the 'old&complete' defs.h named "BSB_LAN_custom_defs.h.default". That you renamed by deleting ".default". Then you had the same defs.h like the previous versions of BSB-LAN.
But: With v3.x we changed that. Since then the custom_defs.h(.default) only supports a few basic parameters. To get complete access to your controller again, you'd have to get the specific file from @fredlcore as mentioned within the chapter 3.3 (see link above).
Then it _may_ be though that certain parameters won't be available anymore which were available in the past with the 'complete' defs.h. The reason is that with the new cusomized generated custom_defs.h really _only_ the parameters will be available which are officially supported by your controller, where within the older version it sometimes happened that you had 'access' to a parameter you normally don't have, just because we got that CommandID from a different controller type and just by conincidence it also somehow worked at your controller.
We discussed this new way here:  https://github.com/fredlcore/BSB-LAN/discussions/482

1coderookie, 2022-11-08:
No problem, I know it's confusing right now when updating and not necessarily checking out the manual if anything 'big' changed and suddenly problems appear.. ;)

proasnet, 2022-11-08:
Can choosing an unsupported parameter make the sd card unreadable?
It is unclear to me that any of the mentioned parameters could be unsupported, when I see values ​​for these parameters in the web interface, and they are quite common parameters.

1coderookie, 2022-11-08:
> "do the query of the mentioned parameters using the webUI" - where is this function? 

..just the regular query /8700 for example using the webinterface..

1coderookie, 2022-11-08:
So just try to completely do a new installation of v3.0 with the steps mentioned above and within the manual like if you'd be new to this project.. Then let's see what happens.. ;)

fredlcore, 2022-11-08:
Thanks for the explanations, @1coderookie :)

1coderookie, 2022-11-08:
Welcome ;)
..I hope that'll solve the problem though..

proasnet, 2022-11-08:
Okay. Step by step with manual, as a new user. At saturday morning, where I have a free time only for me.

dukess, 2022-11-09:
Sorry for my silence, i have lot of work now.
Some time ago we did some experiments with bus frequency:
```
  if (!SD.begin(4, SPI_DIV3_SPEED)) {
    printToDebug(PSTR("failed\r\n")); // change SPI_DIV3_SPEED to SPI_HALF_SPEED if you are still having problems getting your SD card detected
  } else {
    printToDebug(PSTR("ok\r\n"));
  }
```

Maybe try different options again?

fredlcore, 2022-11-10:
Thanks, good idea! And don't worry about reply times, we all have life outside BSB-LAN :)...

proasnet, 2022-11-12:
Hi, now is time for work with BSB. I am going step by step and creating txt file.

proasnet, 2022-11-12:
Thank you. I get a file. I replaced this file and flash.
Setup now write
`Calculating free space on SD...14751 MB free`
The program seems to see the card

proasnet, 2022-11-12:
Everything seems to be working. I tried the requested parameters for logging in the URL and always returned the actual value. So I wrote them down as logging parameters and will wait a few hours to see them on the chart. Then I will write the result.
Thank you very much now.

proasnet, 2022-11-12:
Logger working fine now. Thank you.

closed 2022-11-12 by proasnet