fredlcore/BSB-LAN #513: [BUG] Home Assistant

proasnet, 2022-11-12:
**Before submitting a bug report, please make sure that you have checked [chapter 14](https://1coderookie.github.io/BSB-LPB-LAN_EN/chap14.html) of the manual** ("Problems and their Possible Causes"). Most problems that are reported to us can be solved by going through the solutions provided there.

**BSB-LAN Version**
3.0.1-20221101130752

**Architecture**
Arduino Due

**Bus system**
BSB

**Describe the bug**
Read MQTT data in Home Assistant, because integration from @liudger not working now, I tried add an easy sensor.
IP adress of broker, username and password is right.
Device ID and prefix is as default BSB-LAN

**To Reproduce**
Steps to reproduce the behavior:
1. Go to manual, chapter 8.11
2. Click on configuration.yaml and edit
3. paste
```
sensor:
  - platform: mqtt
    state_topic: "bsb-lan/8310"
    name: BSB-LAN flow temperature
    unit_of_measurement: °C
    device_class: temperature
```
This is not valid configuration.
I tried a new format
```
mqtt:
  sensor:
    - name: "BSB-LAN flow temperature"
      state_topic: "bsb-lan/8310"
      unit_of_measurement: "°C"
      device_class: temperature
```
5. See state of entity, is still unknown in both cfgs

**Logs**
LOG from MQTT Broker in HA:
`2022-11-12 15:52:18: New connection from 192.168.0.150:51415 on port 1883.`

**Additional context**
I am doing some bad, but I dont know what...

proasnet, 2022-11-12:
I tried this, and not working too.

```
rest:
  sensor:
    - name: "BSB-LAN status"
      resource: http://192.168.0.150/JQ=700,1600,8000,8003,6705,710,712,720,721,1610,1612,5400
      method: POST
      username:
      password:
      scan_interval: 60
      value_template: "{{ value_json['6705'].value }}"
      json_attributes:
        - "700"
        - "1600"
        - "6705"
        - "8000"
        - "8003"
        - "710"
        - "712"
        - "720"
        - "721"
        - "1610"
        - "1612"
        - "5400"
```

The check configuration in HA return this
`Invalid config for [rest]: [resource] is an invalid option for [rest]. Check: rest->rest->0->sensor->0->resource. (See /config/configuration.yaml, line 40). `

If I try use resource url in browser, I get this
```
<html><body>
<!--StartFragment--><li class="tabs-menu-item json is-active" role="presentation"><span class="devtools-tab-line"></span></li><li class="tabs-menu-item rawdata " role="presentation"><span class="devtools-tab-line"></span></li><span class="devtools-tab-line"></span><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"></div></div></div></div><div class="toolbar"><div class="devtools-separator"></div><div class="devtools-searchbox"><input class="searchBox devtools-filterinput" placeholder="Filtr JSON" value=""></div></div><div class="panelContent">

  |  
-- | --
700 |  
name | "Operating mode"
dataType_name | "ENUM"
dataType_family | "ENUM"
error | 0
value | "1"
desc | "Automatic"
dataType | 1
readonly | 0
readwrite | 0
unit | ""
710 |  
name | "Comfort setpoint "
dataType_name | "TEMP"
dataType_family | "VALS"
error | 0
value | "21.0"
desc | ""
precision | 0.1
dataType | 0
readonly | 0
readwrite | 0
unit | "°C"
712 |  
name | "Room temp reduced setpoint"
dataType_name | "TEMP"
dataType_family | "VALS"
error | 0
value | "17.0"
desc | ""
precision | 0.1
dataType | 0
readonly | 0
readwrite | 0
unit | "°C"
720 |  
name | "Heating curve slope"
dataType_name | "FP02"
dataType_family | "VALS"
error | 0
value | "1.32"
desc | ""
precision | 0.01
dataType | 0
readonly | 0
readwrite | 0
unit | ""
721 |  
name | "Heating curve parallel displacement"
dataType_name | "TEMP"
dataType_family | "VALS"
error | 0
value | "-1.0"
desc | ""
precision | 0.1
dataType | 0
readonly | 0
readwrite | 0
unit | "°C"
1600 |  
name | "Operating mode"
dataType_name | "ENUM"
dataType_family | "ENUM"
error | 0
value | "1"
desc | "On"
dataType | 1
readonly | 0
readwrite | 0
unit | ""
1610 |  
name | "DHW temperature nominal setpoint"
dataType_name | "TEMP"
dataType_family | "VALS"
error | 0
value | "60.0"
desc | ""
precision | 0.1
dataType | 0
readonly | 0
readwrite | 0
unit | "°C"
1612 |  
name | "DHW temperature reduced setpoint"
dataType_name | "TEMP"
dataType_family | "VALS"
error | 0
value | "35.0"
desc | ""
precision | 0.1
dataType | 0
readonly | 0
readwrite | 0
unit | "°C"
6705 |  
name | "Internal value of diagnostic code"
dataType_name | "UINT"
dataType_family | "VALS"
error | 0
value | "0"
desc | ""
precision | 1
dataType | 0
readonly | 1
readwrite | 1
unit | ""
8000 |  
name | "Status heating circuit 1"
dataType_name | "ENUM"
dataType_family | "ENUM"
error | 0
value | "114"
desc | "Heating mode Comfort"
dataType | 1
readonly | 1
readwrite | 1
unit | ""
8003 |  
name | "Status DHW"
dataType_name | "ENUM"
dataType_family | "ENUM"
error | 0
value | "100"
desc | "Charged, reduced temperature"
dataType | 1
readonly | 1
readwrite | 1
unit | ""

</div><!--EndFragment-->
</body>
</html>
```

1coderookie, 2022-11-12:
I think that's my bad - there was a PR about this HA example for the german language version of the manual with some changes but I didn't made the changes within the english version yet :( Sorry!! :(
Could you please try the examples from the german version of the manual? You'll find it here: https://1coderookie.github.io/BSB-LPB-LAN/kap08.html#811-home-assistant
Thanks

1coderookie, 2022-11-12:
..I now adjusted the english version: https://1coderookie.github.io/BSB-LPB-LAN_EN/chap08.html#811-home-assistant

You probably need to refresh the page - if the code is displayed in different colours now, it's the changed version:
[image]

proasnet, 2022-11-12:
```
mqtt:
  sensor:
    - state_topic: "bsb-lan/8310"
      name: BSB-LAN flow temperature
      unit_of_measurement: °C
      device_class: temperature
```
[image]

proasnet, 2022-11-12:
[image]

1coderookie, 2022-11-12:
I guess that means it doesn't work..? Sorry, but I can't help with that any further, I don't know anything about HA. Maybe @tiger42 , @GSzabados or @liudger knows the solution for this?
And/or maybe you can find help in the belonging thread within the home assistant forum? https://community.home-assistant.io/t/bsb-lan-integration/113501/108

mirkolenz, 2022-11-12:
Are you by any chance using version 2.2 of bsblan? I was faced with the same issue: My sensors were not updating. After some troubleshooting I found out that since the new release, parameters are published only under their new „floar id“. For instance, parameter `700` would be published under `700.00`. This is currently not documented (at least to my knowledge).

1coderookie, 2022-11-12:
It's only one following ".0", so it's e.g. 700.0 - yes, that may be the reason! 
And yes, I didn't yet mentioned the ".0" yet within the manual, that's right - sorry for that, I'll take care about it asap..
But: There's a note at the beginning of the chapter 8 that users have to check stuff like that by their own when it comes down to using the examples, cuz those aren't written by me, they were given to me by other users.. :(
Anyway - did you get it working at the end @mirkolenz ? 
I _think_ it should also be possible with e.g. 700 instead of 700.0, cuz in that case it will just query all possible 700.x - but maybe I'm wrong..

1coderookie, 2022-11-12:
..about the sensors: I recently added new screenshots where the .0/.1/.2 etc are visible (DHT22 still missing tho). But also here: if you just query the sensor number w/o the following .0/.1 etc then it _should_ give you all the possible values/endings anyway. Just if you want to query a specific value from that summarized output, let's say from .1, then you'd have to query that one specifically.

GSzabados, 2022-11-12:
Install MQTT Explorer addon to be able to see what MQTT messages HA receives from BSB-LAN.

mirkolenz, 2022-11-12:
> And yes, I didn't yet mentioned the ".0" yet within the manual, that's right - sorry for that, I'll take care about it asap..
> But: There's a note at the beginning of the chapter 8 that users have to [...]

I really did not intend to insult you in any way, sorry if it came across differently. I am extremely happy with the documentation of this project and found it to be one of the best of a "hobby project" I worked with. With that in mind, I was just surprised that these changes between the versions were not yet documented 😄 Keep up your great work!

> I think it should also be possible with e.g. 700 instead of 700.0, cuz in that case it will just query all possible 700.x - but maybe I'm wrong..

In MQTT you only subscribe to a certain topic. It seems to be the case that currently BSB-LAN only publishes the parameters to the "long" float address and not the shorter one. Personally I think that this is sufficient.

> It's only one following ".0", so it's e.g. 700.0 - yes, that may be the reason!

This I also what I thought initially, but that did not work in my case. After upgrading I first wondered why my states are no longer updating in Home Assistant. Upon investigation, I observed a large amount of dropped messages on my broker. Eventually, I then found out that you have to subscribe to `700.00` with tho trailing zeros. I have this running in my setup for the past weeks and just verified on my broker that I do not get messages on `700.0` and `700`. I uploaded a screenshot, maybe this is helpful for others.

[image]

proasnet, 2022-11-12:
What usage is recomended?
Plain Text, JSON or rich JSON for this example?
Could this be the cause of the problem?
I tried all three variants but without effect

proasnet, 2022-11-12:
It look that only status...
[image]

1coderookie, 2022-11-12:
> I really did not intend to insult you in any way, sorry if it came across differently. I am extremely happy with the documentation of this project and found it to be one of the best of a "hobby project" I worked with. With that in mind, I was just surprised that these changes between the versions were not yet documented smile Keep up your great work!

No, don't worry, I didn't get it that way! But you're right, it's still missing and I need to change that asap..
Oh, and thanks for the flowers! ;)

> In MQTT you only subscribe to a certain topic. It seems to be the case that currently BSB-LAN only publishes the parameters to the "long" float address and not the shorter one. Personally I think [...]

Hmm, ok, that's kinda weird.. I don't really know anything about MQTT tho, but there was just a user who reported a query using mosquitto and he only used the 1010 w/o the trailing .0 or even .00 and it worked (with the most recent version, so he wasn't referring to an old version)..

> > It's only one following ".0", so it's e.g. 700.0 - yes, that may be the reason!
> 
> This I also what I thought initially, but that did not work in my case. After upgrading I first wondered why my states are no longer updating in Home Assistant. Upon investigation, I observed a large amount of dropped messages on my broker. Eventually, I then found out that you have to subscribe to `700.00` with two trailing zeros. I have this running in my setup for the past weeks and just verified on my broker that I do not get messages on `700.0` and `700`. I uploaded a screenshot, maybe this is helpful for others.
> 

@fredlcore Do you know why it is like that..?

1coderookie, 2022-11-12:
@proasnet  So did you try already using .0 or .00 after the parameter number?

proasnet, 2022-11-12:
Yes, now I have theese in cfg
```
mqtt:
  sensor:
    - state_topic: "bsb-lan/8310.0"
      name: BSB-LAN flow temperature
      unit_of_measurement: °C
      device_class: temperature
```
No effect

1coderookie, 2022-11-12:
..and 8310.00?

mirkolenz, 2022-11-12:
> Do you know why it is like that..?

@1coderookie It could also be the case that I have some "strange" configuration that causes this behavior. I once had a bug in BSB-LAN that was caused by a special combination of enabled and disabled functions.

1coderookie, 2022-11-12:
@mirkolenz  Well, but then it would be a bug and that shouldn't happen anyway.. ;) 

@proasnet I dunno which heating system you have, but did you make sure beforehand using the webinterface if you have 8310 anyway?

mirkolenz, 2022-11-12:
@1coderookie That is right :)

@proasnet What log interval did you set up in your config. I think the explorer/client will only show messages/topics that have been published since it has been connected to the broker. If you for example have an interval of one hour, no topic would be shown to you.

1coderookie, 2022-11-12:
> @proasnet What log interval did you set up in your config. I think the explorer/client will only show messages/topics that have been published since it has been connected to the broker. If you for example have an interval of one hour, no topic would be shown to you.

Oh yeah, that's right! So if you aren't actively 'polling' with your broker and you didn't change the default loginterval of BSB-LAN, then you'd have to wait 60min/3600sec! So: either poll or change the loginterval within the webconfig of BSB-LAN..

proasnet, 2022-11-12:
Yes, I have a value
`8310.0 Diagnostics producer - Boiler temp actual value: 41.5 °C`
Here is only one decimal place.
I tried this
```
mqtt:
  sensor:
    - state_topic: "bsb-lan/8310.00"
      name: BSB-LAN flow temperature
      unit_of_measurement: °C
      device_class: temperature
```
and no effect too.
Yes, I have a 3600s. I try set 30, wait a moment please. This is a right!

proasnet, 2022-11-12:
So, in explorer I see
[image]
but status of entity still unknown

GSzabados, 2022-11-12:
> json status is live now, every 30s is more datas

I use plain text.

1coderookie, 2022-11-12:
Nice! And you used "8830" as shown in the screenshot or what exactly?

fredlcore, 2022-11-12:
Just one clarification regarding queries without the decimal point (i.e. 700 instead of 700.0 or 700.00): BSB-LAN is doing a numerical comparison. So if you query 700 it's numerically the same as 700.0, but not as 700.1. So querying 700 will only return the value of parameter 700, which is the same as 700.0, but it will not query 700.1 and following. You have to specifically query these because usually these also have a different meaning and are only logically put next to each other by the heater.

How each home automation system interprets the parameter numbers and what form it requires is a bit difficult to say because it depends on the way these parameters are queried. And this is only partly in our responsibility because it may depend on how any module or plugin deals with this. For example, if a module or plugin is still working with BSB-LAN v3.0 but internally only uses integer numbers (i.e. 700 instead of 700.0), then there may be problems with parameters such as 700.1.

On the other hand, I have not checked if all our way of connecting to the outside world (MQTT, JSON, plain HTTP requests) all use the same format. So it may be that one way uses or even requires a certain number format compared to the other. If anyone of you has clearly identified a discrepancy here, we will certainly try to fix it, or if not possible, mention the different ways of addressing the parameters in the manual.

mirkolenz, 2022-11-12:
The major difference seems to be that @proasnet used a JSON payload. So you also need to update your home assistant config to handle these JSON values correctly with a value template.

In my setup, I use the plain text format and thus subscribe to a different topic for each sensor.

proasnet, 2022-11-12:
Here is more parameters, on screenshot you see only one moment. I changed it to plain text, and see theese.
[image]
refresh now every 30sec

mirkolenz, 2022-11-12:
@fredlcore That all makes sense. So it could be the case that the MQTT integration is handling the numerical data differently. If for example I send a payload of `700` to the bsblan topic (i.e., request a status update), I get a response only on `bsblan/700.00`. But as I said this could also be caused by some configuration combinations on my side.

fredlcore, 2022-11-12:
@mirkolenz: Could you specify again through which way you only were able to query parameters using two trailing digits? For now, I could only think of one trailing digit being required, not two. But if I know where to look, I'll look into it.

proasnet, 2022-11-12:
but the result in the entity is still the same and the value is unknown

mirkolenz, 2022-11-12:
@proasnet I would assume that this is caused by a mismatch of the topic. In your home assistant config, you have `bsb-lan`, but the explorer shows `BSB-LAN`. In MQTT, the capitalization matters if I remember correctly. If that does not help, you could try adding the following line to your sensor config:

`value_template: "{{value | float}}"`

tiger42, 2022-11-12:
> @mirkolenz: Could you specify again through which way you only were able to query parameters using two trailing digits? For now, I could only think of one trailing digit being required, not two. But if I know where to look, I'll look into it.

The code that works in HA is this:
```
mqtt:
  sensor:
  - state_topic: "bsb-lan/8700.00"
    name: BSB-LAN Außentemperatur
    unit_of_measurement: °C
    device_class: temperature
```
That makes sense, as the topics in MQTT are always **strings**, not integers or floats. And as you can see in MQTT Explorer, the topics have two digits:
[image]
So if the topic is for example "bsb-lan/8700.00", you have to query it exactly like that, 8700 or 8700.0 or 8700.00000 won't work.

tiger42, 2022-11-12:
> @proasnet I would assume that this is caused by a mismatch of the topic. In your home assistant config, you have `bsb-lan`, but the explorer shows `BSB-LAN`. In MQTT, the capitalization matters if I [...]

Capitalization in fact matters! In my BSB-LAN instance I've set the topic to lowercase letters for example, because uppercase in MQTT looks kind of awkward... But the default setting, if you haven't changed it, is "BSB-LAN" (uppercase).

mirkolenz, 2022-11-12:
@fredlcore For all log parameters that I define in BSB-LAN (e.g., `700`), the values are only exposed at an address with two trailing zeros (e.g., `700.00). All publishing options work as intended though, so publishing the payload `S700=1` works just as well as `S700.00=1` or `S700.0=1`. So the only area where this "bug" occurs is on the topics that one can subscribe to.

1coderookie, 2022-11-12:
@tiger42 Thanks for that info, so I'll adjust your examples for the manual asap..

@fredlcore Thanks for the clearification about 700/700.0 and .1 etc..

proasnet, 2022-11-12:
Yes, this working
```
mqtt:
  sensor:
  - state_topic: "bsb-lan/8700.00"
    name: BSB-LAN Außentemperatur
    unit_of_measurement: °C
    device_class: temperature
```
And in BSB Lan web I changed BSB-LAN to bsb-lan

proasnet, 2022-11-12:
[image]

mirkolenz, 2022-11-12:
Great to hear that!

proasnet, 2022-11-12:
[image]

proasnet, 2022-11-12:
Thank you.
Tomorow I try add more sensors, and how set room temperature and mode of system, if will work.
Good night

1coderookie, 2022-11-12:
Great!
Thanks to everybody for helping out! :+1:

GSzabados, 2022-11-12:
> Thank you. Tomorow I try add more sensors, and how set room temperature and mode of system, if will work. Good night

I tried to make a MQTT input select for mode selection, but I couldn't figure out how to use the select to change the right MQTT topic and keep the select at the same state.
I spent hours with it to figure out and tried all the options what was possible, but either it did not change the mode or it reverted back immediately through the select to the previous mode.

If you figure out, please share it how it works for you.

mirkolenz, 2022-11-12:
@GSzabados Tgere you go:

```yaml
mqtt:
  select:
    - name: Boiler operating mode
      unique_id: some_id
      state_topic: "bsblan/700.00"
      command_topic: "bsblan"
      availability_topic: "bsblan/status"
      # Example value: 1 - Automatic
      value_template: >
        {% set mapping = {0: 'Protection', 1: 'Automatic', 2: 'Reduced', 3: 'Comfort'} %}
        {% set idx = value.split() | first | int %}
        {{ mapping[idx] }}
      # Example payload: S700=1
      command_template: >
        {% set mapping = {'Protection': 0, 'Automatic': 1, 'Reduced': 2, 'Comfort': 3} %}
        S700={{ mapping[value] }}
      options:
        - Automatic
        - Comfort
        - Away
        - Eco
      icon: mdi:list-box
```

Please note that the specific available modes may be unique to your heating device.

GSzabados, 2022-11-12:
> **BSB-LAN Version**
> 3.0.1-20221101130752

@1coderookie, if I understand correctly the new version is working already and distributed as well on request?

1coderookie, 2022-11-12:
What do you mean by 'distributed as well on request'? 
Yes, v3.x is up now, but there were major changes which you need to pay attention to if you want to upgrade (also the trailing .0 which was mentioned above btw)! 
We discussed some of that here: https://github.com/fredlcore/BSB-LAN/discussions/482
So if you want to update, you should make a complete new install without using any old files (like config.h or whatever). Then you'd need a specific custom_defs.h file from @fredlcore for your system to get access to the parameters your controller has to offer. Check out the manual (chap 2 and 3.3), there you'll find the infors.

1coderookie, 2022-11-12:
@mirkolenz  So there `S700={{ mapping[value] }}` it works without the .00??

dukess, 2022-11-12:
Hi, it is a bug, i think.
If this problem appears when used plain format, i can propose this solution (i hope it help):
in file BSB_LAN/include/mqtt_handler.h replace stings 
```
  } else {
    MQTTTopic.concat(String(param));
  }
```
with this fragment:
```
  } else {
    if(roundf(param * 10) != roundf(param) * 10)
      MQTTTopic.concat(String(param, 0));
    else
      MQTTTopic.concat(String(param, 1));
  }
```

GSzabados, 2022-11-12:
> Please note that the specific available modes may be unique to your heating device.

Thanks! And how does Away and Eco selects Reduced and Protection? Actually I need to convert all of those to the old German phrases, 

```
options:
           - "0 - Schutzbetrieb"
           - "1 - Automatik"
           - "2 - Reduziert"
           - "3 - Komfort"
```

mirkolenz, 2022-11-12:
@1coderookie Yes exactly! That was what I tried to say in my response earlier:

> For all log parameters that I define in BSB-LAN (e.g., 700), the values are only exposed at an address with two trailing zeros (e.g., 700.00). All publishing options work as intended though, so publishing the payload S700=1 to the topic bsblan works just as well as S700.00=1 or S700.0=1. So the only area where this "bug" occurs is on the topics that one can subscribe to: bsblan/700 and bsblan/700.0 do not receive any values, only bsblan/700.00.

So the bug only affects subscriptions, but nu publishing operations.

1coderookie, 2022-11-12:
> > Please note that the specific available modes may be unique to your heating device.
> 
> Thanks! And how does Away and Eco selects Reduced and Protection? Actually I need to convert all of those to the old German phrases,

When you ask for your tailormade custom_defs.h from @fredlcore , let him know that you need the English language version, it should be in there.

1coderookie, 2022-11-12:
@mirkolenz Oh ok, sorry, didn't realize that earlier..

mirkolenz, 2022-11-12:
@GSzabados In your case, you can just insert the German strings to the corresponding numbers. My machine uses the same modes with the same numerical values.

GSzabados, 2022-11-12:
> What do you mean by 'distributed as well on request'?

I read the discussion, but I thought it is still planned as it was saying something about 3.x will have a generic version and for full feature set you will need to get the specific custom_defs.h. And the last released version with tag is v2.2.
But good to know that it is working. I will get in touch with @fredlcore when I will have some time to redo the controller. Unfortunately not at the moment.

1coderookie, 2022-11-12:
Well, in v2.2 there also cae those changes, but that version was still supported with the old&complete defs.h, just named custom_defs.h. Now from v3.x on you _need_ the tailormade custom_defs.h from freetz, otherwise you won't have access to most of the parameters from your controller.
It may be that names changed and certain parameters which were available before won't be availalbe anymore though. That's cuz those IDs came from different controllers which happen to work at your controller also. So now with v3.x you're on the safe side..

1coderookie, 2022-11-12:
> Hi, it is a bug, i think. If this problem appears when used plain format, i can propose this solution (i hope it help): in file BSB_LAN/include/mqtt_handler.h replace lines
> 
...
> 
> If the parameter is an integer, it will be passed without a separator and zero. Otherwise, one decimal place will be printed.

Thanks @dukess !
So please let me/us know if, where and how the HA example from @tiger42 needs to be changed then after you guys did the final tests on that..

@GSzabados in the mentioned example from @tiger42 there's also an example about how to set values/parameters, so maybe you can also try that one and let us know if it works as expected? Thanks..

1coderookie, 2022-11-12:
..I'm afk for today, so have a nice evening/day everybody.. ;)

Devil-of-Chaos, 2022-11-12:
Hi there maybe my HA config helps

MQTT settings are:
Plain Text, Json had working too but I don‘t need this

Topic and ID are  BSB-LAN

MQTT Part (only short part)

```
mqtt:
sensor: 
    - state_topic: "BSB-LAN/7760.00"
      unique_id: "bsb_lan_kessel_temp"
      object_id: "bsb_lan_kessel_temp"
      name: "Kessel"
      value_template: >-
        {% if value == '---' %}
            {{ '0.0' | float }}
        {% else %}
            {{ value }}
        {% endif %}
      unit_of_measurement: '°C'
      device_class: temperature
      availability_topic: "BSB-LAN/status"
      payload_available: "online"
      payload_not_available: "offline"
      device: {
          identifiers: ["00000002"],
          name: "BSB-LAN",
          model: "ESP32",
          manufacturer: "Github"
      }
    - state_topic: "BSB-LAN/8560.00"
      unique_id: "bsb_lan_feststoffkessel_temp"
      object_id: "bsb_lan_feststoffkessel_temp"
      name: "Feststoffkessel"
      value_template: >-
        {% if value == '---' %}
            {{ '0.0' | float }}
        {% else %}
            {{ value }}
        {% endif %}
      unit_of_measurement: '°C'
      device_class: temperature
      availability_topic: "BSB-LAN/status"
      payload_available: "online"
      payload_not_available: "offline"
      device: {
          identifiers: ["00000002"],
          name: "BSB-LAN",
          model: "ESP32",
          manufacturer: "Github"
      } 

```

Send Commands it‘s from tigers example

```
bsb_lan_set_parameter:
    url: http://192.168.23.52/JS
    method: POST
    username: !secret bsb_lan_user
    password: !secret bsb_lan_pass
    # Parameter "type": 1 = SET (default), 0 = INF
    payload: '{"Parameter": "{{ parameter }}", "Value": "{{ value }}", "Type": "{% if type is defined %}{{ type }}{% else %}1{% endif %}"}'
    
bsb_lan_trinkwasser_aus:
    url: http://192.168.23.52/S1600=0
    username: !secret bsb_lan_user
    password: !secret bsb_lan_pass
bsb_lan_trinkwasser_ein:
    url: http://192.168.23.52/S1600=1
    username: !secret bsb_lan_user
    password: !secret bsb_lan_pass 

```

And slow update in this example time pattern normal and reduced Temperatur 

```
platform: rest
  name: BSB-LAN Zeitplan
  resource: "http://192.168.23.52/JQ=501,502,503,504,505,506,507,521,522,523,524,525,526,527,8005"
  method: POST
  username: !secret bsb_lan_user
  password: !secret bsb_lan_pass
  value_template: >-
    {% if value_json['8005'].value == 'unknown' %}
        {{ 'Starting...' }}
    {% else %}
        {{ 'OK' }}
    {% endif %}
  scan_interval: 3600
  timeout: 45
  json_attributes:
    - "501"
    - "502"
    - "503"
    - "504"
    - "505"
    - "506"
    - "507"
    - "521"
    - "522"
    - "523"
    - "524"
    - "525"
    - "526"
    - "527"
    - "8005"
  headers:
    Content-Type: application/json
```

fredlcore, 2022-11-12:
> > Hi, it is a bug, i think. If this problem appears when used plain format, i can propose this solution (i hope it help): in file BSB_LAN/include/mqtt_handler.h replace lines
> > ...
> Thanks @dukess ! So please let me/us know if, where and how the HA example from @tiger42 needs to be changed then after you guys did the final tests on that..

I'm not sure if @dukess can make recommendations regarding HA because if I'm not mistaken, none of us two is using HA. But if I understood his code changes right, then with his suggested changes, you basically get what you request. So if you take @tiger42's example, you wouldn't need to write `8700.00` anymore, but you could just write `8700` or `8700.1` (if applicable). This would have to be tested by someone who actually uses HA with MQTT after making the code change @dukess suggested. If that provides a more coherent way of querying parameters, I shall make the changes in the main code accordingly.

GSzabados, 2022-11-13:
> In your case, you can just insert the German strings to the corresponding numbers.

Thanks @mirkolenz. I used your example, just changed the options of Away and Eco to the correct English ones, ordered as well and all works fine with the German language BSB-LAN values.

I am slowly adding pieces and bits to my HA config when I really start to need it. And this was a big step now. I can build some Automations to turn off the heating when all of us away, etc. I really appreciate this great help!

[19 more replies]

closed 2022-11-13 by fredlcore