fredlcore/BSB-LAN #515: [BUG] JSON error from JI request with no SD card

liudger, 2022-11-13:
**Before submitting a bug report, please make sure that you have checked [chapter 14](https://1coderookie.github.io/BSB-LPB-LAN_EN/chap14.html) of the manual** ("Problems and their Possible Causes"). Most problems that are reported to us can be solved by going through the solutions provided there.

**BSB-LAN Version**
"3.0.1-20221101130752"

**Architecture**
 Arduino Due

**Bus system**
BSB

**Describe the bug**
When doing a request for device info `URL/JI` and no SD card installed. The response has a malformed json.

**To Reproduce**
Steps to reproduce the behavior:
1. no SD card in bsblan device
2. do the JI query
3. See error
```{
    "name": "BSB-LAN",
    "version": "3.0.1-20221101130752",
    "hardware": "Due",
    "freeram": 81271,
    "uptime": 1664668,
    "MAC": "00:80:41:19:69:90",
    "freespace": ,
    "bus": "BSB",
    "buswritable": 0,
    "busaddr": 66,
    "busdest": 0,
    "monitor": 0,
    "verbose": 1
}
```
freespace has no value, should be zero?

**Logs**
If possible, attach or copy/paste a log of the Arduino IDE's serial monitor when performing the above actions. If the error is reproducible, do so after pressing the reset button on the board.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (if applicable, please complete the following information):**
 - OS: [e.g. MacOS 11.1, Windows 11]
 - Browser [e.g. Chrome, Safari]
 - Browser version [e.g. 22]

**Smartphone (if applicable, please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. Safari, FireFox]
 - Browser version [e.g. 22]

**Additional context**
Add any other context about the problem here.

dukess, 2022-11-14:
Question: LOGGER was enabled?

liudger, 2022-11-14:
Not sure. I will check. Still there should be valid json. 0 or Nan?

dukess, 2022-11-14:
In https://github.com/dukess/BSB-LAN/commit/bd43d5a68482ce44da86fedc332c94afd3e46c33 i propose to remove "freespace" field when SD card not initialized.

fredlcore, 2022-11-14:
Hm, I don't know much about JSON parsing, but would it create problems if a field is queried if it's not present? Otherwise I'd suggest `NaN` if the SD card is not present because `0` could also mean that the SD card is full.

dukess, 2022-11-14:
did that

liudger, 2022-11-15:
I updated to the latest and the build failed. I had turned off some defines. And this caused a build issue. (fs_date was not defined) When I enabled most defines the build worked. Maybe we need to test this better? Or hide defines?

Also the issue was now solved!

closed 2022-11-15 by liudger

fredlcore, 2022-11-15:
Well, if certain defines (or the lack thereof) cause an error, then it's of course important to know. And it's one of the reasons why we try to reduce the number of defines used as configurations because this is more difficult to test. But if you remember how you did it and what you did, then please let us know.

liudger, 2022-11-15:
it was `#define LOGGER`
```
Using board 'arduino_due_x_dbg' from platform in folder: /Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12
Using core 'arduino' from platform in folder: /Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12
Detecting libraries used...
/Users/liudger/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sketch-C3F26756AB1E011A300940F4552C2E1B/sketch/BSB_LAN.ino.cpp -o /dev/null
Alternatives for SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.0]
/Users/liudger/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/libraries/SPI/src /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sketch-C3F26756AB1E011A300940F4552C2E1B/sketch/BSB_LAN.ino.cpp -o /dev/null
Alternatives for Wire.h: [Wire@1.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.0]
/Users/liudger/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/libraries/SPI/src -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/libraries/Wire/src /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sketch-C3F26756AB1E011A300940F4552C2E1B/sketch/BSB_LAN.ino.cpp -o /dev/null
Alternatives for EthernetUdp.h: [Ethernet@2.0.1]
ResolveLibrary(EthernetUdp.h)
  -> candidates: [Ethernet@2.0.1]
/Users/liudger/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/libraries/SPI/src -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/libraries/Wire/src -I/Users/liudger/Library/Arduino15/libraries/Ethernet/src /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sket [... cut ...]
/Users/liudger/Git/BSB-LAN/BSB_LAN/BSB_LAN.ino:7788:3: error: 'FsDateTime' has not been declared
   FsDateTime::setCallback(dateTime);
   ^

Using library SPI at version 1.0 in folder: /Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/libraries/SPI 
Using library Wire at version 1.0 in folder: /Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/libraries/Wire 
Using library Ethernet at version 2.0.1 in folder: /Users/liudger/Library/Arduino15/libraries/Ethernet 
exit status 1

Compilation error: 'FS_DATE' was not declared in this scope
```

dukess, 2022-11-15:
Very interesting. On arduino studio this build done with success.probabely 
we need to add #include directive.

Willem-Jan van Rootselaar ***@***.***> 15 ноября 2022 г. 
18:15:12 написал:

dukess, 2022-11-15:
@liudger can you add
#include "src/SdFat/common/FsDateTime.h"
near 
#include "src/SdFat/SdFat.h"
and rebuild project?
Thank you