fredlcore/BSB-LAN #516: [BUG] Lockup when querying Parameter 2505

schlaubstar, 2022-11-14:
**BSB-LAN Version**
Version: 2.2.1-20221101122308

**Architecture**
ESP32

**Bus system**
BSB

**Describe the bug**
When opening /K34 (which contains the Parameters 2200-2682) the page loads and gets stuck in an endless loop repeating: 
“2505.0 Kessel - Schwelle Durchflussdetektion: (parameter not supported)”
So in the webpage keeps loading and growing in size (vertical scroll), repeating the entry above endlessly … and: locking up the whole device so it needs to be rebooted.

**To Reproduce**
Steps to reproduce the behavior:
1. Open WebInterface
2. Click on 'Heizungsfunktionen' (or /K)
3. Click on 'Kessel' (or /K34)
4. See error

**Logs**
LAN->HEIZ QUR 2505.0 Kessel - Schwelle Durchflussdetektion: 
DC C2 00 0B 06 3D 05 1A E1 73 7D
HEIZ->LAN ERR 2505.0 Kessel - Schwelle Durchflussdetektion: error 7
DC 80 42 0C 08 05 3D 1A E1 07 9D D5
#2505:  (parameter not supported)
LAN->HEIZ QUR 2505.0 Kessel - Schwelle Durchflussdetektion: 
DC C2 00 0B 06 3D 05 1A E1 73 7D
HEIZ->LAN ERR 2505.0 Kessel - Schwelle Durchflussdetektion: error 7
DC 80 42 0C 08 05 3D 1A E1 07 9D D5
#2505:  (parameter not supported)
LAN->HEIZ QUR 2505.0 Kessel - Schwelle Durchflussdetektion: 
DC C2 00 0B 06 3D 05 1A E1 73 7D
HEIZ->LAN ERR 2505.0 Kessel - Schwelle Durchflussdetektion: error 7
DC 80 42 0C 08 05 3D 1A E1 07 9D D5
#2505:  (parameter not supported)
LAN->HEIZ QUR 2505.0 Kessel - Schwelle Durchflussdetektion: 
DC C2 00 0B 06 3D 05 1A E1 73 7D
HEIZ->LAN ERR 2505.0 Kessel - Schwelle Durchflussdetektion: error 7
DC 80 42 0C 08 05 3D 1A E1 07 9D D5
#2505:  (parameter not supported)
LAN->HEIZ QUR 2505.0 Kessel - Schwelle Durchflussdetektion: 
DC C2 00 0B 06 3D 05 1A E1 73 7D
HEIZ->LAN ERR 2505.0 Kessel - Schwelle Durchflussdetektion: error 7
DC 80 42 0C 08 05 3D 1A E1 07 9D D5
#2505:  (parameter not supported)

**Expected behavior**
The page should load displaying parameters 2200-2682 or displaying an erromessage but not locking up in an endless loop.

**Screenshots**
See attached screenshot

**Desktop (if applicable, please complete the following information):**
•	Not of relevance: happens independent of the browser/device opening the URL

**Smartphone (if applicable, please complete the following information):**
•	Not of relevance: happens independent of the browser/device opening the URL

**Additional context**
Hardware: ESP32
Version: 2.2.1-20221101122308
Freier Speicher: 219232 Bytes
Uptime: 185239

The whole device is attached to a Brötje EcoTherm Plus (BBS EVO 20), who could be added as “tested” device.

The Debug-Log (Serial Output) displays various error7 messages before, which do not lead to an lockup but the addition "parameter not supported" in the browser.

[image]

fredlcore, 2022-11-14:
You are using an outdated version of BSB-LAN, please update to the current version as we don't fix bugs in older versions.

closed 2022-11-14 by fredlcore

schlaubstar, 2022-11-14:
Hi @fredlcore : I am terribly sorry for that but I just compiled and uploaded the latest release published here. https://github.com/fredlcore/BSB-LAN/releases Where do I get a version 3.x?

1coderookie, 2022-11-14:
@schlaubstar Hmm, didn't you post a comment about an hour ago about where to find the current version..? Anyway - just download it directly from the main repo and please do a complete install from scratch. See also chaps 2 & 3.3 in the manual to note what to do..
https://github.com/fredlcore/BSB-LAN

schlaubstar, 2022-11-14:
@1coderookie : I am sorry for the confusion. My issue was closed with the explanation that I was using an outdated version, that I had downloaded via the official releases link. So I assumed, that the link was pointing to an outdated version, not instantly understanding that you only support bugs on the latest master.

I tried to compile the latest master but it fails for the ESP32 Dev module (which I am using), although I re-verified to have removed the correct directories/renamed the includes and also am on the desired version 2.0.2 of the ESP32 Board definitions as there seems to be an issue with esp32_time.h 
The compiler complains about a lot of multiple definitions for time objects.

`\Arduino\BSB-Lan\BSB-LAN-3.0\BSB_LAN\src\Time/Time.cpp:286: multiple definition of `setTime(int, int, int, int, int, int)'
\Arduino\BSB-Lan\BSB-LAN-3.0\BSB_LAN\src\Time/Time.cpp:133: multiple definition of `year()';
\Arduino\BSB-Lan\BSB-LAN-3.0\BSB_LAN\src\Time/Time.cpp:124: multiple definition of `month()' ...`

Any ideas, what is wrong here ? 
Thank you very much for the support.

schlaubstar, 2022-11-14:
Thank you so much. Removing the folder fixed it and also allowed to compile the binaries and do an OTA Update. I am just compiling the text file with available parameters for the custom_defs.h. 
Thanks again for the support.