fredlcore/BSB-LAN #518: [BUG] compiling with define logger disabled

liudger, 2022-11-15:
**Before submitting a bug report, please make sure that you have checked [chapter 14](https://1coderookie.github.io/BSB-LPB-LAN_EN/chap14.html) of the manual** ("Problems and their Possible Causes"). Most problems that are reported to us can be solved by going through the solutions provided there.

**BSB-LAN Version**
As displayed in the web-configuration or copy from `bsb-version.h` file.

**Architecture**
The architecture BSB-LAN is running on (Arduino Mega, Arduino Due, ESP32 NodeMCU, ESP32 Olimex EVB etc.)

**Bus system**
Which bus BSB-LAN is running on (BSB, LPB, PPS)?

**Describe the bug**
When disable #define Logger the build fails

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Logs**
```
Using board 'arduino_due_x_dbg' from platform in folder: /Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12
Using core 'arduino' from platform in folder: /Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12
Detecting libraries used...
/Users/liudger/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sketch-C3F26756AB1E011A300940F4552C2E1B/sketch/BSB_LAN.ino.cpp -o /dev/null
Alternatives for SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.0]
/Users/liudger/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/libraries/SPI/src /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sketch-C3F26756AB1E011A300940F4552C2E1B/sketch/BSB_LAN.ino.cpp -o /dev/null
Alternatives for Wire.h: [Wire@1.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.0]
/Users/liudger/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/libraries/SPI/src -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/libraries/Wire/src /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sketch-C3F26756AB1E011A300940F4552C2E1B/sketch/BSB_LAN.ino.cpp -o /dev/null
Alternatives for EthernetUdp.h: [Ethernet@2.0.1]
ResolveLibrary(EthernetUdp.h)
  -> candidates: [Ethernet@2.0.1]
/Users/liudger/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/libraries/SPI/src -I/Users/liudger/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/libraries/Wire/src -I/Users/liudger/Library/Arduino15/libraries/Ethernet/src /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sketch-C3F26756AB1E011A300940F4552C2E1B/sketch/BSB_LAN.ino.cpp -o /dev/null
Using cached library dependencies for file: /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sketch-C3F26756AB1E011A300940F4552C2E1B/sketch/src/ArduinoMDNS/MDNS.cpp
Using cached library dependencies for file: /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sketch-C3F26756AB1E011A300940F4552C2E1B/sketch/src/ArduinoMDNS/utility/EthernetUtil.c
Using cached library dependencies for file: /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sketch-C3F26756AB1E011A300940F4552C2E1B/sketch/src/BSB/BSBSoftwareSerial.cpp
Using cached library dependencies for file: /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sketch-C3F26756AB1E011A300940F4552C2E1B/sketch/src/BSB/bsb.cpp
Using cached library dependencies for file: /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sketch-C3F26756AB1E011A300940F4552C2E1B/sketch/src/Base64/src/Base64.cpp
Using cached library dependencies for file: /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sketch-C3F26756AB1E011A300940F4552C2E1B/sketch/src/BlueDot_BME280/BlueDot_BME280.cpp
Using cached library dependencies for file: /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sketch-C3F26756AB1E011A300940F4552C2E1B/sketch/src/CRC32/CRC32.cpp
Using cached library dependencies for file: /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sketch-C3F26756AB1E011A300940F4552C2E1B/sketch/src/DHTesp/DHTesp.cpp
Using cached library dependencies for file: /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sketch-C3F26756AB1E011A300940F4552C2E1B/sketch/src/DallasTemperature/DallasTemperature.cpp
Using cached library dependencies for file: /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sketch-C3F26756AB1E011A300940F4552C2E1B/sketch/src/OneWireNg/OneWireNg.cpp
Using cached library dependencies for file: /private/var/folders/lx/dyph0crj43dd9sml56_1qr280000gn/T/arduino-sketch-C3F26756AB1E011A300940F4552C2E1B/sketch/src/OneWireNg/OneWireNg_BitBang.cpp
[... cut ...]
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (if applicable, please complete the following information):**
 - OS: [e.g. MacOS 11.1, Windows 11]
 - Browser [e.g. Chrome, Safari]
 - Browser version [e.g. 22]

**Smartphone (if applicable, please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. Safari, FireFox]
 - Browser version [e.g. 22]

**Additional context**
Add any other context about the problem here.

liudger, 2022-11-15:
@dukess  Still the same error when compiling

> @liudger can you add
> #include "src/SdFat/common/FsDateTime.h"
> near
> #include "src/SdFat/SdFat.h"
> and rebuild project?
> Thank you

dukess, 2022-11-15:
Oh, got it.
This
// Call back for file timestamps.  Only called for file create and sync().
	#if !defined(ESP32)
	void dateTime(uint16_t* date, uint16_t* time) {
	  // Return date using FS_DATE macro to format fields.
	  *date = FS_DATE(year(), month(), day());
	

	  // Return time using FS_TIME macro to format fields.
	  *time = FS_TIME(hour(), minute(), second());
	

	}
	#endif
and this fragments

#if !defined(ESP32)
	  FsDateTime::setCallback(dateTime);
	#endif
additional should be escaped with
#if defined LOGGER || defined WEBSERVER
#endif

I will fix it tomorrow.
If you don't want to wait you can add these directives by yourself
,

Willem-Jan van Rootselaar ***@***.***> 15 ноября 2022 г. 
22:00:57 написал:

liudger, 2022-11-15:
I have logger on. Not a big deal. Just wanted to give you heads of this bug

fredlcore, 2022-12-09:
Can we consider this closed?

liudger, 2022-12-10:
Yes works fine now

closed 2022-12-10 by liudger