fredlcore/BSB-LAN #520: [BUG] Exception in mqtt_sendtoBroker for mqtt_mode 3

fschaeck, 2022-11-17:
**BSB-LAN Version**
3.0.1-20221101130752

**Architecture**
ESP32 NodeMCU (Joy-It)

**Bus system**
BSB

**Describe the bug**
When using mqtt_mode 3 an exception is thrown AFTER publishing the json message - apparently due to a buffer overflow in outBuf. Using mqtt_mode 1 or 2 this exception is not thrown but the message is also significantly shorter.

Also in function mqtt_callback(...) in BSB_LAN/include/mqtt_handler.h the buffer overflow protection is reserving 4 bytes for the acknowledgement message, but 5 are needed (4 for the ACK_ prefix and one for the terminating nut character).

**To Reproduce**
Steps to reproduce the behavior:
Set mqtt_mode in BSB_LAN_config.h to 3, compile and upload sketch, watch the serial monitor to see the exception happening.

**Logs**
(These logs are taken from an already modified mqtt_handler.h to pin down the problem)
12:41:55.566 -> �f`ADY
12:41:55.598 -> Using settings from config file
12:41:55.598 -> Reading done.
12:41:55.630 -> EEPROM schema v.0 Program schema v.11
12:41:55.630 -> Address EEPROM option 0: 0
12:41:55.630 -> Address EEPROM option 1: 1
12:41:55.630 -> Address EEPROM option 2: 2
12:41:55.630 -> Address EEPROM option 3: 6
12:41:55.630 -> Address EEPROM option 4: 226
12:41:55.630 -> Address EEPROM option 5: 306
12:41:55.662 -> Address EEPROM option 6: 490
12:41:55.662 -> Address EEPROM option 7: 491
12:41:55.662 -> Address EEPROM option 8: 492
12:41:55.662 -> Address EEPROM option 9: 493
12:41:55.662 -> Address EEPROM option 10: 494
12:41:55.662 -> Address EEPROM option 11: 495
12:41:55.662 -> Address EEPROM option 12: 499
12:41:55.662 -> Address EEPROM option 13: 659
12:41:55.662 -> Address EEPROM option 14: 663
12:41:55.662 -> Address EEPROM option 15: 823
12:41:55.662 -> Address EEPROM option 16: 829
12:41:55.662 -> Address EEPROM option 17: 830
12:41:55.694 -> Address EEPROM option 18: 834
12:41:55.694 -> Address EEPROM option 19: 838
12:41:55.694 -> Address EEPROM option 20: 842
12:41:55.694 -> Address EEPROM option 21: 846
12:41:55.694 -> Address EEPROM option 22: 848
12:41:55.694 -> Address EEPROM option 23: 852
12:41:55.694 -> Address EEPROM option 24: 856
12:41:55.694 -> Address EEPROM option 25: 920
12:41:55.694 -> Address EEPROM option 26: 984
12:41:55.694 -> Address EEPROM option 27: 985
12:41:55.694 -> Address EEPROM option 28: 986
12:41:55.694 -> Address EEPROM option 29: 996
12:41:55.726 -> Address EEPROM option 30: 997
12:41:55.726 -> Address EEPROM option 31: 1157
12:41:55.726 -> Address EEPROM option 32: 1158
12:41:55.726 -> Address EEPROM option 33: 1162
12:41:55.726 -> Address EEPROM option 34: 1163
12:42:07.699 -> 
12:42:07.729 -> Starting MDNS service with hostname BSB-LAN
12:42:07.729 -> Update Server started on port 8080.
12:42:07.729 -> Setup complete
12:42:07.729 -> Client ID: BSB-LAN
12:42:07.729 -> Will topic: BSB-LAN/status
12:42:07.762 -> Connect to MQTT broker, updating will topic
12:42:07.762 -> Subscribed to topic 'BSB-LAN'
12:42:07.762 -> Published status 'online' to topic 'BSB-LAN/status'
12:42:26.560 ->  query failed
12:42:27.571 ->  query failed
12:42:28.587 ->  query failed
12:42:28.587 -> 8700
12:42:28.587 -> Output topic: BSB-LAN/json
12:42:28.587 -> 
12:42:28.587 -> Payload Output : {"BSB-LAN":{"id":8700,"name":"Außentemperatur","value": "","desc": "","unit": "°C","error": 261}}
12:42:28.620 -> Successfully published...
12:42:28.620 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
12:42:28.620 -> 
12:42:28.620 -> Core  1 register dump:
12:42:28.620 -> PC      : 0x4008a431  PS      : 0x00060f30  A0      : 0x800d8ab1  A1      : 0x3ffb24a0  
12:42:28.620 -> A2      : 0x3ffc3e4b  A3      : 0x00000000  A4      : 0x000000ff  A5      : 0x0000ff00  
12:42:28.620 -> A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb20e0  
12:42:28.653 -> A10     : 0x3ffc3e4b  A11     : 0xffffffff  A12     : 0x0000000d  A13     : 0x3ffb22f0  
12:42:28.653 -> A14     : 0x0000000d  A15     : 0x3ffb22f0  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
12:42:28.653 -> EXCVADDR: 0x00000000  LBEG    : 0x4008a431  LEND    : 0x4008a43e  LCOUNT  : 0xffffffff  
12:42:28.653 -> 
12:42:28.653 -> 
12:42:28.653 -> Backtrace:0x4008a42e:0x3ffb24a00x400d8aae:0x3ffb24b0 0x400ddc0e:0x3ffb2530 0x400e9bc9:0x3ffb2820 
12:42:28.685 -> 
12:42:28.685 -> 
12:42:28.685 -> 
12:42:28.685 -> 
12:42:28.685 -> ELF file SHA256: 0000000000000000
12:42:28.685 -> 
12:42:28.685 -> Rebooting...

**Expected behavior**
A published json message without a subsequent exception.

**Screenshots**
n/a

**Desktop (if applicable, please complete the following information):**
n/a

**Smartphone (if applicable, please complete the following information):**
n/a

**Additional context**
Nothing to add.

fredlcore, 2022-12-09:
Can we consider this closed due to your PR, @fschaeck?

fschaeck, 2022-12-10:
Yes, we can! Closing it…

closed 2022-12-10 by fschaeck