fredlcore/BSB-LAN #525: [BUG] parameter 1132 decoded incorrectly

DE-cr, 2022-11-21:
**BSB-LAN Version**
3.0.1-20221101130752

**Architecture**
ESP32 NodeMCU

**Bus system**
BSB

**Describe the bug**
BSB-LAN decodes as follows:
`1132.0 Heizkreis 2 - Antrieb-Regelungsart Heizkreis: 255 - not found`
while ISR shows in the OEM menues (and only there):
[image]

fredlcore, 2022-11-21:
Please make sure that you have installed the most recent version of BSB-LAN before submitting a bug. Current version is 3.0.3 and the new data type VT_BINARY_ENUM has been added already. 
Depending on how old your generated BSB_LAN_custom_defs.h is, you may need to request a new one or change the type of the affected parameters from VT_ENUM to VT_BINARY_ENUM yourself.

closed 2022-11-21 by fredlcore

DE-cr, 2022-11-21:
> Please make sure that you have installed the most recent version of BSB-LAN before submitting a bug. Current version is 3.0.3 and the new data type VT_BINARY_ENUM has been added already. Depending [...]

Sorry, from now on, I'll always install the most recent version of BSB-LAN before submitting a bug.

How do I know what the affected parameters are? Or is it all VT_ENUMs?

fredlcore, 2022-11-21:
It's only those parameters that give you a `255 - not found` in the value of an ENUM. Just change that parameter in your _custom_defs.h from VT_ENUM to VT_BINARY_ENUM and you're done.