fredlcore/BSB-LAN #531: Bind log_interval to RTC not millis()

dukess, 2022-12-18:
Hi, @fredlcore , @1coderookie 
For some technical reasons, I had the idea to slightly rework logging (and sending via MQT) the program values specified in log_parameters:
I would like to link these operations not to the millis() timer, which counts down the time from the start of the microcontroller, but to the "real-time clock", which is initialized by the boiler clock.
Pros: Data will always be sent at the same time, regardless of the time the controller is started.
Cons: if I understood correctly, we can't set the time in the controller via PPS in any way. This problem can be circumvented by continuing to use the millis() timer for this bus.

Are there any other problems that I forget about? Do all boilers have clocks?

fredlcore, 2022-12-18:
Yes, PPS is a (the) problem with this. And it might be that BSB-LAN is not connected to the heater (or the heater is off) when the microcontroller starts. Then we would have to take such scenarios into account as well. To me the pro is not "strong" enough to introduce such a "mix" of clock sources because most people have the logging interval at less than a (few) minute(s), and the microcontroller usually isn't started that often, so the interruptions shouldn't happen that often...

dukess, 2022-12-19:
Ok then i will leave it for future :)

closed 2022-12-19 by fredlcore