fredlcore/BSB-LAN #533: [BUG] bsb-lan fails to close http connection after delivering home page

DE-cr, 2022-12-19:
**BSB-LAN Version**
3.0.3-202212094171550 (github head version from 2022-12-19 07:51)

**Architecture**
ESP32 NodeMCU

**Bus system**
BSB

**Describe the bug**
On my system, the http connection is not closed properly after completely delivering the home page (http://bsb-lan/) contents (including the final `\</html\>\r\n\r\n`).
After a while, the client then complains (Firefox: "Fehler: Verbindung unterbrochen [...]") or just hangs (wget).
_This only seems to happen when "check for updates" is enabled!_

fredlcore, 2022-12-19:
Can you try and change line 110 in print_webpage.h from
flushToWebClient();
to
forcedflushToWebClient();
and see if the problem persists?

DE-cr, 2022-12-19:
Thank you for the quick reply - even with this less-than-substantial bug!
I've changed what you've suggested: The problem persists.

DE-cr, 2022-12-19:
Here's the WireShark (Linux) protocol: 
[2022-12-19_18-27_-cr_wireshark.zip](https://github.com/fredlcore/BSB-LAN/files/10261496/2022-12-19_18-27_-cr_wireshark.zip)

dukess, 2022-12-20:
BSB-LAN flush all bytes to client browser but doesn't connection closing after ~1000 msec.
May it rebooted in this time for some reason?

DE-cr, 2022-12-21:
Yes, that's it! Uptime in /C shows that it does reboot after delivering / contents. But why?

fredlcore, 2022-12-21:
What does SerMo log say? Is there a crash?

DE-cr, 2022-12-21:
There's no SerMo: The unit sits inside the boiler casing.
Should I try telnet monitoring?

fredlcore, 2022-12-21:
No point. In case there is a crash, how should the crash message be sent to the telnet client?

DE-cr, 2022-12-21:
@fredlcore, could you please remove the bogus `9` from http://bsb-lan.de/bsb-version.h and let me see if the problem persists?

fredlcore, 2022-12-21:
Why should this have anything to do with it and why don't you remove it yourself locally and test it then?

DE-cr, 2022-12-21:
The problem could be writing beyond the end of `version_number[15]` in `include/print_webpage.h: webPrintSite()`.
The version check in calling bsb-lan's root page reads http://bsb-lan.de/bsb-version.h, not my local copy.

DE-cr, 2022-12-21:
I've included a fix in my still open PR  https://github.com/fredlcore/BSB-LAN/pull/499

DE-cr, 2022-12-21:
Btw, this also explains a slightly different phenomenon I had been experiencing lately:
Instead of memorizing /K, /DG, etc. (OK, I've got these two memorized now), I usually start my http sessions with my bsb-lan unit via http://bsb-lan, and then use the menus. For the last couple of weeks, (quickly) selecting one of the menu entries after loading http://bsb-lan lead to a "server cannot be reached" message - which I hadn't worried much about, as a subsequent F5=refresh loaded the desired page alright. I did not open a bug report for that because I thought that it could have been due to a weak WLAN connection in my network. Now I know that the server could not be reached because it was rebooting at those times. Of course this only happens upon loading the bsb-lan http root page if the "newer version available?" check is activated - and `COMPILETIME` in http://bsb-lan.de/bsb-version.h contains one digit too many.

fredlcore, 2022-12-21:
Thanks for spotting this, I wouldn't have thought about this if I had had time to start bugfixing now. No need for a fix as this number will be overwritten with a correct COMPILETIME value once I push the next version number (which I might do before the holidays, as soon as I have confirmation on some other matter).

closed 2022-12-21 by fredlcore