fredlcore/BSB-LAN #540: [BUG]

tiger42, 2023-02-18:
**BSB-LAN Version**
Commit hash: 8a1ec49557618c35d9a0c0946108464995afc37e (compiletime: 20230217224441)

**Architecture**
Arduino Due with W5500 ethernet shield

**Bus system**
BSB

**Describe the bug**
Since the version of 02/17 BSB-LAN does not work for me any longer. If I rollback to the previous version of Feb, 4th (commit hash f3e6deb5d008cb9833725f4ebeeee9a8a75f52b8), everything works again. The error manifests itself in the fact that the Arduino just stops responding shortly after startup. The serial monitor output looks like this:
```
Reading EEPROM
Reading done.
EEPROM schema v.10 Program schema v.10
Address EEPROM option 0: 0
Address EEPROM option 1: 1
... (shortened)
Address EEPROM option 62: 1506
Address EEPROM option 63: 1507
Size of cmdtbl: 14952
free RAM: 85307
BSB-LAN version: 3.0.4-20230217224441
3.0.4-20230217224441
Using RX/TX pins 19, 18
PPS settings:
Starting SD..failed
Waiting for DHCP address
192.168.0.95
255.255.255.0
192.168.0.100
Waiting 3 seconds to give Ethernet shield time to get ready...
Calculating free space on SD...0 MB free
Start network services
Device family: 97
Device variant: 100

Starting MDNS service with hostname BSB-LAN
Setup complete
Client ID: BSB-LAN
Will topic: bsb-lan/status
Connect to MQTT broker, updating will topic
Subscribed to topic 'bsb-lan'
Published status 'online' to topic 'bsb-lan/status'
DSP1->HEIZ QUR 8310.0
```
After that nothing happens and the Arduino is playing dead...

fredlcore, 2023-02-18:
I have added  bugfix, can you test this one and let me know if this fixes it?

tiger42, 2023-02-19:
Thanks, the bugfix is working! The Arduino is behaving normally again. I've done multiple reset cycles just to be sure that the system is coming up every time.

fredlcore, 2023-02-19:
Great, thanks a lot for testing it so quickliy!

closed 2023-02-19 by fredlcore