fredlcore/BSB-LAN #541: [BUG] ESP32 resets in a loop by watchdog

dwittenb, 2023-02-19:
**BSB-LAN Version**
3.0.4-20230217224441

**Architecture**
The architecture BSB-LAN is running on ESP32 Olimex EVB EA IND

**Bus system**
BSB-LAN is running on BSB

**Describe the bug**
After poweron or pressing the reset button on the olimex-board  it seems that the wachdogtimer resets the board after starting some queries. Have a look at the log at timestamp 20:05:05.573.

**Logs**
```
20:04:50.559 -> ets Jul 29 2019 12:21:46
20:04:50.559 -> 
20:04:50.559 -> rst:0x1 (POWERON_RESET),boot:0xb (HSPI_FLASH_BOOT)
20:04:50.621 -> invalid header: 0x00000000
20:04:50.667 -> invalid header: 0x00000000
20:04:50.702 -> invalid header: 0x00000000
20:04:50.735 -> invalid header: 0x00000000
20:04:50.780 -> invalid header: 0x00000000
20:04:50.825 -> invalid header: 0x00000000
20:04:50.873 -> invalid header: 0x00000000
20:04:50.906 -> invalid header: 0x00000000
20:04:50.950 -> ets Jul 29 2019 12:21:46
20:04:50.950 -> 
20:04:50.950 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
20:04:50.950 -> configsip: 0, SPIWP:0xee
20:04:50.950 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:04:50.950 -> mode:DIO, clock div:1
20:04:50.950 -> load:0x3fff0030,len:1184
20:04:50.950 -> load:0x40078000,len:12804
20:04:50.950 -> ho 0 tail 12 room 4
20:04:50.950 -> load:0x40080400,len:3032
20:04:50.950 -> entry 0x400805e4
20:04:51.294 -> Rffe5
20:04:51.339 -> Reading EEPROM
20:04:51.339 -> Using settings from config file
20:04:51.369 -> Reading done.
20:04:51.369 -> EEPROM schema v.10 Program schema v.10
20:04:51.369 -> Address EEPROM option 0: 0
20:04:51.369 -> Address EEPROM option 1: 1
20:04:51.369 -> Address EEPROM option 2: 2
20:04:51.369 -> Address EEPROM option 3: 6
20:04:51.369 -> Address EEPROM option 4: 226
20:04:51.369 -> Address EEPROM option 5: 306
20:04:51.369 -> Address EEPROM option 6: 490
20:04:51.369 -> Address EEPROM option 7: 491
20:04:51.369 -> Address EEPROM option 8: 492
20:04:51.369 -> Address EEPROM option 9: 493
20:04:51.369 -> Address EEPROM option 10: 494
20:04:51.369 -> Address EEPROM option 11: 495
20:04:51.402 -> Address EEPROM option 12: 499
20:04:51.402 -> Address EEPROM option 13: 659
20:04:51.402 -> Address EEPROM option 14: 663
20:04:51.402 -> Address EEPROM option 15: 823
20:04:51.402 -> Address EEPROM option 16: 829
20:04:51.402 -> Address EEPROM option 17: 830
20:04:51.402 -> Address EEPROM option 18: 834
20:04:51.402 -> Address EEPROM option 19: 838
20:04:51.402 -> Address EEPROM option 20: 842
20:04:51.402 -> Address EEPROM option 21: 846
20:04:51.402 -> Address EEPROM option 22: 848
20:04:51.402 -> Address EEPROM option 23: 852
20:04:51.435 -> Address EEPROM option 24: 856
20:04:51.435 -> Address EEPROM option 25: 920
20:04:51.435 -> Address EEPROM option 26: 984
20:04:51.435 -> Address EEPROM option 27: 985
20:04:51.435 -> Address EEPROM option 28: 986
20:04:51.435 -> Address EEPROM option 29: 996
20:04:51.435 -> Address EEPROM option 30: 997
20:04:51.435 -> Address EEPROM option 31: 1157
20:04:51.435 -> Address EEPROM option 32: 1158
20:04:51.435 -> Address EEPROM option 33: 1162
20:04:51.435 -> Address EEPROM option 34: 1163
20:04:51.435 -> Address EEPROM option 35: 1164
20:04:51.469 -> Address EEPROM option 36: 1165
20:04:51.469 -> Address EEPROM option 37: 1169
20:04:51.469 -> Address EEPROM option 38: 1234
20:04:51.469 -> Address EEPROM option 39: 1299
20:04:51.469 -> Address EEPROM option 40: 1364
20:04:51.469 -> Address EEPROM option 41: 1396
20:04:51.469 -> Address EEPROM option 42: 1398
20:04:51.469 -> Address EEPROM option 43: 1399
20:04:51.469 -> Address EEPROM option 44: 1400
20:04:51.469 -> Address EEPROM option 45: 1401
20:04:51.469 -> Address EEPROM option 46: 1433
20:04:51.469 -> Address EEPROM option 47: 1497
20:04:51.502 -> Address EEPROM option 48: 1517
20:04:51.502 -> Address EEPROM option 49: 1537
20:04:51.502 -> Address EEPROM option 50: 1557
20:04:51.502 -> Address EEPROM option 51: 1558
20:04:51.502 -> Address EEPROM option 52: 1559
20:04:51.502 -> Address EEPROM option 53: 1560
20:04:51.502 -> Address EEPROM option 54: 1561
20:04:51.502 -> Address EEPROM option 55: 1562
20:04:51.502 -> Address EEPROM option 56: 1563
20:04:51.502 -> Address EEPROM option 57: 1595
20:04:51.502 -> Address EEPROM option 58: 1596
20:04:51.502 -> Address EEPROM option 59: 1597
20:04:51.542 -> Address EEPROM option 60: 1598
20:04:51.542 -> Address EEPROM option 61: 1600
20:04:51.542 -> Address EEPROM option 62: 1602
20:04:51.542 -> Address EEPROM option 63: 1603
20:04:51.542 -> Size of cmdtbl: 3668
20:04:51.542 -> free RAM: 282944
20:04:51.542 -> BSB-LAN version: 3.0.4-20230217224441
20:04:51.542 -> 3.0.4-20230217224441
20:04:51.542 -> Microcontroller: ESP32/Olimex EVB
20:04:51.542 -> Using RX/TX pins 36, 17
20:04:51.542 -> PPS settings:
20:04:51.542 -> Starting SD..Setting up WiFi interface......................
20:04:54.290 -> Attempting to connect to WPA SSID: FritzBox2..
20:04:57.324 -> You're connected to the network:
20:04:57.324 -> SSID: FritzBox2
20:04:57.324 -> IP Address: 192.168.3.38
20:04:57.324 -> Signal strength (RSSI): -42Bm
20:04:57.324 -> Waiting 3 seconds to give Ethernet shield time to get ready...
20:04:57.324 -> Calculating free space on SD...1377739 Bytes free
20:05:00.537 -> Start network services
20:05:01.342 -> Device family: 163
20:05:01.342 -> Device variant: 5
20:05:01.342 -> 
20:05:01.389 -> Starting MDNS service with hostname BSB-LAN
20:05:01.389 -> Setup complete
20:05:01.527 -> DSP1->HEIZ QUR      0D3D0519 
20:05:01.527 -> DC 8A 00 0B 06 3D 0D 05 19 4F 8C
20:05:01.606 -> HEIZ->DSP1 ANS      0D3D0519 00 06 86
20:05:01.606 -> DC 80 0A 0E 07 0D 3D 05 19 00 06 86 E4 63
20:05:01.722 -> RGT1->HEIZ QUR      11000C12 
20:05:01.722 -> DC 86 00 0B 06 00 11 0C 12 76 A5
20:05:01.802 -> HEIZ->RGT1 ERR      11000C12 07
20:05:01.802 -> DC 80 06 0C 08 11 00 0C 12 07 4B 0A
20:05:04.617 -> RGT1->HEIZ QUR 8700.0 ets Jul 29 2019 12:21:46
20:05:05.573 -> 
20:05:05.573 -> rst:0x8 (TG1WDT_SYS_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
20:05:05.573 -> configsip: 0, SPIWP:0xee
```

**Expected behavior**
The watchdog should not reset

fredlcore, 2023-02-19:
Please install the most recent version from GitHub, that should fix the problem.

closed 2023-02-19 by fredlcore

dwittenb, 2023-02-19:
Thank you for your quick response. Updated to new version / patch and problem is fixed.