fredlcore/BSB-LAN #550: [BUG] Just FYI because for old version: lines 2505 and 2504 in cmdtbl are mixed up

dukess, 2023-03-05:
In old BSB_LAN_defs.h two lines in cmdtbl array are mixed up:

```
{0x053D1AE1,  CAT_KESSEL,           VT_LITERPERMIN,   2505,  STR2505,  0,                    NULL,         FL_OEM,       DEV_ALL}, // Schwelle Durchflussdetektion
{0x053D0FE8,  CAT_KESSEL,           VT_SECONDS_SHORT5,2504,  STR2504,  0,                    NULL,         DEFAULT_FLAG, DEV_ALL}, // Baxi Luna Platinum+ Min Einschaltzeit Schalter

```
which cause infinite loop for category 34.
It should be
```
{0x053D0FE8,  CAT_KESSEL,           VT_SECONDS_SHORT5,2504,  STR2504,  0,                    NULL,         DEFAULT_FLAG, DEV_ALL}, // Baxi Luna Platinum+ Min Einschaltzeit Schalter
{0x053D1AE1,  CAT_KESSEL,           VT_LITERPERMIN,   2505,  STR2505,  0,                    NULL,         FL_OEM,       DEV_ALL}, // Schwelle Durchflussdetektion
```

fredlcore, 2023-03-05:
Yes, the problem is that I cannot change files in past releases - or is there a way to do so?

dukess, 2023-03-05:
You can create new branch from last commit with "old version" and leave it 
for bugfixes.

closed 2023-03-05 by fredlcore