fredlcore/BSB-LAN #564: [BUG] Crash on opening sensor page (Guru Meditation Error: Core  1 panic'ed)

deki, 2023-03-26:
**BSB-LAN Version**
3.1.5-20230326095055

**Architecture**
ESP32 NodeMCU

**Bus system**
BSB

**Describe the bug**
`Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.` and reboot happens when opening sensors page.

**To Reproduce**
Steps to reproduce the behavior:
Open sensors page /K44

**Logs**
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
READY
Reading EEPROM
Reading done.
EEPROM schema v.11 Program schema v.11
Address EEPROM option 0: 0
Address EEPROM option 1: 1
Address EEPROM option 2: 2
Address EEPROM option 3: 6
Address EEPROM option 4: 226
Address EEPROM option 5: 306
Address EEPROM option 6: 490
Address EEPROM option 7: 491
Address EEPROM option 8: 492
Address EEPROM option 9: 493
Address EEPROM option 10: 494
Address EEPROM option 11: 495
Address EEPROM option 12: 499
Address EEPROM option 13: 819
Address EEPROM option 14: 823
Address EEPROM option 15: 1143
Address EEPROM option 16: 1149
Address EEPROM option 17: 1150
Address EEPROM option 18: 1154
Address EEPROM option 19: 1158
Address EEPROM option 20: 1162
Address EEPROM option 21: 1166
Address EEPROM option 22: 1168
Address EEPROM option 23: 1172
Address EEPROM option 24: 1176
Address EEPROM option 25: 1240
Address EEPROM option 26: 1304
Address EEPROM option 27: 1305
Address EEPROM option 28: 1306
Address EEPROM option 29: 1316
Address EEPROM option 30: 1317
Address EEPROM option 31: 1637
Address EEPROM option 32: 1638
Address EEPROM option 33: 1642
Address EEPROM option 34: 1643
Address EEPROM option 35: 1644
Address EEPROM option 36: 1645
Address EEPROM option 37: 1649
Address EEPROM option 38: 1714
Address EEPROM option 39: 1779
Address EEPROM option 40: 1844
Address EEPROM option 41: 1876
Address EEPROM option 42: 1878
Address EEPROM option 43: 1879
Address EEPROM option 44: 1880
Address EEPROM option 45: 1881
Address EEPROM option 46: 1913
Address EEPROM option 47: 1977
Address EEPROM option 48: 1997
Address EEPROM option 49: 2017
Address EEPROM option 50: 2037
Address EEPROM option 51: 2038
Address EEPROM option 52: 2039
Address EEPROM option 53: 2040
Address EEPROM option 54: 2041
Address EEPROM option 55: 2042
Address EEPROM option 56: 2043
Address EEPROM option 57: 2075
Address EEPROM option 58: 2076
Address EEPROM option 59: 2077
Address EEPROM option 60: 2078
Address EEPROM option 61: 2080
Address EEPROM optionj
Address EEPROM option 63: 2083
Address EEPROM option 64: 2084
Power-saving activated.
Size of cmdtbl: 35812
free RAM: 286212
BSB-LAN version: 3.1.5-20230326095055
3.1.5-20230326095055
Microcontroller: ESP32/NodeMCU
Using RX/TX pins 16, 17
PPS settings:
Starting SD..Setting up WiFi interface..................................................
Attempting to connect to WPA SSID: MyWifi..E (17757) wifi:sta is connecting, return error
..
You're connected to the network:
SSID: MyWifi
IP Address: 192.168.xx.xx
Signal strength (RSSI): -44Bm
Waiting 3 seconds to give Ethernet shield time to get ready...
Calculating free space on SD...1433600 Bytes free
Start network services
Device family: 163
Device variant: 5

Starting MDNS service with hostname BSB-LAN
Setup complete
Client ID: BSB-LAN
Will topic: heating/status
Connect to MQTT broker, updating will topic
Subscribed to topic 'heating'
Published status 'online' to topic 'heating/status'
DSP1->HEIZ QUR 8310.0 Diagnose Erzeuger - Kesseltemperatur-Istwert: 
DC 8A 00 0B 06 3D 0D 05 19 4F 8C
HEIZ->DSP1 ANS 8310.0 Diagnose Erzeuger - Kesseltemperatur-Istwert: 34.5 °C
DC 80 0A 0E 07 0D 3D 05 19 00 08 A1 93 E9
DSP1->HEIZ QUR 8310.0 Diagnose Erzeuger - Kesseltemperatur-Istwert: 
DC 8A 00 0B 06 3D 0D 05 19 4F 8C
HEIZ->DSP1 ANS 8310.0 Diagnose Erzeuger - Kesseltemperatur-Istwert: 34.0 °C
DC 80 0A 0E 07 0D 3D 05 19 00 08 82 87 E8
DSP1->HEIZ QUR 8310.0 Diagnose Erzeuger - Kesseltemperatur-Istwert: 
DC 8A 00 0B 06 3D 0D 05 19 4F 8C
HEIZ->DSP1 ANS 8310.0 Diagnose Erzeuger - Kesseltemperatur-Istwert: 33.5 °C
DC 80 0A 0E 07 0D 3D 05 19 00 08 62 7A C6
Client ID: BSB-LAN
Will topic: heating/status
Connect to MQTT broker, updating will topic
Subscribed to topic 'heating'
Published status 'online' to topic 'heating/status'
GET / HTTP/1.1

GET /K44 HTTP/1.1

56
Virtual parameter 20000 queried. Table line 1243
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008ab6d  PS      : 0x00060d30  A0      : 0x800d6461  A1      : 0x3ffb1f00  
A2      : 0x3ffc5720  A3      : 0x00000000  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb1b40  
A10     : 0x3ffc5720  A11     : 0xffffffff  A12     : 0x00000006  A13     : 0x3ffb1d50  
A14     : 0x00000006  A15     : 0x3ffb1d50  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008ab6d  LEND    : 0x4008ab81  LCOUNT  : 0xffffffff  

Backtrace: 0x4008ab6a:0x3ffb1f00 0x400d645e:0x3ffb1f10 0x400d9a1d:0x3ffb1f30 0x400db7d3:0x3ffb1f70 0x400df00d:0x3ffb1fa0 0x400ee89d:0x3ffb2290

ELF file SHA256: f393d899c49b5b86

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
READY
```

**Expected behavior**
Sensor page opens instead of reboot

**Desktop (if applicable, please complete the following information):**
 - OS: Windows 10 Pro
 - Browser: Firefox
 - Browser version 111.0

fredlcore, 2023-03-26:
Thanks. please also post your BSB_LAN_config.h so I can recreate the issue here.

fredlcore, 2023-03-26:
And please also call the following parameters directly via the URL http://bsb-lan/PARAMETERNUMBER
20000
20006
20050
20500
20700

deki, 2023-03-26:
```
/************************************************************************************/
/************************************************************************************/
/* Settings -   BEGIN                                                               */
/************************************************************************************/
/************************************************************************************/

// Upon first installation, rename this file from BSB_lan_config.h.default to BSB_lan_config.h and adjust settings accordingly
// Users who still use Arduino Mega2560 boards instead of Arduin Due: Please have a look at the very end of this configuration
// where certain modules are disabled by default due to limited flash memory.

/* Select language; so far German is the most complete, with English following.
 * Available languages are: Czech (CS), German (DE), Danish (DA), English (EN), Spanish (ES), Finnish (FI),
 * French (FR), Greek (EL), Hungarian (HU), Italian (IT), Dutch (NL), Polish (PL), Russian (RU), Swedish (SV),
 * Slovenian (SI) and Turkish (TR).
 * Incomplete languages will automatically be filled up with English translations first, and if no English translation
 * is available, fallback will take place to German.
*/
#define LANG DE

/*
Allow to initialize program configuration by reading settings from EEPROM
byte UseEEPROM = 0; // Configuration is read from this config file.
                    // Configuration can be stored in EEPROM but will not used while UseEEPROM is zero.
                    // Set zero for fallback startup in case EEPROM configuration is broken.
byte UseEEPROM = 1; // Configuration will be read from EEPROM. This is the default.
*/
byte UseEEPROM = 1;

/*
 *  Enter a MAC address, found either on the EthernetShield or use the one below.
 *  Change this if you have more than one BSB-LAN adapter in your LAN, so that there aren't any address conflicts.
 *  MAC address here only affects the LAN shield, it doesn't apply to the WiFi-ESP-solution.
*/
byte mac[6] = { 0x00, 0x80, 0x41, 0x19, 0x69, 0x90 };

/*
 * Initialize the Ethernet server library
 * with the IP address and port you want to use
 * (port 80 is default for HTTP):
*/

uint16_t HTTPPort = 80;
bool useDHCP = true;                 // Set to false if you want to use a fixed IP.
byte ip_addr[4] = {192,168,178,88};     // Please note the commas instead of dots!!!  Set useDHCP to true if you want DHCP.
byte gateway_addr[4] = {192,168,178,1}; // Gateway address. This is usually your router's IP address. Please note the commas instead of dots!!! Ignored if first value is 0.
byte dns_addr[4] = {192,168,178,1};     // DNS server. Please note the commas instead of dots!!! Ignored if first value is 0.
byte subnet_addr[4] = {255,255,255,0};  // Subnet address. Please use commas instead of dots!!! Ignored if first value is 0.

#define WIFI                          // Activate this definement if you want to use WiFi. Note: MAC address can't be set individually.
char wifi_ssid[32] = "MyWifi"; // enter your WiFi network name (SSID) here
char wifi_pass[64] = "MyPassword";// enter your WiFi password here
#define WIFI_SPI_SS_PIN 12              // defines SPI-SS pin for Arduino-ESP8266 connection

#define MDNS_SUPPORT                   // Advertises the hostname in the local network. Disable this if you don't want your device to be found under this name in your network.
char mDNS_hostname[32] = "BSB-LAN";

#define DEBUG         // Compile with verbose DEBUG module if defined
byte debug_mode = 1;  // Debug mode: 0 - disabled, 1 - send debug messages to serial interface, 2 - send debug messages to telnet client
byte verbose = 1;     // If set to 1, all messages on the bus are printed to debug interface
byte monitor = 0;     // Bus monitor mode. This is only necessary for in-depth debug sessions.
bool show_unknown = true; // true - show all parameters, false - hide unknown parameters from web display (parameters will still be queried and therefore take time!)
//#define DEVELOPER_DEBUG //Very verbose debug output. This option will dramatically slow down MCU performance. Useful for developers not for users.

/* SECURITY OPTIONS
 * There are several options to control and protect access to your heating system. However, keep
 * in mind, that even activating all three options are no guarantee that a versatile intruder with
 * access to your (W)LAN won't be able to gain access. In any case, no encryption of data streams
 * is provided from the Arduino itself. Use VPN or a SSL proxy if that is a must for you and connect
 * the Arduino wired to the VPN server or SSL proxy. On the other hand, someone with this amount
[... 221 lines cut ...]
// Pins on Mega can be (Digital) 2, 3, 18, 19, 20, 21
// On Due any Digital pins can be selected excluding 12, 16-21, 31, 33, 53.
// Make sure you aren't using pins which are already in use for sensors (default: 2, 3, 7) or change them accordingly.
#define BUTTONS
uint8_t button_on_pin[4] = {0, 0, 0, 0}; //Order: TWW push, presence ROOM1, presence ROOM2, presence ROOM3

// Variables for future use:
// Compile room unit replacement extension
#define ROOM_UNIT
byte UdpIP[4] = {0,0,0,0}; // (destination IP address for sending UDP packets to, the room unit replacement of FHEM user @fabulous uses that)
uint16_t UdpDelay = 15;    // (interval in seconds to send UDP packets)

// Compile off-site logger extension
#define OFF_SITE_LOGGER
[... cut ...]
uint32_t destinationDelay = 84600;  // interval in seconds to send values

/************************************************************************************/
/************************************************************************************/
/* Settings -   END                                                                 */
/************************************************************************************/
/************************************************************************************/

```

deki, 2023-03-26:
Calling parameters via URL doesn't work as well.

DE-cr, 2023-03-26:
I was just to about to report the same thing, then I noticed this bug report here.
My setup (WLAN SSID and password erased) for this post):
[image]

DE-cr, 2023-03-26:
> And please also call the following parameters directly via the URL http://bsb-lan/PARAMETERNUMBER 20000 20006 20050 20500 20700

20050 works as expected, the others crash my system.

fredlcore, 2023-03-26:
@DE-cr: I can'r reproduce this here, could you add some kind of debug messages in lines 4100ff (queryVirtualPrognr) and see if the variable `line` holds an invalid value or if the function call to `recognizeVirtualFunctionGroup` creates the problem? I assume that the error happens somewhere in these lines.

DE-cr, 2023-03-26:
I'll see what I can see. ;)
In the meantime, I can report that config changing via /CI (to enable telnet debugging) also crashed my system.

DE-cr, 2023-03-26:
A printFmtToDebug() call right in the first line of queryVirtualPrognr() does not give me any output, so I guess the problem is before that.
Where is /Kn (with n being the category number) handled? I've only found
```
        // list categories
        if (p[1]=='K' && !isdigit(p[2])) {
```
i.e. /K without n.

fredlcore, 2023-03-26:
That would be line 6869 ff. and then 6907.

This error with the sensors really sounds familiar and I'm sure we fixed it or at least figured out which settings had to be made, but I can't find it right now, will keep searching...

fredlcore, 2023-03-26:
Can _both_ of you please just install the most recent version from master and only change your network settings in BSB_LAN_config.h, please, and start out with UseEEPROM = 0. If then there is no error, set UseEEPROM = 1 and test again. If that also works, please adjust your settings one-by-one, save the settings and access the sensors page again. That would be the only way to figure out where the problem lies.
I have downloaded the master repo and just changed the network settings and it worked without problems.

DE-cr, 2023-03-27:
Downloaded BSB-LAN-master, changed default config.h to UseEEPROM=0, enabled WiFi, set SSID and password, compiled and uploaded to ESP32, called http://bsb-lan/ (ok), called sensors (/K28) => crash.

DE-cr, 2023-03-27:
> That would be line 6869 ff. and then 6907.

Thanks, I thought that would be in the JSON handling part only.

Some debug code at 6869 ff ...
```
            if (range[0]=='K') {
              uint8_t cat = atoi(&range[1]) * 2; // * 2 - two columns in ENUM_CAT_NR table
printFmtToDebug("range=%s,cat=%d,sizeof(*ENUM_CAT_NR)=%ld,sizeof(ENUM_CAT_NR=%ld),ENUM_CAT_NR[%ld]\n",range,cat,sizeof(*ENUM_CAT_NR),sizeof(ENUM_CAT_NR),sizeof(ENUM_CAT_NR)/sizeof(*ENUM_CAT_NR));//-cr
              start = ENUM_CAT_NR[cat];
              end = ENUM_CAT_NR[cat+1];
printFmtToDebug("start=%g,end=%g\n",start,end);//-cr
              if (end > ENUM_CAT_NR[cat+2]) {
                end = ENUM_CAT_NR[cat+2]-1;
              }
printFmtToDebug("end=%g\n",end);//-cr
            } else {
```
... gives me:
```
GET /K28 HTTP/1.1

range=K28,cat=56,sizeof(*ENUM_CAT_NR)=4,sizeof(ENUM_CAT_NR=232),ENUM_CAT_NR[58]
start=20000,end=20899
end=-211.125
30
Virtual parameter 20000 queried. Table line 447
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```
=> 
```
              if (end > ENUM_CAT_NR[cat+2]) {
                end = ENUM_CAT_NR[cat+2]-1;
              }
```
... reads past the end of ENUM_CAT_NR!

DE-cr, 2023-03-27:
... as would a user's manually calling /K99, e.g. ;)

fredlcore, 2023-03-27:
Ok, this line 
```
end = ENUM_CAT_NR[cat+2]-1;
```
obviously makes no sense, but I still don't get it why it works here without a problem.
Anyway, I'll add a check for maximum category number, that should fix it.

fredlcore, 2023-03-27:
Sorry, it does make sense because some devices generate parameter lists with overlapping parameter numbers. So taking the start of the next category minus one as the end of the current category if it overlaps into the next is already a safety check.

DE-cr, 2023-03-27:
Even with overlapping parameter lists: Couldn't this be removed?:
```
              if (end > ENUM_CAT_NR[cat+2]) {
                end = ENUM_CAT_NR[cat+2]-1;
              }
```

DE-cr, 2023-03-27:
Will do, when I'm back at home.

DE-cr, 2023-03-27:
I can tell you now that my system crashes in the last line of this code excerpt:
```
char *build_pvalstr(bool extended) {
  int len = 0;
  outBuf[len] = 0;
  if (extended && decodedTelegram.error != 257) {
    len+=sprintf_P(outBuf, PSTR("%g "), decodedTelegram.prognr);

    len+=strlen(strcpy_PF(outBuf + len, decodedTelegram.catdescaddr));
```
...with decodedTelegram.catdescaddr==0 and decodedTelegram.prognr==20000 and decodedTelegram.cat==30.

Figuring out why .catdescaddr doesn't point to a C string here might be easier for you than for me.
Otherwise I could look at it again tomorrow evening.

fredlcore, 2023-03-27:
Thanks, that resembles the problem I think I had fixed some weeks ago, but maybe at a different location in code. I'll see what I can do...

fredlcore, 2023-03-27:
Hm, if decodedTelegram.cat is still 30, then that's the problem. It has to be 28. 30 is again beyond the size of the category ENUM.
I wonder where the `30` in the output of your debug code comes from. I don't see it in your code and neither in the general code. But the problem lies there where the category gets bumped up by two.

fredlcore, 2023-03-27:
Ok, I think I found it: https://github.com/fredlcore/BSB-LAN/commit/34707cd653c1a7ff308f8e67c83e5fcc90047bd7
Let me know if that really fixes the problem.

DE-cr, 2023-03-28:
No more crashes on my system. :+1:

deki, 2023-03-28:
Great to see the progress. I'm away from home right now but will try on the weekend.

DE-cr, 2023-03-28:
/CI: enabled telnet debugging (this is where my system crashed once, as reported above), disabled debugging again => works as expected

fredlcore, 2023-03-28:
Great, so I'll close this as fixed now...

closed 2023-03-28 by fredlcore