fredlcore/BSB-LAN #565: won't compile esp32_save_energy

stonie81, 2023-03-26:
`Arduino: 1.8.19 (Mac OS X), Board: "OLIMEX ESP32-EVB, 80MHz, 115200, Default, None, Disabled"

In file included from /Users/stonie81/Downloads/BSB-LAN/BSB_LAN/BSB_LAN.ino:876:
BSB_LAN_EEPROMconfig.h:177:122: error: 'esp32_save_energy' was not declared in this scope
   {CF_ESP32_ENERGY_SAVE,11,CCAT_GENERAL,  CPI_SWITCH,    CDT_BYTE,           OPT_FL_ADVANCED, CF_ENERGY_SAVE_TXT, sizeof(esp32_save_energy)}, //need reboot
                                                                                                                          ^~~~~~~~~~~~~~~~~
/Users/stonie81/Downloads/BSB-LAN/BSB_LAN/BSB_LAN.ino: In function 'void setup()':
BSB_LAN:7516:57: error: 'esp32_save_energy' was not declared in this scope
   registerConfigVariable(CF_ESP32_ENERGY_SAVE, (byte *)&esp32_save_energy);
                                                         ^~~~~~~~~~~~~~~~~
Mehrere Bibliotheken wurden für "WiFiUdp.h" gefunden
 Benutzt: /Users/stonie81/Library/Arduino15/packages/esp32/hardware/esp32/2.0.6/libraries/WiFi
 Nicht benutzt: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
'esp32_save_energy' was not declared in this scope

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
`
actual git version with commit [426a018]

fredlcore, 2023-03-26:
There is actually a template for bug reports that you are being offered. Why didn't you follow the instructions there but rather just copy-pasted an error message here with no further context?
And no, you are not using the actual git version, because the error results in an outdated BSB_LAN_config.h that you use. Copy over the current BSB_LAN_config.h.default, make the necessary adjustments and flash again.

closed 2023-03-26 by fredlcore

stonie81, 2023-03-26:
BSB-LAN Version
Master

Architecture
Arduino IDE 2.0.4

Bus system
Olimes ESP32 EVB

Describe the bug
Compile Error

To Reproduce
Compile

hmm so maybe I'm too dumb, but I used `git clone https://github.com/fredlcore/BSB-LAN` and the error remains. How I have to use git to get the version that works? `git clone` an `git pull` should be all I need.

fredlcore, 2023-03-26:
Of course the error remains. Do you expect a git pull to overwrite your configuration in BSB_LAN_config.h?

stonie81, 2023-03-26:
Sorry I'm not the programming guy, I'm just a user. So I `diff` it and update accordingly.
Maybe the next time be a little more helpful not pissed, thanks.

fredlcore, 2023-03-26:
No problem with being a user, but if people don't read the manual and then post bug reports without using the corresponding template which actually mentions to check this matter first before reporting, then I'm not too amused to spend time on such matters.
if you did a `diff`, it should have shown a difference between your old BSB_LAN_config.h.default and the new one, and Section 2.1.3 which deals specifically with updates clearly highlights:
> The existing and (if necessary) adapted file BSB_LAN_config.h can usually be taken over when updating to a newer version, but it is advisable to use the current file BSB_LAN_config.h.default instead of the existing file BSB_LAN_config.h. To do this, the file BSB_LAN_config.h.default must be renamed as before and, if necessary, adapted to the previous settings. This way you can be sure that you have made a complete update of the BSB-LAN software.

So please forgive me if my replies remain rather short under these circumstances, thanks.

stonie81, 2023-03-26:
I now used `diff` 
I understand that sucks, and I'm sorry I don't have it on my mind, that every time I `git pull` I should `diff` my config. But I am a guy who's willing to learn.
A hint to chapter 2.1.3 in the first place would have been nice. That's all..

fredlcore, 2023-03-26:
The "hint" is already in the proper template for the bug report which reads:
` Please make sure that you have installed the most recent version from the master repository and also updated your BSB_LAN_config.h accordingly.`
In this case I was able to identify that this would be the reason for the error because this function was just added recently and thus I still had it in the back of my mind. But if a user updates after a year or so and gives me a bug report like the one above, I have a hard time telling what the problem could be. That's why a proper bug report would have been a nice way to show that you do your part in solving the problem, that's all.

stonie81, 2023-03-26:
OK I'm too dumb to bugreport 👍