fredlcore/BSB-LAN #573: [BUG] WiFi problems on OpenWrt

WMP, 2023-04-10:
**BSB-LAN Version**
80c792e8f12cb1fec1f0ae4a9a856c4d014be27be1 (3.1 with WiFi fixes)

**Architecture**
ESP32 NodeMCU 30pin

**Describe the bug**
OnTp-Link Archer C7 AC1750 and  Openwrt 19.07 wifi works when i set Encryption: mixed WPA/WPA2 PSK, cipehers: auto. After upgrade to OpenWrt 22.03 i must set:
1. Encryption: mixed WPA/WPA2 PSK
2. Ciphers: AES
3.
```
//  esp_wifi_set_bandwidth(WIFI_IF_STA, WIFI_BW_HT20);  // W.Bra. 23.03.23 HT20 - reduce bandwidth from 40 to 20 MHz. In 2.4MHz networks, this will increase speed and stability most of the time, or will at worst result in a roughly 10% decrease in transmission speed.
// Enable the following command if you have problems connecting to your WiFi router. The tradeoff are (possibly significantly) lower transmission rates.
  esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N);  // W.Bra. 23.03.23 LR
```

**What i trying and dont works**:
1. a) Encryption: WPA2 PSK
b) Ciphers: AES
c) 
```
//  esp_wifi_set_bandwidth(WIFI_IF_STA, WIFI_BW_HT20);  // W.Bra. 23.03.23 HT20 - reduce bandwidth from 40 to 20 MHz. In 2.4MHz networks, this will increase speed and stability most of the time, or will at worst result in a roughly 10% decrease in transmission speed.
// Enable the following command if you have problems connecting to your WiFi router. The tradeoff are (possibly significantly) lower transmission rates.
  esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N);  // W.Bra. 23.03.23 LR
```
2. a) Encryption: WPA2 PSK
b) Ciphers: auto
c) 
```
//  esp_wifi_set_bandwidth(WIFI_IF_STA, WIFI_BW_HT20);  // W.Bra. 23.03.23 HT20 - reduce bandwidth from 40 to 20 MHz. In 2.4MHz networks, this will increase speed and stability most of the time, or will at worst result in a roughly 10% decrease in transmission speed.
// Enable the following command if you have problems connecting to your WiFi router. The tradeoff are (possibly significantly) lower transmission rates.
  esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N);  // W.Bra. 23.03.23 LR
```

3. a) Encryption: WPA2 PSK OR mixed WPA/WPA2 PSK
b) Ciphers: auto OR AES
c) 
```
//  esp_wifi_set_bandwidth(WIFI_IF_STA, WIFI_BW_HT20);  // W.Bra. 23.03.23 HT20 - reduce bandwidth from 40 to 20 MHz. In 2.4MHz networks, this will increase speed and stability most of the time, or will at worst result in a roughly 10% decrease in transmission speed.
// Enable the following command if you have problems connecting to your WiFi router. The tradeoff are (possibly significantly) lower transmission rates.
  esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_LR);  // W.Bra. 23.03.23 LR
```

4. a) Encryption: WPA2 PSK OR mixed WPA/WPA2 PSK
b) Ciphers: auto OR AES
c) 
```
esp_wifi_set_bandwidth(WIFI_IF_STA, WIFI_BW_HT20);  // W.Bra. 23.03.23 HT20 - reduce bandwidth from 40 to 20 MHz. In 2.4MHz networks, this will increase speed and stability most of the time, or will at worst result in a roughly 10% decrease in transmission speed.
// Enable the following command if you have problems connecting to your WiFi router. The tradeoff are (possibly significantly) lower transmission rates.
  esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_LR);  // W.Bra. 23.03.23 LR
```

5. a) Encryption: OPEN or WEP
b) Ciphers: auto
c) 
```
esp_wifi_set_bandwidth(WIFI_IF_STA, WIFI_BW_HT20);  // W.Bra. 23.03.23 HT20 - reduce bandwidth from 40 to 20 MHz. In 2.4MHz networks, this will increase speed and stability most of the time, or will at worst result in a roughly 10% decrease in transmission speed.
// Enable the following command if you have problems connecting to your WiFi router. The tradeoff are (possibly significantly) lower transmission rates.
  esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_LR);  // W.Bra. 23.03.23 LR
```

BraweProg, 2023-04-10:
"WIFI_PROTOCOL_LR" is the long range protocol from espressif and only works, if both partners are of ESP32-WROOM type. Makes only sense, if the distance is too long for the other protocols.

fredlcore, 2023-04-10:
I'm surprised that OpenWRT is creating these problems (@BraweProg: You were also using OpenWRT, correct?), but there are no problems when using a Fritz!Box, for example. While I would like to support the most compatible WiFi-Setting for all routers, I have no other router than a FB myself, so I cannot test any of this, so there is no way for me to help here except adding code that works for you as well as for me and then include such a solution into BSB-LAN.

If you say that you have to enable mixed WPA/WPA2 PSK in order to get it to work, this probably means that the connection falls back to WPA, right? But then I don't understand why option 4 doesn't work which is also mixed WPA/WPA2 PSK.

But as @BraweProg mentioned, remove WIFI_PROTOCOL_LR first. I will remove this probably even from in the commented-out form because the benefits probably do not outweigh the potential problems. Users with a long distance to the router should rather use an additional WiFi repeater.

fredlcore, 2023-04-11:
Thanks for the detailed infos, unfortunately I'm not a WiFi-tech kind of guy. However, I find it strange that a router would completely abandon an older, but still functional transmission standard. This would render quite a few older devices in my household unusable. Maybe in line with this, the ESP32 is offering a default cipher that OpenWRT does not accept (anymore) and doesn't try another one by itself.

In any case, it would be great if you and @BraweProg could figure out which setting works fastest with current OpenWRT routers. Priority here is, as mentioned before, transmission speed due to the already rather slow transmission speed for graph display etc.

BraweProg, 2023-04-11:
Unfortunately I'm not using openwrt nor have any experience with that. My router is a Fritzbox, but, because of the too long distance between the router and BSB-LAN I have installed an ESP-IOT-Bridge (https://github.com/espressif/esp-iot-bridge) in between. Both, the bridge and BSB-LAN are basing on ESP32-EVB with ESP32Wroom chip. This, and only this one, supports the LR protocol. NodeMCU has the same chip. If you are running openwrt on a board with another MCU than ESP32Wroom, LR communication would not work.
Principially the wifi STA initialization applies for the AP mode as well. So, if you are choosing LR mode for STA, you would have to set it back again for the AP mode. Otherwise you cannot establish a connection to the AP from another device. That is the reason for:
esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N);

If you don't use the LR initialization, you would not need to go back to 11bgn. LR did not solve my wifi distance problem, so currently I'm no longer using it. With 11bgn I'm getting -69dBm from the bridge, which is not much and it is slow. The graphical log for 2 days and a sample rate of 600s takes 2 minutes (through the bridge).
Did you test the AP behaviour and the beacon values? I'm getting b11/g54/n72 and WPA2(AES/TKIP) 20Mhz. Disadvantage with the AP is that the BSB-LAN cannot draw a graphic log due to the missing internet connection.

fredlcore, 2023-04-11:
Ok, I'll then remove the LR settings from the code if it is only used in this configuration and will keep the 20MHz in place.

WMP, 2023-04-12:
I, on the other hand, have a server room in a boiler room, so the distance from ap to bsb-lan is 3m. I am not able to test the speed, at my place everything is very fast. In theory, the younger the wifi protocol and the higher the MHz range the faster.

I am most interested in encryption, why only AES works. Where is the code that is responsible for encryption? Bsb-lan uses esp-idf or Arduino?

fredlcore, 2023-04-12:
It uses the Arduino-Framework of ESP. BSB-LAN just uses the standard libraries and function calls, so such specific details would have to be addressed elsewhere.

fredlcore, 2023-05-12:
WiFi security is set to WPA2 (CCMP), which is one of the two options available on the FritzBox. I'm not sure why I don't use the better one, WPA2 + WPA3, could be because of the ESP32, but also some of the other, sometimes quite old devices I have running here in my home...

closed 2024-02-26 by fredlcore