fredlcore/BSB-LAN #576: [BUG] brute-force.pl script does not query the right HEX code

nrobadey, 2023-04-22:
Hardware: ESP32
Version: 3.1.15-20230402082920

When the "brute-force.pl scriptis run, it seems to repetitively query the same variable, and not the advertised HEX code.
Telnet log:
GET /4444/Y06,0x053D0000 HTTP/1.1
LAN->HEIZ QUR  0.0 Uhrzeit und Datum - Datum/Zeit: 
78 0E 00 42 C0 02 00 14 06 3D 05 00 0B F4 FD
HEIZ->LAN ANS  0.0 Uhrzeit und Datum - Datum/Zeit: 22.04.2023 09:56:53
78 17 42 00 0C 02 00 14 07 05 3D 00 0B 00 7B 04 16 06 09 38 35 01 EC 6E
#0: 22.04.2023 09:56:53
GET /4444/Y06,0x053D0000 HTTP/1.1
LAN->HEIZ QUR  0.0 Uhrzeit und Datum - Datum/Zeit: 
78 0E 00 42 C0 02 00 14 06 3D 05 00 0B F4 FD
HEIZ->LAN ANS  0.0 Uhrzeit und Datum - Datum/Zeit: 22.04.2023 09:56:54
78 17 42 00 0C 02 00 14 07 05 3D 00 0B 00 7B 04 16 06 09 38 36 01 EC 6F
#0: 22.04.2023 09:56:54
GET /4444/Y06,0x053D0000 HTTP/1.1
LAN->HEIZ QUR  0.0 Uhrzeit und Datum - Datum/Zeit: 
78 0E 00 42 C0 02 00 14 06 3D 05 00 0B F4 FD
HEIZ->LAN ANS  0.0 Uhrzeit und Datum - Datum/Zeit: 22.04.2023 09:56:54
78 17 42 00 0C 02 00 14 07 05 3D 00 0B 00 7B 04 16 06 09 38 36 01 EC 6F
#0: 22.04.2023 09:56:54
GET /4444/Y06,0x053D0001 HTTP/1.1
LAN->HEIZ QUR  0.0 Uhrzeit und Datum - Datum/Zeit: 
78 0E 00 42 C0 02 00 14 06 3D 05 00 0B F4 FD
HEIZ->LAN ANS  0.0 Uhrzeit und Datum - Datum/Zeit: 22.04.2023 09:56:55
78 17 42 00 0C 02 00 14 07 05 3D 00 0B 00 7B 04 16 06 09 38 37 01 EC 70
#0: 22.04.2023 09:56:55
GET /4444/Y06,0x053D0001 HTTP/1.1
LAN->HEIZ QUR  0.0 Uhrzeit und Datum - Datum/Zeit: 
78 0E 00 42 C0 02 00 14 06 3D 05 00 0B F4 FD
HEIZ->LAN ANS  0.0 Uhrzeit und Datum - Datum/Zeit: 22.04.2023 09:56:55
78 17 42 00 0C 02 00 14 07 05 3D 00 0B 00 7B 04 16 06 09 38 37 01 EC 70
#0: 22.04.2023 09:56:55
GET /4444/Y06,0x053D0001 HTTP/1.1
LAN->HEIZ QUR  0.0 Uhrzeit und Datum - Datum/Zeit: 
78 0E 00 42 C0 02 00 14 06 3D 05 00 0B F4 FD
HEIZ->LAN ANS  0.0 Uhrzeit und Datum - Datum/Zeit: 22.04.2023 09:56:56
78 17 42 00 0C 02 00 14 07 05 3D 00 0B 00 7B 04 16 06 09 38 38 01 EC 71
#0: 22.04.2023 09:56:56
GET /4444/Y06,0x053D0002 HTTP/1.1
LAN->HEIZ QUR  0.0 Uhrzeit und Datum - Datum/Zeit: 
78 0E 00 42 C0 02 00 14 06 3D 05 00 0B F4 FD
HEIZ->LAN ANS  0.0 Uhrzeit und Datum - Datum/Zeit: 22.04.2023 09:56:56
78 17 42 00 0C 02 00 14 07 05 3D 00 0B 00 7B 04 16 06 09 38 38 01 EC 71
#0: 22.04.2023 09:56:56
GET /4444/Y06,0x053D0002 HTTP/1.1
LAN->HEIZ QUR  0.0 Uhrzeit und Datum - Datum/Zeit: 
78 0E 00 42 C0 02 00 14 06 3D 05 00 0B F4 FD
HEIZ->LAN ANS  0.0 Uhrzeit und Datum - Datum/Zeit: 22.04.2023 09:56:56
78 17 42 00 0C 02 00 14 07 05 3D 00 0B 00 7B 04 16 06 09 38 38 01 EC 71
#0: 22.04.2023 09:56:56
GET /4444/Y06,0x053D0002 HTTP/1.1
LAN->HEIZ QUR  0.0 Uhrzeit und Datum - Datum/Zeit: 
78 0E 00 42 C0 02 00 14 06 3D 05 00 0B F4 FD
HEIZ->LAN ANS  0.0 Uhrzeit und Datum - Datum/Zeit: 22.04.2023 09:56:57
78 17 42 00 0C 02 00 14 07 05 3D 00 0B 00 7B 04 16 06 09 38 39 01 EC 72
#0: 22.04.2023 09:56:57
GET /4444/Y06,0x053D0003 HTTP/1.1
LAN->HEIZ QUR  0.0 Uhrzeit und Datum - Datum/Zeit: 
78 0E 00 42 C0 02 00 14 06 3D 05 00 0B F4 FD
HEIZ->LAN ANS  0.0 Uhrzeit und Datum - Datum/Zeit: 22.04.2023 09:56:58
78 17 42 00 0C 02 00 14 07 05 3D 00 0B 00 7B 04 16 06 09 38 3A 01 EC 73
#0: 22.04.2023 09:56:58
GET /4444/Y06,0x053D0003 HTTP/1.1
LAN->HEIZ QUR  0.0 Uhrzeit und Datum - Datum/Zeit: 
78 0E 00 42 C0 02 00 14 06 3D 05 00 0B F4 FD
HEIZ->LAN ANS  0.0 Uhrzeit und Datum - Datum/Zeit: 22.04.2023 09:56:58
78 17 42 00 0C 02 00 14 07 05 3D 00 0B 00 7B 04 16 06 09 38 3A 01 EC 73
[... 144 lines cut ...]
#0: 22.04.2023 09:57:10
GET /4444/Y06,0x053D000B HTTP/1.1
LAN->HEIZ QUR  0.0 Uhrzeit und Datum - Datum/Zeit: 
78 0E 00 42 C0 02 00 14 06 3D 05 00 0B F4 FD
HEIZ->LAN ANS  0.0 Uhrzeit und Datum - Datum/Zeit: 22.04.2023 09:57:11
78 17 42 00 0C 02 00 14 07 05 3D 00 0B 00 7B 04 16 06 09 39 0B 01 EC 45
#0: 22.04.2023 09:57:11
GET /4444/Y06,0x053D000C HTTP/1.1
LAN->HEIZ QUR  0.0 Uhrzeit und Datum - Datum/Zeit: 
78 0E 00 42 C0 02 00 14 06 3D 05 00 0B F4 FD
HEIZ->LAN ANS  0.0 Uhrzeit und Datum - Datum/Zeit: 22.04.2023 09:57:11
78 17 42 00 0C 02 00 14 07 05 3D 00 0B 00 7B 04 16 06 09 39 0B 01 EC 45
#0: 22.04.2023 09:57:11
GET /4444/Y06,0x053D000C HTTP/1.1
LAN->HEIZ QUR  0.0 Uhrzeit und Datum - Datum/Zeit: 
78 0E 00 42 C0 02 00 14 06 3D 05 00 0B F4 FD
HEIZ->LAN ANS  0.0 Uhrzeit und Datum - Datum/Zeit: 22.04.2023 09:57:11
78 17 42 00 0C 02 00 14 07 05 3D 00 0B 00 7B 04 16 06 09 39 0B 01 EC 45
#0: 22.04.2023 09:57:11
GET /4444/Y06,0x053D000C HTTP/1.1
LAN->HEIZ QUR  0.0 Uhrzeit und Datum - Datum/Zeit: 
78 0E 00 42 C0 02 00 14 06 3D 05 00 0B F4 FD
HEIZ->LAN ANS  0.0 Uhrzeit und Datum - Datum/Zeit: 22.04.2023 09:57:12
78 17 42 00 0C 02 00 14 07 05 3D 00 0B 00 7B 04 16 06 09 39 0C 01 EC 46
#0: 22.04.2023 09:57:12

fredlcore, 2023-04-22:
The brute-force.pl script was intended to work with versions lower than 2.0 and is no longer developed further since the findings were always more than doubtful. If you insist on using it, you have to install a version lower than 2.0.

closed 2023-04-22 by fredlcore

nrobadey, 2023-04-22:
Actually it seems to work when the "4444" at the end of the suggested URL is removed.

fredlcore, 2023-04-22:
Ok, good to know!