fredlcore/BSB-LAN #582: [BUG] Web-UI does not show any parameters other than "0 .. 4"

drfjordan, 2023-05-19:
**BSB-LAN Version** 3.1.16-20230509001116 (as shown in Web-UI)
Used current master from 20230519

**Architecture** Arduino Due

**Bus system** BSB

**Describe the bug**
Navigating to any link shown in "Heizfunktionen" shows only parameters of the time section:
[ As shown when calling the device with "K0" ]
0 Uhrzeit - Aktuelles Datum / Aktuelle Uhrzeit: 19.05.2023 12:01:30 
1 Uhrzeit - Uhrzeit: 12:01:30 
2 Uhrzeit - Datum 1: 19.05. 
3 Uhrzeit - Datum 2: 2023  

**To Reproduce**
Steps to reproduce the behavior:
1. Go to main web page of 'BSB LAN" device
2. Click on 'Heizungsfunktionen'
3. Click on any other presented link than 'Urzeit'
4. Always the output of section 0 (like page had been called with "/K<x>" (with x=0) 

[BSB_LAN_config.h.zip](https://github.com/fredlcore/BSB-LAN/files/11515849/BSB_LAN_config.h.zip)

closed 2023-05-19 by fredlcore

fredlcore, 2023-05-19:
Sorry, I closed this prematurely. There was a small but effective bug that prevented /Kx from working properly. This was now fixed in https://github.com/fredlcore/BSB-LAN/commit/f7ab5c5c23e760403cf3aad5a2a94c3213425d38 and should work fine now.
@dukess: I'm not sure whether this could affect other parts of the new code as well, but `parsingStringToParameter` only works if it starts with a number, so parsing for example `K17` would always evaluate zero which is why only parameters from category 0 were shown. Letting the parsing begin one character later did the quick fix for me, but maybe there is a better approach.

dukess, 2023-05-20:
@fredlcore your patch is elegant and sufficient.

P.S. I have been worried for a long time about why the categories are displayed by the "Kxx" command, while all other commands for working with parameters have the form "Y=xxxxx", and I could suggest breaking compatibility, but bringing the type of commands to uniformity. But is it worth the effort?

fredlcore, 2023-05-20:
This is actually on me: When Gero first started with the software, there was just the parameter range and the category command if I'm not mistaken. Later, I added the /G command to query GPIO pins and then also set them using /Gx=y. Which made sense because you are setting a pin to a value. With the logging command /L, things got more confused as I used commas in the /L,interval,parameter notation since the `=` didn't seem to make sense there. And then when the /J commands came up, I used /JK=17 instead of /JK17. Why? I don't know ;). With other URL commands it's likewise. Would it be better to streamline this? Yes. Would it be better for the users who would have to change their home automation software which sometimes has been running for years? No, because this is more than just adjusting BSB-LAN's config file. So I suggest we leave it in this not too perfect state at least until we have a real major breaking change (such as changing the platform or something like that). Until then I hope everybody can live with it ;)...