fredlcore/BSB-LAN #586: [BUG] v3.1.17 (3.2?) causes many timeouts for web-requests/url-commands

schlaubstar, 2023-06-06:
**BSB-LAN Version**
3.1.17-20230520012002

**Architecture**
ESP32 

**Bus system**
BSB

**Describe the bug**
When upgrading from v.3.1.15 to 3.1.17 (3.2 according to comments in Ino-file) there is suddenly a massive timeout issue, when using URL-commands to set parameters.

I use a RaspberryPI to query the real temperatures and emulate a room-unit, thus setting parameter 10000 every 3 minutes with the current temperature, using a python script. As I forgot to define a connection timeout my PI crashed due to many open connections, eating up available memory. Looking at the problem, I found out that BSB-Lan on my ESP32 is not replying to many of the URL-requests. I can not find out why but approx. 10% of the Requests do never get a response.

**To Reproduce**
Steps to reproduce the behavior:

Use e.g. this small python script with v3.15 and 3.17

```
#!/usr/bin/python 	
# -*- coding: utf-8 -*-
import requests, time 				

def test():
	requrl = "http://192.168.xxx.xx/I10000=21.09"
	jsonResp = requests.post(requrl, auth=('user', 'pass'), timeout=15.00)
	print("Response.status_code", jsonResp.status_code)

if __name__ == '__main__':
	for x in range(100):
		test()
		time.sleep(3)
```

**Logs**
with 3.15 you will get 100times:
Response.status_code 200

with 3.17 you will get approx each 10th call:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/urllib3/response.py", line 443, in _error_catcher
    yield
  File "/home/pi/.local/lib/python3.7/site-packages/urllib3/response.py", line 566, in read
    data = self._fp_read(amt) if not fp_closed else b""
  File "/home/pi/.local/lib/python3.7/site-packages/urllib3/response.py", line 532, in _fp_read
    return self._fp.read(amt) if amt is not None else self._fp.read()
  File "/usr/lib/python3.7/http/client.py", line 461, in read
    n = self.readinto(b)
  File "/usr/lib/python3.7/http/client.py", line 505, in readinto
    n = self.fp.readinto(b)
  File "/usr/lib/python3.7/socket.py", line 589, in readinto
    return self._sock.recv_into(b)
socket.timeout: timed out

**Expected behavior**
The reliability of using url-commands should remain the same for version 3.15 and 3.17

**Screenshots**

**Additional context**
- I tried out:
  3.1.17-20230520012002 with arduino ESP32 libraries in version 2.0.2 (as this was rock-solid in the past)
  3.1.17-20230520012002 with arduino ESP32 libraries in version 2.0.9 (latest version)
  Both builds cause the same issues.
- "Downgrading" to 3.1.15 instantly fixes the issue.
- ESP32 PowerSave is de-activated.
- Upgrades/Downgrades were carried out OTA

fredlcore, 2023-06-06:
I have somewhat narrowed down the problem, but I can't explain it:
When doing the testing mentioned before, could you change the `client.write`line in `print2webclient.h`'s `flushToWebClient` function to this:
```
printlnToDebug(PSTR("1"));
    client.write(bigBuff, bigBuffPos);
printlnToDebug(PSTR("2"));
```
It seems to be that sometimes there is an (unexplicable to me) long delay between the output of `1` and `2`, apparently resulting in the timeout you mention. But since this is a general library call, I don't see where I can change that behaviour...

schlaubstar, 2023-06-06:
I was just testing so feedback to your initial question: 

- Problem is also there for official 3.1. release 3.1.17-20230520012002

[noMonitoring_BSBCap.txt](https://github.com/fredlcore/BSB-LAN/files/11662947/noMonitoring_BSBCap.txt)

I will modify the lines proposed and will check the results

fredlcore, 2023-06-06:
Also, are you using WiFi or Ethernet? If you have a possibility to switch (temporarily) to Ethernet, this might also be a helpful indication.
However, if you say that it did work rock-solid with a previous version, the only solution I can see right now is to go back in steps and install the software based on each previous pull request and see where the error takes effect. I'm willing to do all the changes necessary to fix this, but since I'm not on site right now, I have no means to do this process.

schlaubstar, 2023-06-06:
Maybe your "output 1/2 fix" already gives some indications. 
Although I did not dig into the code to deep but to me it looks as if the corresponding function sometimes gets called subsequently and "to often" 

``
POST /I10000=21.09 HTTP/1.1
1
2
1
2
1
2
1
2
set ProgNr 10000 = 21.09
setting line: 2.86257e-311 val: 05 45 00
LAN->HEIZ INF      3D2D0215 05 45 00
DC C2 00 0E 02 3D 2D 02 15 05 45 00 F5 B4
1
2
LAN->HEIZ INF      3D2D0215 05 45 00
DC C2 00 0E 02 3D 2D 02 15 05 45 00 F5 B4
POST /I10000=21.09 HTTP/1.1
1
2
1
2
1
2
1
2
set ProgNr 10000 = 21.09
setting line: 2.86257e-311 val: 05 45 00
LAN->HEIZ INF      3D2D0215 05 45 00
DC C2 00 0E 02 3D 2D 02 15 05 45 00 F5 B4
1
2
LAN->HEIZ INF      3D2D0215 05 45 00
DC C2 00 0E 02 3D 2D 02 15 05 45 00 F5 B4
POST /I10000=21.09 HTTP/1.1
1
2
1
2
1
2
1
2
set ProgNr 10000 = 21.09
setting line: 2.86257e-311 val: 05 45 00
LAN->HEIZ INF      3D2D0215 05 45 00
DC C2 00 0E 02 3D 2D 02 15 05 45 00 F5 B4
1
2
LAN->HEIZ INF      3D2D0215 05 45 00
DC C2 00 0E 02 3D 2D 02 15 05 45 00 F5 B4
``

fredlcore, 2023-06-06:
No, that the frequent calls are correct because the output is often times too large to fit in the buffer. You can see the contents to be sent if you change the code above to:
```
printlnToDebug(PSTR("1"));
Serial.write(bigBuff, bigBuffPos);
    client.write(bigBuff, bigBuffPos);
printlnToDebug(PSTR("\r\n2"));
```

I was a bit suspicious that it could be WiFi-related, but if you say that it worked rock-solid with 3.1.15, then that cannot be it...

schlaubstar, 2023-06-06:
As I am not so familiar with "expert functions" on GitHub I'd need some hints how to retrieve each version per pull request and how to restrict the search to those requests made after: Version 3.1.15-20230402082920
The rest should be fine. (Compiling / Uploading / Testing)

fredlcore, 2023-06-06:
You go to this page:
https://github.com/fredlcore/BSB-LAN/commits/master
And then go to the commit you want to test and click on the `<>`. Then the repository opens with the code at that time and you can download it as usual via the green download button.

But before you do this you might want to try this:
I erased the EEPROM using /NE and then flashed anew using the default BSB_LAN_config.h just modified for the networking parameters. Now I don't have any delays or timeouts anymore. If that is the same with you, it might be easier to isolate the causing factor by re-activating the configuration options from before and see where the problem starts to occur. Of course take note of all your web-configuration settings before erasing the EEPROM :)!

fredlcore, 2023-06-06:
I also realized that there is still a bug in the preferred BSSID functionality. If the BSSID parameter is empty, it will lead to a reconnect every 60 seconds. I'll fix this later, but you may want to look out for the serial monitor and see if that is a problem with you, too.

fredlcore, 2023-06-06:
And have you disabled Energy Saving for the ESP32 in the web configuration? I just realized that I had it turned on and that also resulted in bad connection.

fredlcore, 2023-06-06:
Enery saving seems to be the root cause of this. Try this first before anything else.

schlaubstar, 2023-06-06:
ESP32 PowerSave was and is constantly de-activated but I am already on May 16th ;-)

fredlcore, 2023-06-06:
Ok, too bad :(. But great that you're making such fast progress :)...

schlaubstar, 2023-06-06:
Some insights but no 100% explanation, yet.

- Enabling and using /NE did have a positive effect ... now the timeouts have reduced from ~10% to ~1% and I had to change my python script to wait less and perform more rounds
- I never achieved a timeout until commit: b87ea8dd3c16d67beeb08f551ed6117a06cfd01c
- I also encounter the frequent "WiFi Re-Scans" you mentioned with Versions having this option for a fixed BSSID when left in default config.
- I will enter the correct BSSID and try out a later version. I can't fight the feeling that the Wifi-Scans might interfere with the URL-command

schlaubstar, 2023-06-06:
Final insight: The problem is caused with the introduction of the ability to lock WiFi to a certain BSSID if you do not really use it and provide the MAC address.  As you observed this causes frequent "disconnects" where the ESP32 simply seems to loose connectivity for a short period of time, leaving existing requests "unhandled".

After compiling the latest commit (https://github.com/fredlcore/BSB-LAN/commit/960c326ed723896c0c1841aba5d5332118f3b746) but providing the MAC Address of my 2.4GHz Wifi ... I got 1000 requests with 1 sec. delay without any timeout or loss of connectivity and the Debug output also does not show the Wifi Scans anymore.

From my view you can close this issue, but I think it would make sense to reasonably deal with a "default BSSID"